dresden-elektronik/deconz-rest-plugin #4878: eWelink Zigbee Door sensor ZM-CG206

dioneimiodutzki, 2021-05-14:
## Device

- Product name: Zigbee Door Window Sensor for Tuya 
- Manufacturer: Tuya (?)
- Model identifier: ZM-CG206
- Device type : 
  - Sensor

### Aditional Info

This is a chinese generic door/window sensor (open/close sensor) but somehow is not listed like that on Deconz. 

https://www.aliexpress.com/item/1005002348194512.html?spm=a2g0s.9042311.0.0.19be4c4duwG9If

## Screenshots

[image]

### Power Configuration

[image]

### IAS Zone

[image]

Smanar, 2021-05-14:
I think it s not correclty included, it miss some cluster and have "ffff".
Can be repaired naturaly after a time, else you can use command from the first circle to finish it.

dioneimiodutzki, 2021-05-14:
> 
> 
> I think it s not correclty included, it miss some cluster and have "ffff".
> Can be repaired naturaly after a time, else you can use command from the first circle to finish it.

Sorry, I never hit an unsupported device before so I really don't know how to proceed. 
How can I try to send a command to finish it?

Smanar, 2021-05-14:
Easy, when you select the first circle on the node title, you will have a popup with some commands. Just resend them (specially the descriptor ones), for that you just need to click on the square.
But you will probably need to wake up the device in same time (this issue happen generaly when the device go to sleep during the inclusion)

But it can be repaired itslef after a time.

dioneimiodutzki, 2021-05-15:
Thanks! I was able to find the buttons and try, but to no luck, even waking up the sensor, the buttons have no effect. I tried then to repair the sensor (remove it on deconz and started the search on phoscon again, and the sensor appeared again the same way (ffff). I've bought a pack of 4 of those sensors and they are all with the same behaviour. Not sure if it will be fixable. Anything else I can try?
Again, thanks for the help.

Smanar, 2021-05-15:
IDK for this brand, but for aqara, pressing shortly the "reset button" every 2/3 s after the pairing procedure (long press) help to prevent the device going to sleep.

closed 2021-06-14 by github-actions[bot]