dresden-elektronik/deconz-rest-plugin #4882: API config "externalwindowopen" for Danfoss Ally thermostat works only for a few seconds

luka6000, 2021-05-16:
## Describe the bug

API config "externalwindowopen" for Danfoss Ally thermostat doesn't work properly.
Setting it true with API call makes it true only for a few seconds.
I'm talking about sensors -> config attributes -> ZHAThermostat -> externalwindowopen

## Steps to reproduce the behavior

curl -X PUT --data '{"externalwindowopen": true}' http://homeassistant.local:40850/api/91F4CF2034/sensors/14/config

## Expected behavior

Setting attribute externalwindowopen permanently.

## Screenshots

```
curl -X PUT --data '{"externalwindowopen": true}' http://homeassistant.local:40850/api/91F4CF2034/sensors/14/config
[{"success":{"/sensors/14/config/externalwindowopen":true}}]                                                                                                                                                 

while [ true ]; do date; curl -X GET http://homeassistant.local:40850/api/91F4CF2034/sensors/14; echo; echo; sleep 1; done
Sun May 16 13:52:23 CEDT 2021
{"config":{"battery":38,"displayflipped":true,"externalsensortemp":-8000,"externalwindowopen":true,"heatsetpoint":2000,"locked":false,"mountingmode":false,"offset":0,"on":true,"reachable":true,"schedule":{"W127":[{"heatsetpoint":2100,"localtime":"T07:30"},{"heatsetpoint":2000,"localtime":"T23:00"}]},"schedule_on":false},"ep":1,"etag":"0520faebf6b56a9715e92d561e27605f","lastseen":"2021-05-16T11:49Z","manufacturername":"Danfoss","modelid":"eTRV0100","name":"Bigroom trv","state":{"errorcode":"0000","lastupdated":"2021-05-16T11:26:56.503","mountingmodeactive":false,"on":false,"temperature":2279,"valve":1,"windowopen":"Closed"},"swversion":"01.08.0008 01.08","type":"ZHAThermostat","uniqueid":"cc:cc:cc:ff:fe:4e:02:b9-01-0201"}

Sun May 16 13:52:24 CEDT 2021
{"config":{"battery":38,"displayflipped":true,"externalsensortemp":-8000,"externalwindowopen":true,"heatsetpoint":2000,"locked":false,"mountingmode":false,"offset":0,"on":true,"reachable":true,"schedule":{"W127":[{"heatsetpoint":2100,"localtime":"T07:30"},{"heatsetpoint":2000,"localtime":"T23:00"}]},"schedule_on":false},"ep":1,"etag":"0520faebf6b56a9715e92d561e27605f","lastseen":"2021-05-16T11:49Z","manufacturername":"Danfoss","modelid":"eTRV0100","name":"Bigroom trv","state":{"errorcode":"0000","lastupdated":"2021-05-16T11:26:56.503","mountingmodeactive":false,"on":false,"temperature":2279,"valve":1,"windowopen":"Closed"},"swversion":"01.08.0008 01.08","type":"ZHAThermostat","uniqueid":"cc:cc:cc:ff:fe:4e:02:b9-01-0201"}

Sun May 16 13:52:25 CEDT 2021
{"config":{"battery":38,"displayflipped":true,"externalsensortemp":-8000,"externalwindowopen":true,"heatsetpoint":2000,"locked":false,"mountingmode":false,"offset":0,"on":true,"reachable":true,"schedule":{"W127":[{"heatsetpoint":2100,"localtime":"T07:30"},{"heatsetpoint":2000,"localtime":"T23:00"}]},"schedule_on":false},"ep":1,"etag":"0520faebf6b56a9715e92d561e27605f","lastseen":"2021-05-16T11:49Z","manufacturername":"Danfoss","modelid":"eTRV0100","name":"Bigroom trv","state":{"errorcode":"0000","lastupdated":"2021-05-16T11:26:56.503","mountingmodeactive":false,"on":false,"temperature":2279,"valve":1,"windowopen":"Closed"},"swversion":"01.08.0008 01.08","type":"ZHAThermostat","uniqueid":"cc:cc:cc:ff:fe:4e:02:b9-01-0201"}

Sun May 16 13:52:26 CEDT 2021
{"config":{"battery":38,"displayflipped":true,"externalsensortemp":-8000,"externalwindowopen":true,"heatsetpoint":2000,"locked":false,"mountingmode":false,"offset":0,"on":true,"reachable":true,"schedule":{"W127":[{"heatsetpoint":2100,"localtime":"T07:30"},{"heatsetpoint":2000,"localtime":"T23:00"}]},"schedule_on":false},"ep":1,"etag":"0520faebf6b56a9715e92d561e27605f","lastseen":"2021-05-16T11:49Z","manufacturername":"Danfoss","modelid":"eTRV0100","name":"Bigroom trv","state":{"errorcode":"0000","lastupdated":"2021-05-16T11:26:56.503","mountingmodeactive":false,"on":false,"temperature":2279,"valve":1,"windowopen":"Closed"},"swversion":"01.08.0008 01.08","type":"ZHAThermostat","uniqueid":"cc:cc:cc:ff:fe:4e:02:b9-01-0201"}

Sun May 16 13:52:28 CEDT 2021
{"config":{"battery":38,"displayflipped":true,"externalsensortemp":-8000,"externalwindowopen":false,"heatsetpoint":2000,"locked":false,"mountingmode":false,"offset":0,"on":true,"reachable":true,"schedule":{"W127":[{"heatsetpoint":2100,"localtime":"T07:30"},{"heatsetpoint":2000,"localtime":"T23:00"}]},"schedule_on":false},"ep":1,"etag":"114f4b87cc6d334cfcae886f6d825250","lastseen":"2021-05-16T11:49Z","manufacturername":"Danfoss","modelid":"eTRV0100","name":"Bigroom trv","state":{"errorcode":"0000","lastupdated":"2021-05-16T11:26:56.503","mountingmodeactive":false,"on":false,"temperature":2279,"valve":1,"windowopen":"Closed"},"swversion":"01.08.0008 01.08","type":"ZHAThermostat","uniqueid":"cc:cc:cc:ff:fe:4e:02:b9-01-0201"}

``` 

## Environment

 - Host system: Raspberry Pi4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26680700
 - deCONZ version: 2.10.4
 - Device: ConBee II
 - Do you use an USB extension cable: yes, USB2
 - Is there any other USB or serial devices connected to the host system? ssd on USB3

## deCONZ Logs

## Additional context

SwoopX, 2021-05-16:
Hm, apparently then only value sent on zigbee level is `false` no matter what has been sent via API... I'm currently running a heavily modified API version for setting config parameters, but I doubt this is the reason. Gotta dig some deeper.

luka6000, 2021-05-16:
> Hm, apparently then only value sent on zigbee level is `false` no matter what has been sent via API

I've added additional context that it works as expected when set with deconz UI.

hanskroner, 2021-05-16:
I suspect `data` might need to be converted to a `uint8_t` before being passed on to `addTaskThermostatReadWriteAttribute ` here
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/42ceb25b64a493fa86c7afd33b2375e5e4fe83f4/rest_sensors.cpp#L1987-L1993

I'll push a fix for that once I've gotten a chance to try it out and confirm that's the issue.

SwoopX, 2021-05-16:
That won't help, you can spare that time 😉

hanskroner, 2021-05-18:
From a very quick test of v2.11.05, this seems to be working again. Refreshing to know it wasn't something I broke, for a change 🤣

SwoopX, 2021-05-20:
Confirmation from Hans and myself should be enough to close this 😉

closed 2021-05-20 by SwoopX

luka6000, 2021-05-21:
Affirmative - works like a charm! Thank you!