dresden-elektronik/deconz-rest-plugin #4884: Conbee II NOT connect

RitaSghe, 2021-05-17:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Raspbian)
 - Firmware version: (26660700)
 - deCONZ version: (2.10.04)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: usb pen

## deCONZ Logs

## Additional context

Mimiix, 2021-05-17:
So you connected them with Phoscon?

RitaSghe, 2021-05-17:
Yes .....

RitaSghe, 2021-05-17:
I tried this istruction:
$ deCONZ --dbg-error=2 --dbg-info=1 -platform minimal

and I have this error ...
COM failed to open /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2431343-if00: 
Permission error while locking the device

Mimiix, 2021-05-17:
Aah. That's because you have the deconz service running.

You are trying to start another deconz. But you can't because deCONZ is already running and that deconz is using the stick.

Sudo systemctl deconz stop
sudo systemctl deconz-gui start

RitaSghe, 2021-05-17:
With ..
sudo systemctl deconz stop

I have ...
Unknown operation deCONZ

RitaSghe, 2021-05-17:
But I have two different port for deconz 40849 and 40850 ...
I shouldn't I have this problem ....

Mimiix, 2021-05-17:
My mistake:
sudo systemctl stop deconz
sudo systemctl start deconz-gui

deCONZ is the Headless version.

If you want to run the map you need to run deconz-gui instead of deconz.

RitaSghe, 2021-05-17:
OK, thank you ...

Mimiix, 2021-05-17:
YOu can, but it can't use the same stick.

RitaSghe, 2021-05-18:
When the last week I had installed Conbee II on my pc with kubuntu 20.04, I had configured one port for deconz and another port for deconz-gui and worked perfectly both ....
Ok, it's not a problem but it's weird .....

Mimiix, 2021-05-18:
I am not sure what you mean.

If you run deconz-gui, you get both deconz and Phoscon.

RitaSghe, 2021-05-18:
In Raspberry I have the Conbee II not connected (img.1)

[image]

and so I cannot see the network (img. 2)

[image].

To see the network form I have to stop deconz service .....

Well, in kubuntu I had not this problem; I had deconz service and deconz-ui service start both ....

Mimiix, 2021-05-18:
Deconz and deconz-gui are exactly the same. You can run deconz-gui and have the same functionality as deconz service + the gui....

You can not have 2 deconz running on 1 stick. that is impossible.

Smanar, 2021-05-18:
When an application use this kind of usb device, it use an exclusive connexion, so it s not possible to another app to use the same USB key.

RitaSghe, 2021-05-19:
Ok, I understand, but why you not responding to my question?
In kubuntu 20.04 I had Conbee connected and I saw network both and now with Raspberry I cannot do that?
In this guide:
https://indomus.it/guide/come-installare-e-configurare-deconz-su-raspbian-di-raspberry-pi/
is written as how set two different port to use both and (I repeat) in kubuntu it's work fine ......

RitaSghe, 2021-05-22:
UPDATE:
My Raspberry has two ip address:
lan 192.168.1.28
wifi 192.168.1.29
Today I realized that Conbee II is connected on 192.168.1.29 (wifi).
Can't be the reason of my problem?
Can I change the ip?

Smanar, 2021-05-22:
But you still have problem ?
Which one deconz version you want to use ?

RitaSghe, 2021-05-25:
Yes, I still have problem!!!!
I cannot see the network because I have the conbee disconnect, How am I going to explain?

[image]

RitaSghe, 2021-05-25:
The problem is :
Raspberry (Pi4) at start, started deconz and I cannot see the deconz-gui.
To see the network I got to stop deconz and start deconz-gui manually.
How can I start deconz-gui (at start of raspberry) in the place of deconz?

Mimiix, 2021-05-25:
@RitaSghe Deconz-gui = Deconz. IT is the same,.but then also adding the gui.

RitaSghe, 2021-05-25:
I don't understand if you make fun of me or what is the problem ....

Mimiix, 2021-05-25:
...?

As i said. you can run deconz-gui. It is the same as deconz but without the gui. If you have deconz-gui running, you dont need to have deconz running.

RitaSghe, 2021-05-25:
Evidently, we have communication problems, I give up, I'll try and get help elsewhere ....

closed 2021-05-26 by RitaSghe