dresden-elektronik/deconz-rest-plugin #4886: MoesHouse ZigBee Bluetooth Smart Brightness Thermometer Sensor Light Temperature Humidity Detector

MiAutomations, 2021-05-17:
## Device

https://s.click.aliexpress.com/e/_9HESSD

Product name: MoesHouse ZigBee Bluetooth Smart Brightness Thermometer Sensor Light Temperature Humidity Detector
Manufacturer: _TYZB01_kvwjujy9
Model identifier: TS0222
Device type : Please remove all unrelated device types.
Sensor

## Screenshots

[image]

Smanar, 2021-05-17:
Heelo, can you show the attribute table for the cluster 0x0001 (power configuration) pls ? after having press the "read" button.

MiAutomations, 2021-05-18:
> Hello, can you show the attribute table for the cluster 0x0001 (power configuration) pls ? after having press the "read" button.

Thank you for your reply, I finish right now the installation of the new version in the HA the deconz plugin and now is detected the device as you can see in the image bellow. But I still without data from the temperature after a host rebot the temperature data appears for a seconds but disappear after this ... Can you please help me in order to get this to work properly? Thank you in avance for your support and best regards 
[image]

Smanar, 2021-05-18:
So the humidity is working but not the temperature ?
Can you check the device json direclty ? You can have them using this link http://phoscon.de/pwabeta
It s the phoscon beta version then after go in help / API information

If the value is still 0, go in cluster 0x0402, attribute 0x0000 It need to display the temperature, if I m right if you try to read the value, it will update the value in the API (and HA)

MiAutomations, 2021-06-09:
Hello I still with the same issue 

please don't close this request 

Thank you

Smanar, 2021-06-09:
Still waiting for the JSON.

GabrieleCastellani, 2021-06-22:
I cannot see it in Phoscon I don't see any new sensor

 I can see it in Deconz here the screenshots:
[image]
[image]
[image]
[image]
[image]
[image]

in home assistant i see:
[image]

Thank you

Beast-Panik, 2021-06-22:
I have also bought these sensors, they can be taught but do not appear in the sensor overview
[image]
in IOBroker, however, the values appear, although not the correct or the same as on the sensor itself.
[image]

Smanar, 2021-06-22:
Ok, so I realy don't understand how you can have so differents results ?
- one have only 1 sensor created with "good value"
- one other have 4 but with strange value

They can be invisible in phoscon, this application use white/black list.

@Beast-Panik the sensor values are 38 39 40 41 ? Can you check in the JSON if the problem was from the api or the third app pls ?

Beast-Panik, 2021-06-22:
These are the values that are displayed in the API. 
`"38": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "ep": 1,
        "etag": "693cecfd8d1a13af593e333a0339bff1",
        "lastseen": "2021-06-22T14:04Z",
        "manufacturername": "_TYZB01_kvwjujy9",
        "modelid": "TS0222",
        "name": "LightLevel 38",
        "state": {
            "dark": false,
            "daylight": true,
            "lastupdated": "2021-06-22T14:04:27.036",
            "lightlevel": 26885,
            "lux": 488
        },
        "type": "ZHALightLevel",
        "uniqueid": "5c:02:72:ff:fe:05:19:17-01-0400"
    },
    "39": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "ad7b1df54feef10176f5ff5b0b37612a",
        "lastseen": "2021-06-22T14:04Z",
        "manufacturername": "_TYZB01_kvwjujy9",
        "modelid": "TS0222",
        "name": "Temperature 39",
        "state": {
            "lastupdated": "2021-06-22T14:02:04.990",
            "temperature": 0
        },
        "type": "ZHATemperature",
        "uniqueid": "5c:02:72:ff:fe:05:19:17-02-0402"
    },
    "40": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "b2b222aecf5a8b889748bba73d5534b5",
        "lastseen": "2021-06-22T14:04Z",
        "manufacturername": "_TYZB01_kvwjujy9",
        "modelid": "TS0222",
        "name": "Humidity 40",
        "state": {
            "humidity": 4990,
            "lastupdated": "2021-06-22T14:02:05.359"
        },
        "type": "ZHAHumidity",
        "uniqueid": "5c:02:72:ff:fe:05:19:17-02-0405"
    },
    "41": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "ep": 2,
        "etag": "f9c2ea6583f786250a8abcdc75c11e83",
        "lastseen": "2021-06-22T14:04Z",
        "manufacturername": "_TYZB01_kvwjujy9",
        "modelid": "TS0222",
        "name": "LightLevel 41",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "lightlevel": null,
            "lux": null
        },
        "type": "ZHALightLevel",
        "uniqueid": "5c:02:72:ff:fe:05:19:17-02-0400"
    }
}

                    
                `
and these are the current values
[image]

Smanar, 2021-06-22:
> "humidity": 4990,
> "lux": 488
> "temperature": 0

So only the temperature have issue ?

GabrieleCastellani, 2021-06-23:
It works for me the temperature is aligned with the display

[image]

GabrieleCastellani, 2021-06-23:
I can see only lux and battery in API (see above my JSON)
Can you give me hint on how to investigate on this?

Smanar, 2021-06-23:
You can set again phoscon in permit join, and read descriptors using the first circle in the node title, it will update sensors.
But you will need to awake the device.

And BTW, for this bug
>But I still without data from the temperature after a host rebot the temperature data appears for a seconds but disappear after this

Someone can share logs with the flag "info_l2" only during this action ? it will log websocket notifications.

Beast-Panik, 2021-07-01:
Sorry my sensor has a defect currently I can not provide values or comparisons. I am waiting for a replacement.
https://user-images.githubusercontent.com/55361239/124163853-51fba180-daa0-11eb-8e27-30983e308277.mp4

Smanar, 2021-07-02:
Ha, it have flashing strobe feature too ?

Seriously, it happen naturaly or after a bad manipulation ?

Beast-Panik, 2021-07-03:
I think it was already defective on arrival. Have moved it twice to other locations and from then on it had dropouts.

closed 2021-08-01 by github-actions[bot]