dresden-elektronik/deconz-rest-plugin #4890: _TZ3000_dfgbtub0

ReichertD, 2021-05-18:
## Device

- Product name: Scene Switch
- Manufacturer: _TZ3000_dfgbtub0
- Model identifier: TS0042
- Device type : 
  - Switch

[Moes Scene Switch at Amazon](https://www.amazon.de/gp/product/B08XJTJ379/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

## Screenshots

Cluster info:
First time add new switch (same switch as second and third time)
 
[image]

Second time add new switch (same switch as first and third time) and klicking a button
 
[image]

Third time add new switch (same switch as in first and second time)
 
[image]

Node info:
 
[image]

API Info Switch:
{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "516d6483ed4eb2cc7445852637b76551",
    "lastseen": "2021-05-18T10:25Z",
    "manufacturername": "_TZ3000_dfgbtub0",
    "mode": 1,
    "modelid": "TS0042",
    "name": "Schalter",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "84:71:27:ff:fe:e2:f3:81-01-0006"
}

API Info Light 35 (automatically added during add switch)
{
    "etag": "fae6ebde3ad6a8e99334839d85c9e621",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-18T10:25Z",
    "manufacturername": "_TZ3000_dfgbtub0",
    "modelid": "TS0042",
    "name": "On/Off light 35",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "84:71:27:ff:fe:e2:f3:81-01"
}

API Info Light 36 (automatically added during add switch)
{
    "etag": "3151ede34a3fea8d651dd66d3e320808",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-05-18T10:25Z",
    "manufacturername": "_TZ3000_dfgbtub0",
    "modelid": "TS0042",
    "name": "On/Off light 36",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "84:71:27:ff:fe:e2:f3:81-02"
}

PhosconApp:
Switches:
[image]

Lights:
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-05-18:
Hello, the result is "normal".

Do you have a linux machine to test a new code ?

ReichertD, 2021-05-18:
Hi,

yes I heard so from the support team. ;)
I ran deconz and phoscon in a docker container on a raspberry. Will this help?

Smanar, 2021-05-18:
>I ran deconz and phoscon in a docker container on a raspberry. Will this help?

lol, not for me, it s the worse situation ^^.
Not easy to compile the code on docker, you don't have a machine with full OS ? even a small raspberry.

gfxdude-dus, 2021-05-18:
hi,
is this one the same? [amazon link](https://www.amazon.de/gp/product/B08LB9WKCJ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1) have issues to pair or get any answers. running on ubtuntu 20.

Smanar, 2021-05-18:
Impossible to reconize a tuya device with a picture, the only way is including it on a zigbee network and check for the manufacture name.

ReichertD, 2021-05-18:
I have a Windows running, this will be the easiest for me. Maybe I will be able to setup an older Raspberry. Will try this out and will come back to you to get the code/beta.

Smanar, 2021-05-18:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_switch_4 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

Remove the old entries before.

You will need to edit too the json file. It will be in /usr/share/deCONZ/devices/button_maps.json and the change will be https://github.com/Smanar/deconz-rest-plugin/commit/1fd60e9ec2cd3477a0dc2ec10c00ce351c81bb50#diff-a3adce467fe93b9c612c198c187dad34f5284d6c9adf66aa28127f0f619979b5

ReichertD, 2021-05-19:
Hi, now I were able to do test some things.
I installed Ubunto VM and cloned your branch. After the make (need additionally "sudo apt install build-essentia") I updated the json, opened deCONZ and phoscon app and started a search for the switch. The switch were found and connected fast!
First I saw 2 on/off lights in phoscon, too. After a reboot these lights are gone in phoscon!

After that the switch is not usable in groups.
But(!) I connected this test phoscon bridge to my openhab and added the Switch to the things and created the channels. Now with your update I were able to get the events and states from the switch.

Looks like a good job!

Smanar, 2021-05-19:
This device don't have group cluster (like almost all tuya one), so no "magic" group for direct bind.

So can make the PR nothing is missing ?

ReichertD, 2021-05-19:
OK. Thanks.
Yes, from my point everything fine and well done! You can do the PR.
Shall I close the request or will you do that?

closed 2021-05-22 by manup