dresden-elektronik/deconz-rest-plugin #4892: unable to login when NGINX is in front of phoscon to provide HTTPS encryption

anubisg1, 2021-05-18:
## Describe the bug
as described.

If i access phoscon via HTTP on the DECONZ_WEB_PORT i can login no problem but if i place phoscon behind NGINX Reverse Proxy in order to provide SSL i can no longer login .

i can see that when i press the login button i get error 403 for the "/api" call and this error is returned:

[{"error":{"address":"/","description":"link button not pressed","type":101}}]

docker compose is the following:

```
version: '3.4'
services:
  deconz:
    image: marthoc/deconz:stable
    container_name: deconz
    restart: always
    network_mode: host
    volumes:
      - /volume1/docker/deconz/deCONZ:/root/.local/share/dresden-elektronik/deCONZ
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyACM0
    environment:
      - DECONZ_WEB_PORT=4080
      - DECONZ_WS_PORT=4443
      - DECONZ_VNC_PORT=5900
      - DECONZ_NOVNC_PORT=6080
      - DECONZ_DEVICE=/dev/ttyACM0
      - DECONZ_VNC_PASSWORD=changemehere
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
```

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Synology NAS
 - Running method:  Marthoc Docker container 
 - Firmware version: 26660700
 - deCONZ version: 2.11.05 / 5/9/2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no
 - NGINX reverse proxy is the built in NAS reverse proxy
 
[image]
[image]
[image]

## deCONZ Logs

## Additional context

CuBiC3D, 2021-05-20:
Having the same problem - can't login anymore. API returns 403 to that request:
```
:authority: deconz.domain.net
:method: POST
:path: /api/
:scheme: https
accept: application/json, text/javascript, */*; q=0.01
accept-encoding: gzip, deflate, br
accept-language: en-GB,en;q=0.9,de;q=0.8
authorization: Basic [redacted - base64 encoded "delight:password"]
cache-control: no-cache
content-length: 52
content-type: application/json; charset=UTF-8
dnt: 1
origin: https://deconz.domain.net
pragma: no-cache
referer: https://deconz.domain.net/pwa/login.html
sec-fetch-dest: empty
sec-fetch-mode: cors
sec-fetch-site: same-origin
sec-gpc: 1
user-agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36
x-requested-with: XMLHttpRequest
```
The response is:
```
[{"error":{"address":"/","description":"link button not pressed","type":101}}]
```
I'm running deconz in Kubernetes cluster with nginx ingress terminating TLS.

I can easily replicate through ingress:
```
❯ curl -X POST -H "Accept: application/json" -H "Authorization: Basic [redacted]" -d '{"devicetype":"Phoscon#[redacted]","login":"delight"}' https://deconz.domain.net/api/
[{"error":{"address":"/","description":"link button not pressed","type":101}}]
```
From inside pod:
```
# curl -X POST -H "Accept: application/json" -H "Authorization: Basic [redacted]" -d '{"devicetype":"Phoscon#[redacted]","login":"delight"}' http://localhost/api/
[{"success":{"username":"[redacted]"}}]
```

SwoopX, 2021-05-20:
I'm sorry guys, but this is neither any kind of bug nor is it related to deconz REST plugin. Therefore, I'm closing this issue. However, please raise your question in Discord. I know there's folks around actually using reverse proxies and it gives you a much wider spread.

closed 2021-05-20 by SwoopX

CuBiC3D, 2021-05-20:
@SwoopX can you elaborate on why this is an expected behaviour and why it got introduced? I downgraded to `2.11.3` and it works fine. Stops at `2.11.4`. Maybe #4864?

SwoopX, 2021-05-23:
Where have I mentioned that this is expected bahvior or that anything has been introduced? And Phoscon uses `/api/` ever since as far as I can tell.

So please follow my previous advice to seek some (way faster) feedback. We're discussing individual system setup issues way to often in the issues.

CuBiC3D, 2021-06-05:
tl;dr REST API can't handle headers that are lower case (not case-insensitive comparison).

Since I see some thumbs up here I assume there are couple of people running into this, update on that:

I found the issue and posted on discord as suggested but unfortunately it went lost in the chat (one reason why I think it's not a really suitable place to handle issues). I'll try to re-post it here.

> I'm sorry guys, but this is neither any kind of bug nor is it related to deconz REST plugin.

Looks like is a bug and related to REST plugin. The deconz REST API does not handle HTTP requests properly - according to RFC2616:
*Each header field consists of a name followed by a colon (":") and the field value. **Field names are case-insensitive**.*

I guess after the changes of converting headers to strings from version `.3` to `.4` headers are not converted to lowercase before comparison anymore. After some sniffing on the overlay network I could see that ingress-nginx transforms all headers to lowercase (as required for HTTP2) even if HTTP1.1 is used on the backend if a request was done using HTTP2 on frontend.

Quick workaround at least for now: Turn off HTTP2 support in Nginx which keeps the `authorization` header as `Authorization` for me.

JSchlarb, 2021-06-05:
@CuBiC3D: Great investigation! You safed me a lot of time. Downgraded back to `.3` as dropping http2 is not an option

Realted mr #4862

anubisg1, 2021-06-06:
@SwoopX  given the latest finding, don't you think this bug should be re-opened? 

@JSchlarb  how would you do that specifically? i'm using the container so i guess i just need to get the older version?

JSchlarb, 2021-06-06:
> @JSchlarb how would you do that specifically? i'm using the container so i guess i just need to get the older version?
Yep, simple change the image to `marthoc/deconz:2.11.03` :)

CuBiC3D, 2021-06-20:
Looks like it's been fixed in `2.12.1-beta`