dresden-elektronik/deconz-rest-plugin #4893: DIYRuZ_Flower

habitoti, 2021-05-18:
Please include this very useful soil moisture sensor: https://zigbee.blakadder.com/DIYRuZ_Flower.html

It's supported via ZIGBEE2MQTT and also ioBroker Zigbee. It shows up in Deconz, but is currently not recognised by Phoscon. It [has been requested before](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4624) by [@ingo01978](https://github.com/ingo01978) , but was automatically closed after a month w/o any comment. It's very sad to be locked in like that with Deconz and not be able to extend it through community :-( 

## Device

Product name: DIYRuZ_FlowerV2
Manufacturer: modkam.ru
Model identifier: DIYRuZ_Flower
Device type : Sensor , Other (Soil moisture)

## Screenshots

[image]

[image]

[image]

[image]

### Basic

https://modkam.ru/?p=1700
https://github.com/diyruz/flower

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

habitoti, 2021-06-07:
Is this the right way to ask for additional integrations? Pretty silent here...

Smanar, 2021-06-07:
Yes, it s the good place ^^.
But I think there is no one with free time for this device.

And As cluster are not the good one, need some hacks in the code too.

habitoti, 2021-06-07:
Hm. Seems to be the only outdoor Zigbee humidity sensor that I could find, and I am seeing lots of people switching to ZIGBEE2MQTT just for the support of this.
Can't you provide kind of an "open integration" type of proxy where 3rd parties (or nice hobbyists) can hook in with generic additional interfaces to their hardware? Whenever new devices pop up it feels we're locked-in with Deconz until it officially delivers support through an update, while it's not possible to just dynamically add something simply at runtime. This would relief that waiting cycle and would also allow prof. and hobby devs to work on preliminary or "casual" integrations until an official one can be delivered (if that happens at all).

Smanar, 2021-06-07:
The code is open, some modification just need 4 lines to edit, and to compile people just need to copy paste 5 lines from the homepage.

People are afraid by c++, even it s no more complicated to edit a c++ code than a javascript one.
But yes, the API "will be improved soon", to make new device addition easier by users.

And your device will not be "basic" to include, as we need to manage "output" cluster as "input" cluster.

guny74, 2021-06-19:
I also use the conbee II and deconz, because a lot of people suggest to do it theright way.
But now I have 4 of this sensors (flower moisture sensors) and I have to switch to an other cc2552 controller
and zigbee2mqtt to support this. At the moment I have both running, but this is not my preferred solution.

So I also would be very happy if the conbee II / deconz can work with this device....

Thank you

TPertenava, 2021-07-11:
Hi, just wanted to add my 5 cents here. I’ve been using DeCONZ for quite a while now, thanks for making it!
But now had to switch to zigbee2mqtt which is not that user friendly, just to get support of my 15 brand-new soil mosture sensors.
It would be really nice to have an option to either add them to DeCONZ natively or configure them as a “generic” device.
Thanks in advance!

guny74, 2021-07-11:
Hello,

Any idea if / when this device will be supported ? I am at the decision to switch to another stick & zigbee2mqtt incl. to leave conbee+deconz or - my preferred way - conbee/deconz support also such flower moisture stick.

Thank you

Regards

Smanar, 2021-07-11:
If you realy need this device, I think you can use another project.
This device need hack to be used, and ATM I think one of the bigger problem on deconz is removing all hacks.

Smanar, 2021-07-11:
It's just my opinio, but this device is not zigbee compliant, so need hacks to make working and ATM one priority in deconz is removing all hacks, so I don't think this device will be in the todo list one day.

TPertenava, 2021-07-12:
@Smanar IMO it is totally valid point, it is better to stay within standards to avoid hacks. I would like to give it a try and communicate with the DIYRuZ_Flower FW developer if it may be tuned to match the standards.
Would you be able to describe what exactly is wrong with that device in a nutshell, so it would be obvious to the FW author?

Smanar, 2021-07-12:
Ha yes, if you can make something to correct it, I will make the device integration myself ^^.
So my problem is simple, clusters are output cluster, instead of input, on the screenshoot they are grey instead of blue, and it s for that you have (0) as attributes available (except for luminance, but here the bug is from deconz I think.)

guny74, 2021-07-13:
Hello,

Maybe the link to the github Page will be helfen.
https://github.com/diyruz/flower

I am not an expert, but in the lower Part these are Screenshots from other Software zigbee2mqtt where input and output Cluster seams ok...or I am wrong?

I have also some running at zigbee2mqtt and could deliver Info.... or I cool also try to connect them to the deconz on windows (so far not successfull)

Smanar, 2021-07-14:
He nope look at this one https://raw.githubusercontent.com/diyruz/flower/master/images/DIYRuZ_Flower_14.png, all usefull cluster are in output
But i s an old version, you realy don't have same.

guny74, 2021-07-14:
[image]
Here at zigbee2mqtt webui it lokks like this....

guny74, 2021-07-14:
Ok, so only change of the device firmware would solve this, right !? "sh..."

(I do not know about the meaning of cluster, so I could only deliver screenshots)

Smanar, 2021-07-14:
Yep ^^
You can look all devices with temperature or humidity, for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4801
Clusters are always blue.

guny74, 2021-07-14:
So I think the problem is, that we are / I am no experts describing the problem to the firmware project, and if there would be a change to the firmware, the other project (zigbee2mqtt, iobroker) which supports it now in this version, it would not work anymore ( I fear).

Any other ideas than to use second controller+zigbee2mqtt ?

TPertenava, 2021-07-21:
> Ha yes, if you can make something to correct it, I will make the device integration myself ^^.
> So my problem is simple, clusters are output cluster, instead of input, on the screenshoot they are [...]

Hi Smanar, with some very valuable help from the original FW developer, it seems I've got FW tuned, so all the EPs are blue now. I am able to read all the values besides the main "unsupported" one (cluster 01, endpoint 0408, attrid 0x0002) which stands for soil humidity.
Please let me know what would be next steps to get this sensor supported by DeCONZ. Thanks a lot for your willingness to help!

[image]

Smanar, 2021-07-21:
Gz ^^, this device seem to have a good support.

I will do it tommorow, just some questions.
I will use defaut value for cluster 0x0400 0x0402 0x0403 0x405
The cluster 0x408 is new on deconz, I will add it.
Just need the attribute table on cluster 0x001 to be sure on wich one attribute you have voltage.
And I have a question, why 2 endpoint with 2 temperature cluster ? both are usefull ?

TPertenava, 2021-07-21:
> Gz ^^, this device seem to have a good support.
Indeed!
> 
> Have started the code, you will have it tommorow, just some questions.
> I will use defaut attribute for cluster 0x0400 0x0402 0x0403 0x405
> The cluster 0x408 is new on deconz, I will add it.
Great, thanks!
> Just need the attribute table on cluster 0x001 to be sure on wich one attribute you have voltage.
[image]
0x0020 is for battery voltage (should multiply by 100 to get voltage in mV)
0x0021 is for battery percentage (in percents)
> And I have a question, why 2 endpoint with 2 temperature cluster ? both are usefull ?
Jep, both are useful. The first one is for air temperature and the second one is for soil temperature.

habitoti, 2021-07-22:
Question not directly related to deconz, but a prerequisite then: how would I update the firmware of the sensor once fixed? So far I did that via deconz, which would be a hen&egg problem now... 😉

Mimiix, 2021-07-22:
@habitoti That's a different question. Please ask that in the forums (forum.phoscon.de) or read the documentation of deCONZ :)

Smanar, 2021-07-22:
Ok so I have choosed the battery level. Code is updated

You can try the code and tell me what is working or not.

I don't understand your problem for firmware ? If you already do it with deconz, you can continue with it, just need the firmware file, and BTW if you can share it too I think others users will be interested too.

habitoti, 2021-07-22:
While the sensor shows up in the deconz map, it's not in the OTAU plugin list, so how would I push a firmware then?

Smanar, 2021-07-22:
>While the sensor shows up in the deconz map, it's not in the OTAU plugin list, so how would I push a firmware then?

But the device is visible on the deconz map with the bad firmware , just the cluster are not on good side, no ?

TPertenava, 2021-07-22:
> While the sensor shows up in the deconz map, it's not in the OTAU plugin list, so how would I push a firmware then?

Hi, am I getting it right you are trying to update the SW on the sensor itself? I am not 100% sure it supports OTA updates at all. I think the easiest way would be to just reprogram it from scratch. The updated FW is now found in releases section of Flower sensor depository [https://github.com/diyruz/flower/releases](url)

TPertenava, 2021-07-22:
> Ok so I have choosed the battery level. Code is updated
> 
> You can try the code and tell me what is working or not.
> 
Thank you, @Smanar, really overwhelmed with your support. I will try to figure out the testing procedure as per your instructions and get back here as soon as I get it checked!

habitoti, 2021-07-22:
> Hi, am I getting it right you are trying to update the SW on the sensor itself? I am not 100% sure it supports OTA updates at all. I think the easiest way would be to just reprogram it from scratch. [...]

Doesn‘t seem to support OTA. As said it‘s not showing up in OTA plugin.
Maybe it‘s just me, but I have no clue how to get the firmware uploaded to the device. The device or firmware site don‘t mention anything in this regard. Any supporting hint welcome 😬

TPertenava, 2021-07-22:
> Maybe it‘s just me, but I have no clue how to get the firmware uploaded to the device. The device or firmware site don‘t mention anything in this regard. Any supporting hint welcome 😬

I used this tutorial together with Arduino board I had laying around. https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html#via-arduino-unoesp8266-with-ccloader-3min
Just give it a try, if no luck when I think the most appropriate would be to ask help in Flower rep. You can tag me there if you want.
Pinout:
Vcc - 3.3V from Arduino
RST - pin 4
P2.2 - pin 5
P2.1 - pin 6
GND - GND from Arduino

Smanar, 2021-07-23:
> I am not 100% sure it supports OTA updates at all.

Right, the device have not the OTAU cluster.

guny74, 2021-07-30:
I flashed the sensor with the newest firmware.

I the device code in the "latest" Version right now ? I user the docker version

TPertenava, 2021-07-30:
> I flashed the sensor with the newest firmware.
> 
> I the device code in the "latest" Version right now ? I user the docker version

Hi, I guess it is not in the main branch as for today. It is on me to build deconz-restapi from @Smanar ’s repository and test the functionality. Unfortunately I’ve been struggling to make my lab setup working at all when building it from development branch. Hope to get this done by Monday and give feedback to Smanar.
Meanwhile you can do the same as instructed by Smanar [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4893#issuecomment-884441056)

Mimiix, 2021-07-30:
@TPertenava are you the developer of the firmware?

TPertenava, 2021-07-30:
> @TPertenava are you the developer of the firmware?

No, I am not. But I've consulted the developer and made a pull request to fix the in/out cluster issue which was accepted into the main branch of the sensor FW.

habitoti, 2021-07-31:
So I flashed the device now successfully with the latest release firmware, however in deconz I am still seeing all those old values (and e.g. nothing under Power Configuration, like shown above). Should I expect it to show correct data now in the cluster infos?

Smanar, 2021-08-01:
>So I flashed the device now successfully with the latest release firmware, however in deconz I am still seeing all those old values (and e.g. nothing under Power Configuration, like shown above). Should I expect it to show correct data now in the cluster infos?

You mean same color for blue an gray cluster ?
At opposite to this one  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4893#issuecomment-884433478 

Try to delete and re-include the device.

habitoti, 2021-08-01:
Removing and adding back did the trick, thx. 
I am seeing now properties for most clusters:

[image]

However all report 0/empty values only. Is that expected at this stage w/o according deconz update, or should I see any meaningful values for the properties already?

[image]

[image]

The only two properties from all listed are manufacturer and model id:

[image]

Smanar, 2021-08-01:
Have you used the "read" button ?

And this device is a sensor, so you probably need to awake it to have values.
BUT what is your machine ? Can you try to compile the deconz code too, to test the device inside the API ?

habitoti, 2021-08-01:
Read doesn't work (sensor shows red bullet then). When would the sensor push values? On change only or regularly? How would I awake it explicitly?

I am running Deconz on Homeassistant (with Conbee II attached to a PI 4). Not sure it's so easy (for me 😬) to update the docker sandbox it's running in...

Smanar, 2021-08-01:
>Read doesn't work (sensor shows red bullet then). When would the sensor push values? On change only or regularly? How would I awake it explicitly?

I think it s exactly the problem ^^
The GUI don't use queue list, if the device is sleeping you will have the red error, on some device, a short press on the reset button make the trick.
Or just wait for natural reporting, but not sure the GUI update them in real time.

And no, docker are not easy, there is a procedure https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing , but I realy can't help you for it.

habitoti, 2021-08-01:
When pushing reset, I can exec a „read“ and the bullet in deconz VNC flashes green shortly. But all values stay 0, especially the „measured value“ for temp. and humidity.

Smanar, 2021-08-01:
You don't have same result than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4893#issuecomment-884433478

You need to have too some attributes disabled. and the color need to be the blue, not the green.

habitoti, 2021-08-01:
Yeah, seems like it has a very short "awake" time to read the data. So playing a bit with the timing, I get e.g. the power config.:

[image]

For humidity it's still 0, though the others get disabled then. The sensor is standing in a glas of water right now, so I suppose the reading should be different from 0 then:

[image]

Same for temperature:

[image]

Smanar, 2021-08-01:
Ha yes it s stange here ....
But it have worked for the other user, can come from the firmware, perhaps it need some hours of working mode before sending an average ...

habitoti, 2021-08-01:
Yeah, I'll wait another 24h, let's see whether it's "grooving in"...

TPertenava, 2021-08-01:
Hi everyone, I've finally managed to test it under beta setup with @Smanar 's branch of the rest plugin applied. Beta version obviously has some strange bug which prevents any sensor to pair with the coordinator (deCONZ just NOT entering into pairing mode, this is totally unrelated to this topic), but finally it just worked somehow, so I can confirm it works and I was able to read all the correct values via deCONZ GUI.
Please let me know what has to be tested/verified before this sensor may be included into the main repo?

@habitoti there is literally no lag with values readout, so no reason to wait 24h. Every time you push _physical_ button on the sensor it reads current Soil humidity (or any other) value and sends it out to coordinator. Note you should also press READ button in the cluster info, it _does not_ request values from the sensor, but it will keep updating numbers in the table for some time. 
[image]

habitoti, 2021-08-01:
Hm…what could be the reason for cluster 408 to show up as „unknown“ with no properties on my sensor after flashing the new firmware, while yours seems to correctly report „soil moisture“ here? All other clusters are the same. Do you have any property readings <> 0 for any of the other clusters?

TPertenava, 2021-08-02:
> Hm…what could be the reason for cluster 408 to show up as „unknown“ with no properties on my sensor after flashing the new firmware, while yours seems to correctly report „soil moisture“ here? All other clusters are the same. Do you have any property readings <> 0 for any of the other clusters?

The reason for 408’s name  not showing up properly is that it is non-standard cluster for deCONZ and you need to use modified rest plugin and xml file to fix that.
Yes, I have got correct readings in deCONZ GUI for all sensors (2xtemperatures, pressure, RH%, soil moisture).

Smanar, 2021-08-02:
Ha yes the xml file is /usr/share/deCONZ/zcl/general.xml on Raspberry
And the new file is here https://github.com/Smanar/deconz-rest-plugin/blob/flower/general.xml

>Please let me know what has to be tested/verified before this sensor may be included into the main repo?

Just the working mode in the API, it need to be visible in third app (except the ZHAMoisture, this one will be visible only in the API, can use http://phoscon.de/pwabeta in help / api Information)

habitoti, 2021-08-02:
What will be the regular timeline when it would show up in Homeassistant?

Smanar, 2021-08-02:
Officialy, IDK, we are relay late ATM for PR validation.
But if you compile it on your side, immediatly.

guny74, 2021-08-09:
What are the next todos  and timeline, that the sensor is completly indegrated in deconz / Photon app (incl docker version) ?

Smanar, 2021-08-10:
I just need your confirmation all is working using the API (for exemple in HA) and I will make the PR.

guny74, 2021-08-11:
I (personaly) have a problem to test ist, because I use Deconz as Docker....and the "test" description seams  not to  work in the docker image .... or I do it wrong....

Smanar, 2021-08-11:
I can help you for docker, but you have followed the procedure

```
git clone https://github.com/franciscogouveia/docker-deconz-rest-plugin-test.git
cd docker-deconz-rest-plugin-test
docker pull marthoc/deconz
docker build -t my-deconz-image --build-arg PLUGIN_REPOSITORY=https://github.com/Smanar/deconz-rest-plugin.git --build-arg PLUGIN_GIT_COMMIT=flower
```

guny74, 2021-08-12:
No, I tried the commands descriped upwards and tried them inside the docker container....

I will now try your commands....

guny74, 2021-08-12:
After that, how could I exaclty test the sensor / api ?

guny74, 2021-08-12:
I had to change the docker file to arch64 instead of amd64, because I use an RPI4, but then I get an error, that deconz / deconz-dev is not in package list....

[image]

guny74, 2021-08-14:
I also tried with arch=arm64 ...same issue.
Any Idea ? I really want to help to test it, but....

Or any other could test ist and it goes to the "offical" release....

Thanks

Smanar, 2021-08-14:
I prefer someone can test it, at least one time to be sure sensors are visible and send good vlaues in the API.

No one can make a test on a VM, you are all using this fu***ng dockers ?

[29 more replies]

closed 2021-11-09 by manup