dresden-elektronik/deconz-rest-plugin #4894: CC2652RB stick with Konkee Firmware (used as router)

samsam-rolon, 2021-05-19:
## Device

- Product name: https://slae.sh/projects/cc2652/#the-stick
- Manufacturer: TI CC2652RB 
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Other: Dev stick as router or coordinator
https://github.com/Koenkk/Z-Stack-firmware/

## Screenshots

[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-05-19:
And what do you want to do with it ? Display it as entry in the API ?

Mimiix, 2021-05-20:
IT is already repeating right?

You can't use it as coordinator as deCONZ only works with Conbee/Raspbee.

samsam-rolon, 2021-05-20:
Yeah ! I'm using it as repeater. Yeah It would be nice to see it in api (and GUI) to facilitate pairing first, then to track it (alive or not) and finally for any evolution regarding issue concerning the network map readable through API ?

Smanar, 2021-05-21:
>any evolution regarding issue concerning the network map readable through API

There is 2 Pr about that in waiting list, go make notification ^^

For your device, I have a problem, the device type = 0x00FF.
There is realy no way to chnage it for exemple in 0x0008 for "Range extender"

0x00FF mean "reserved", not fantastic to make something generic.

samsam-rolon, 2021-06-14:
> 
> 
> > any evolution regarding issue concerning the network map readable through API
> 
> There is 2 Pr about that in waiting list, go make notification ^^
> 
> For your device, I have a problem, the device type = 0x00FF.
> There is realy no way to change it for exemple in 0x0008 for "Range extender" as the code is editable no ?
> 
> 0x00FF mean "reserved", not fantastic to make something generic.

I didn't see stuff like that in firmware ... :-(
I used this firmware : https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin

No matter, deCONZ GUI show it correctly so let's keep it like this !

closed 2021-07-15 by github-actions[bot]