dresden-elektronik/deconz-rest-plugin #4897: _TZ3000_bmg14ax2

ReichertD, 2021-05-19:
## Device

- Product name: TEEKAR Drahtlose Türsensor, ZigBee Smart Door Window Alarm, Wireless Security Alarm, APP Remote Notification, geeignet für ZigBee Home Automation -ZigBee Gateway erforderlich [Link](https://www.amazon.de/gp/product/B08XQN4WSP/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1)
- Manufacturer: _TZ3000_bmg14ax2
- Model identifier: TS0203
- Device type :
  - Sensor (for door/window)

## Screenshots
Node Info:
[image]

Cluster Info:
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

pavime, 2021-06-02:
Hello,
You may add this similar device at the same time

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4340

Thank you

Regards

Smanar, 2021-06-03:
Hello, do you have linux machine to test code modification ?

ReichertD, 2021-06-08:
Hi, 

sorry for the late answer. Currently I do not have the possibility to test the code.
I use deconz in docker on rasperry.
Hopefully I will receive a second sd card next days, then I will be able to setup a second environmet without docker only for test. If so I will let you know.

Regards, Dominik

ReichertD, 2021-06-30:
hey, sorry for the late response. I got the sd cards last week but I were not able to test anything because I had a broken arm, too.
So next week I am on holiday, but when you may explain me how to do the test will try to do it asap.

Regards

Dominik

closed 2021-07-30 by github-actions[bot]

270R, 2021-08-05:
Hi, 
I've just updated to Version 2.12.03 (12.07.2021) and Firmware 266F0700, but the device (Model: ZM-CG205, Zigbee ID: TS0203) is still not found. 
(I could test on a debian machine in headless mode.)

Smanar, 2021-08-05:
This device is not added yet, because I can't find someone to test it.
A debian machine is good, but it s too late for today, will make the code tommorow

Smanar, 2021-08-06:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_door_sensor_3 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

You haven't provided capture so I m using the same than on the first post
```
{ VENDOR_EMBER, "TS0203", silabs11MacPrefix }, // Tuya door sensor TEEKAR
const quint64 silabs11MacPrefix   = 0xcc86ec0000000000ULL;
VENDOR_EMBER = 0x1002
```

Rick-Git, 2021-08-15:
Hi! I tried this, with the manual. Because I use Docker, even with the manual, it failed :-(
That's annoying. 

Now I have 2 Tuya devices that aren't working, that's also kinda funny because I didn't check compatibility first. 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4904

Is there another way to test it?

Smanar, 2021-08-15:
>even with the manual, it failed :-(

The compilation have failed, and the test ?

>Is there another way to test it?

Sure, on a classic Linux OS for exemple, it s possible too on VM, only using docker need lot of skills

Rick-Git, 2021-08-15:
Some random error.
But don't want to compromise my production Linux machine :-)

So, I will be setting up a RPI3b tomorrow. I will test both....

Rick-Git, 2021-08-16:
Soooooo... https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4904# worked out.
This one isn't. Phoscon is not connecting, but the device is in some way in my network via deconz.
But, after a stop/start, it's gone again. 

I'm a newby with this stuff, please be patient :-)

Smanar, 2021-08-16:
Lol, np, I m like you, I discover the device.
So you mean at a moment you are able to see it in deconz, but not for a long time ?

As the capture was not from you, have you compared with your device ?

```
{ VENDOR_EMBER, "TS0203", silabs11MacPrefix }, // Tuya door sensor TEEKAR
const quint64 silabs11MacPrefix   = 0xcc86ec0000000000ULL;
VENDOR_EMBER = 0x1002
Manufacturer: _TZ3000_bmg14ax2
```

Rick-Git, 2021-08-17:
I just hate IT-stuff frequently..... And, I'm in IT for 20 years already.... 
That said....

I tried connecting this device to my test and production Deconz/Conbee. (btw, I cleared the testplugin first at testsystem, just to see differences, but none). 
Result is the same. The device isn't showing up in a good way. 

[image]
[image]

That's it. Tried to push the button a couple of times to push some new info.

Maybe it's bwoken! (F1 joke)

Mimiix, 2021-08-17:
It isn't paired properly  @Rick-Git ,

Mimiix, 2021-08-17:
Despite being written for Xiaomi, you can try this:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

Rick-Git, 2021-08-17:
Right. Thanks.

Rick-Git, 2021-08-17:
@Mimiix 
No luck. Tried it several times, including helping hands

Mimiix, 2021-08-17:
Can you share some logs while pairing?

Smanar, 2021-08-17:
Ha yes right, this device have somes issue to connect.
It use a router or direct connexion ?
Try to press shortly the button, to prevent the device going to sleep during the inclusion. Are you sure for battery level ?

Rick-Git, 2021-08-17:
@Mimiix I used 
`deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1 > log
`
When pairing, what part do you want to see? It's long

@Smanar 
Direct connection, tried, idk

This sucks.

Mimiix, 2021-08-17:
Can we have APS on 2 and add - -dbg-error=2?

Also, just post the whole log :)

270R, 2021-08-17:
Soooo, sorry for the long break. It's a busy period. Here are some infos from the device I have on hands. 
[image]
[image]

Rick-Git, 2021-08-17:
@Mimiix Sure, I will generate again.
You were just in time, testing a other Zigbee device also and was cleaning the files haha....

270R, 2021-08-17:
Just in case another log is required: 
(Quite long, too.)
[2021_08_17_log_GUI.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7001342/2021_08_17_log_GUI.txt)
0x00124b001bdc0962

Smanar, 2021-08-17:
Next time, pls avoid --dbg-http=1, too much talkative.
But it s not the same device than on the first post ?
And on your capture the device have left the network again ? The name is grayed.

270R, 2021-08-18:
Hi,
in fact name is different (either ZM-CG205 or TY-ZDR06, both mentioned on package/manual) but ID is TS0203. It's following device: https://zigbee.blakadder.com/Tuya_ZM-CG205.html

Smanar, 2021-08-18:
But your device seem have so much connection issue.
- It have left on the capture
- Still missing the manufacture name and model ID.
- The Mac address range seem too much different.

270R, 2021-08-19:
I've just setup a clean ubuntu VM for testing and the device got linked in deconz: 
[image]
How can I obtain further information for this device?

Smanar, 2021-08-19:
This time it is still in the network, but inclusion have failed again ...

No more thing to do than the one in this link https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues since step 5

270R, 2021-08-19:
Mhmm I tried, but no difference....
It's visible in deconz, but implementation in Phoscon App failed and I can see "actions" indicated in deconz-gui but I cannot read node desciption etc....

Rick-Git, 2021-08-19:
Same, same....

270R, 2021-08-23:
Soooo, is there anything that can be done?
From MAC POV it seems to be similar to this device: https://github.com/Koenkk/zigbee2mqtt/issues/7553

Smanar, 2021-08-23:
Not realy much ....
If the device is not reconized in deconz, we can do nothing with the API.

And I don't see a special trick to make it better included.

270R, 2021-09-06:
Not sure if this helps but these sensors connect flawlessly to a tuya gateway.

270R, 2021-09-06:
I opened the device, to check the inside and it seems to be a CC2530 (F256) SOC from TI.

[image]
[image]
[image]

270R, 2021-09-20:
Would it be beneficial to use ZSHARK?

Smanar, 2021-09-21:
If you can't include it correctly in deconz, not realy.
You can see the problem in Zshark, but will be another story to search why this device can't be included.

270R, 2021-09-21:
So to sniff the communication between Tuya gateway and sensor doesn't change anything.... that's a pity.

270R, 2022-01-31:
Here's a short update on some information I've obtained through Zigbee2MQTT: 
modelZigbee:TS0203
type:EndDevice
nwk:30423
manuf id:0
manufacturer:_TYZB01_xph99wvr

270R, 2022-02-06:
It's in the network and information is transmitted, but not shown in phoscon: (Zone Status 0 | 1 - closed | open).
[image]

Smanar, 2022-02-07:
But the device work in deconz now ?
On previous post, it was not possible ?

270R, 2022-02-08:
> But the device work in deconz now ? On previous post, it was not possible ?

As I tested the latest beta for another device on test VM (Smart plug, where DDF shall be modified to show metering, voltage & current), I gave this sensor a shot. Adding in Phoscon Web App failed, but it appeard in deconz Gui as regular sensor and transmitted status information. Unfortunately Manufacturer and model ID is still missing. 
From Zigbee2MQTT I get manuf id:0, manufacturer:_TYZB01_xph99wvr;

Smanar, 2022-02-08:
>Unfortunately Manufacturer and model ID is still missing.

Awaking the device (using the button or the device itself) when trying to read them still not working ?

270R, 2022-02-08:
FYI it survives reboot, too.
[image]
[image]