dresden-elektronik/deconz-rest-plugin #4898: Moes Scene Remote (4 button switch - battery operated)

Sjonnie2018, 2021-05-19:
## Device

- Product name: Moes Scene Remote (no model number on package bujt probably ZS-SR4-2169-MS)
- Manufacturer: 0x1002 (_TZ3000_abci1hiu)
- Model identifier: TS0044
- Device type :
  - Remote
 
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

## Additional info

The remote has three settings per button (4 buttons in total). Link to the device: https://www.moeshouse.com/collections/zigbee/products/tuya-smart-life-zigbee-smart-home-wireless-switch-4-gang-remote-point-to-point-control

Smanar, 2021-05-19:
I m seing it on the code I have, what is your deconz version ?
Have you done the double inclusion ?

Sjonnie2018, 2021-05-19:
Hi Smanar,

Thanks for your fast reply!

Deconz version: 2.11.05
Conbee II firmware version: 26660700

I am not sure what you mean about "double inclusion" and "problem with Mac address filter". Sorry for my Noobiness :-|

Smanar, 2021-05-19:
2.11.05
You need to have it.

You realy have nothing ?
Even in "light" device ?

For almost all tuya stuff, you need to include the device 2 time, without deleting the older entry, the code use the data retreived on the previous inclusion to reconize the device on the second one.

But the code use Mac adress filter, and your device have one "less standard" than other `0x60a423....`
If it s that, I need to update the code.

Can you share logs during the inclusion ? with "info" and "info_l2" ?

Sjonnie2018, 2021-05-19:
Hi Smanar,

I have the remote in the switches section of Phoscon but no option to add the remote to a scene (it doesn't show up in Phoscon "wizard").

I have tried adding the device multiple times using all the buttons and made sure I added the device using the first button more than once. Made no difference.

How do I get the logs? Can you tell me? Than I'll be more han happy to share :-)

Smanar, 2021-05-19:
> have the remote in the switches section of Phoscon but no option to add the remote to a scene/group (it doesn't show up in Phoscon "wizard").

So it can be normal, phoscon don't support all devices.
But if you can see it, I think it will work on third app.

I think all is ok, what is your third app ? if you can see it in phoscon, it mean the device is in the API.

Sjonnie2018, 2021-05-20:
Sorry for my late reply. I use Domoticz as my primary application. I will try and do some testing later today and report back.

Smanar, 2021-05-20:
>Only thing I don't understand is the ID that changes when pressing button 2 (54). All other buttons have the same ID (53)

Hu ? this is not normal, you need to have only one entry, if you realy have that can you share the 2 device JSON for the ID 54 and 53 ?
It s possible on domoticz on custom/deconz else you can use help/api information with this url https://github.com/dresden-elektronik/phoscon-app-beta

>EDIT: Would you like me to open an issue on the Domoticz forum if it is at all possible to change the properties/behaviour of this device?

Yep if you have this kind of issue, better to use the plugin github or domoticz forum. Except if you have different ID (mean there is a problem in the API), your result seem normal, so if domoticz have issue it's the plugin fault.

Sjonnie2018, 2021-05-21:
Hi,

Thanks for your relentless support! Much appreciated!!

I deleted the node in DeCONZ and re-added it. I thought that if I made an error configuring it would be best to start over. The remote is stubborn though. I now have ID55 and ID56 as new devices.

The logs and the JSON:

Hi I deleted the node in DeCONZ and re-added it. I thought that if I made an error configuring it would be best to start over. The remote is stubborn though. I now have ID55 and ID56 as new devices.

I pressed every button, starting with button 1, three times: one short press, double short press and long press

The logs and the JSON:
 2021-05-21 12:48:09.421 Error: ConBee2: (ConBee2) Unknow MAJ: {'e': 'changed', 'id': '56', 'name': 'AB MOES', 'r': 'sensors', 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-04-0006'}
2021-05-21 13:01:33.297 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 1002, 'lastupdated': '2021-05-21T11:01:33.135'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:01:40.737 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 1004, 'lastupdated': '2021-05-21T11:01:40.673'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:01:49.797 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 1003, 'lastupdated': '2021-05-21T11:01:49.750'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:01:55.136 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 2002, 'lastupdated': '2021-05-21T11:01:55.086'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:01:57.641 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 2004, 'lastupdated': '2021-05-21T11:01:57.604'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:02:05.110 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 2003, 'lastupdated': '2021-05-21T11:02:05.077'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:02:10.641 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 3002, 'lastupdated': '2021-05-21T11:02:10.587'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:02:17.141 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 3004, 'lastupdated': '2021-05-21T11:02:17.086'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:02:25.226 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '55', 'r': 'sensors', 'state': {'buttonevent': 3003, 'lastupdated': '2021-05-21T11:02:25.183'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-21 13:02:28.756 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '56', 'r': 'sensors', 'state': {'buttonevent': 4002, 'lastupdated': '2021-05-21T11:02:28.717'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-04-0006'}
2021-05-21 13:02:32.125 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '56', 'r': 'sensors', 'state': {'buttonevent': 4004, 'lastupdated': '2021-05-21T11:02:32.095'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-04-0006'}
2021-05-21 13:02:46.615 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '56', 'r': 'sensors', 'state': {'buttonevent': 4003, 'lastupdated': '2021-05-21T11:02:46.605'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-04-0006'} 

Domoticz -> Custom -> Deconz -> Device data (AB MOES):

{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "ep": 4,
  "etag": "98ffed1f8fc6c77f5d8993dbfe2a6788",
  "lastseen": "2021-05-21T11:02Z",
  "manufacturername": "_TZ3000_abci1hiu",
  "mode": 1,
  "modelid": "TS0044",
  "name": "AB MOES",
  "state": {
    "buttonevent": 4003,
    "lastupdated": "2021-05-21T11:02:46.605"
  },
  "type": "ZHASwitch",
  "uniqueid": "60:a4:23:ff:fe:d2:23:2e-04-0006",
  "id": "56",
  "deviceclass": "sensors"
}

Domoticz -> Custom -> Deconz -> Device data (switch):

{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "aead66b7fa5dfe6d947ba0f550e87150",
  "lastseen": "2021-05-21T11:02Z",
  "manufacturername": "_TZ3000_abci1hiu",
  "mode": 1,
  "modelid": "TS0044",
  "name": "Switch",
  "state": {
    "buttonevent": 3003,
    "lastupdated": "2021-05-21T11:02:25.183"
  },
  "type": "ZHASwitch",
  "uniqueid": "60:a4:23:ff:fe:d2:23:2e-01-0006",
  "id": "55",
  "deviceclass": "sensors"
}

The remote shows up as a switch in Domoticz. But it has only ten levels were it should have 13 (Off and three states for every button). Could that cause the double entry?

Mimiix, 2021-05-21:
@Smanar This is the same as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3611 ?

Sjonnie2018, 2021-05-21:
Yes, it most definitely looks like the same issue. I "scanned" the thread but can't see a solution. Maybe that's because I use Domoticz instead of HA.

It looks like a problem with the plugin that does not communicate with Domoticz correctly. I will wait for confirmation by @Smanar and if so, open a thread in de Domoticz forum where he has a thread for his plugin.

Smanar, 2021-05-21:
@Mimiix nope, not the same model, and I think I have find why this one and only this one have the issue
The endpoint 0x04 don't have "basic" cluster, so the code can't reconize the device. It can use the information from other endpoint, but the code don't do it.
And this issue can happen on other device too.

And there is no issue on #3611 ?

>Maybe that's because I use Domoticz instead of HA

No don't worry, the problem is on the API, so you will have same on both

>The remote shows up as a switch in Domoticz. But it has only ten levels were it should have 13 (Off and three states for every button)

This contrariwise is probably my fault ^^, but will see later, need first to correct the API.

BTW I don't remember, are you able to compile the code on your side to test a new code ?

Smanar, 2021-05-21:
I think I have found an easy workaround
Try this curl command to delete the bad entry

`curl -X DELETE http://IP:PORT/api/KEY/sensors/56`

For IP, PORT, KEY, you can see then in domoticz logs too, else it the same you are using than fo the plugin.

This command will just delete the bad entry the device ID 56, to reconize it

"uniqueid": "60:a4:23:ff:fe:d2:23:2e-01-0006" > good one
"uniqueid": "60:a4:23:ff:fe:d2:23:2e-04-0006" > bad one

There is a weakness in API code if the code check for endpoint 0x04 before the other.
And I can't put a hack just for this device, because all tuya are almost same, It need lot of hacky code to solve the problem, and we will have "soon" a new feature in code that prevent that, so for the moment I prefer try the workaround.

And as the code seem to be able to handle the issue just if you delete the bad one ....

BTW, If I m right in domoticz you can delete sensor too ? Using custom/deconz ? If yes it will work too, but don't make mistake.

Sjonnie2018, 2021-05-21:
Hi Smanar,

I will have a go at it tomorrow morning and report back to you.

Sjonnie2018, 2021-05-22:
Hi Smanar,

I used Domoticz -> Custom -> Deconz -> 60:a4:23:ff:fe:d2:23:2e-04 -> Remove.

As a result both devices have disappeared from Domoticz and the device is no longer visible in DeCONZ. Can I safely rejoin the remote or is there something else you want me to do first. I ask specifically because I believe you expected a different result and I don't want to ruin the "crime-scene" :-)

Smanar, 2021-05-22:
Lol no you can re-include the device.
The inclusion is random, it can be correctly included on next try.

But I need to check why both are deleted .... It s not more possible to delete only 1 sensor without deleting all associated ?

Sjonnie2018, 2021-05-22:
OK, re-added the remote and it has become ID57 (Surprise!)

Logs in Domoticz (which seem promissing sofar):

 2021-05-22 12:52:39.985 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '57', 'r': 'sensors', 'state': {'buttonevent': 1002, 'lastupdated': '2021-05-22T10:52:39.943'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-22 12:52:46.271 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '57', 'r': 'sensors', 'state': {'buttonevent': 2002, 'lastupdated': '2021-05-22T10:52:46.261'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-22 12:52:50.806 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '57', 'r': 'sensors', 'state': {'buttonevent': 3002, 'lastupdated': '2021-05-22T10:52:50.779'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'}
2021-05-22 12:52:54.647 ConBee2: (ConBee2) ### WebSocket Data : {'e': 'changed', 'id': '57', 'r': 'sensors', 'state': {'buttonevent': 4002, 'lastupdated': '2021-05-22T10:52:54.612'}, 't': 'event', 'uniqueid': '60:a4:23:ff:fe:d2:23:2e-01-0006'} 

Will be performing all button sequences and report back if anything strange happens.

Thanks a bundle for your support!

Smanar, 2021-05-22:
Ha yes, perfect ^^ (we are lucky)

The inclusion is random, it s "event based" so all inclusion don't happen same, and you need a special combinaison to have the issue.

I have asked for the delete command to other devs by curiosity.

>The remote shows up as a switch in Domoticz. But it has only ten levels were it should have 13 (Off and three states for every button)

For this isue, if you still have it, you can contact me on the plugin github, domoticz forum, domoticz PM or discord, this one is more domoticz based.

Sjonnie2018, 2021-05-22:
I will open a thread on Domoticz forum. I believe the DeCONZ part is solved and I will mark the thread accordingly.

Sjonnie2018, 2021-06-19:
OK, So I got this device working but it's "eating batteries for breakfast". The battery will only last for two days max even if you only use it (say) once a day. So if you are considering buying the remote? Don't!!

I will get myself a new remote control 'cause this one is going to the dumpster :-(

Sjonnie2018, 2021-06-19:
Warn potential buyers

closed 2021-06-19 by Sjonnie2018

Smanar, 2021-06-19:
2 Days ?
Hu, IDK if it s normal, but this device haven't pool cluster, so there is not a lof thing we can do.

It use standard values like for others devices, but have you check values for reporting ? In cluster 0x0006, attribute 0x0000, press button "read config" with awaking the device in same time.

You have tied with different battery ? I already have a full set of 10 "news" empty battery on amazon.

Sjonnie2018, 2021-06-20:
Hi Smanar,

I tested the batteries before use. They all had 12 Volts or a little more. After two days the voltage was down to 0.2 Volts. I have given up on the device and bought a 4 button wall switch that is supposed to arrive any day now.

getriebesand, 2021-12-22:
I have the same battery problem. Remote works fine but after 3 days the battery is empty

Smanar, 2021-12-22:
Hello, are you able to check reporting values ?

> In cluster 0x0006, attribute 0x0000, press button "read config" with awaking the device in same time.

getriebesand, 2021-12-22:
I tried but can't find the IDs of clusters.
Can you show me where to find this? 
e.g. I have 0000 Basic and 0006 On/Off. Is that the cluster ID?
Sorry for this noob question :)

Sjonnie2018, 2021-12-22:
I hate to say but you'd better give up on this device. It really does not get any better. I can tell you that the remote is not fire proof; it melts really quickly :-)

Smanar, 2021-12-22:
>e.g. I have 0000 Basic and 0006 On/Off. Is that the cluster ID?

Yep exactly, cluster 0x0006, select the attribute 0x0000 (on the table below) then on the new pop up "read config" with making the device awake.

>the remote is not fire proof; it melts real quickly :-)

Lol, you have fire issue ?

Sjonnie2018, 2021-12-22:
> Lol, you have fire issue ?

(ROFL) No I have not. The remote did however after I set it on fire :)

getriebesand, 2021-12-25:
[image]

I have your requested reading. I hope it will help

getriebesand, 2021-12-25:
NOT_FOUND is the response on reading config in cluster 0x0006, attribute 0x0000

Smanar, 2021-12-25:
And you have used" read config" not "read" ?

Can be an explanation.
The reporting is perhaps not working/not supported/missing.

You have try to awake the device in same time ?

Edit:
I m stupid, if deconz have disabled the 3 fields, I think it mean it s not supported.

But I don't find information how work this device, if it need to be polled, or if it make natural report.
And for the moment DDF are nor working for battery switches.

getriebesand, 2022-01-21:
The dealer sent me a new one for free. I have not connected the device to deconz. Once a day I pushed a button and saw the blue light. 2 weeks or so and no empty battery.

as soon as I connected them, the next day the battery was dead

Sjonnie2018, 2022-01-21:
> The dealer sent me a new one for free. I have not connected the device to deconz. Once a day I pushed a button and saw the blue light. 2 weeks or so and no empty battery.
> 
> as soon as I connected them, the next day the battery was dead

@getriebesand : I think it is save to say the remote is utterly useless. You can still set it on fire like I did :)

Smanar, 2022-01-21:
Same manufacture name "_TZ3000_abci1hiu" ?

getriebesand, 2022-01-21:
yes, same manufacture name "_TZ3000_abci1hiu"
maybe it needs something to go to battery saving mode after connect.

caldir65, 2023-03-12:
Hi,
has the problem with the batteries been solved in the meantime? i read somewhere that the remote's firmware is to blame if it is not operated with a moes gateway.

Smanar, 2023-03-12:
Probably, can you try this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6787
If you have one of thoses TS0044 
`
"manufacturername": ["_TZ3000_ygvf9xzp", "_TZ3000_ee8nrt2l", "_TZ3000_dziaict4", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_abci1hiu", "_TZ3000_wkai4ga5"],`

caldir65, 2023-03-12:
How do I have to integrate this? My device is a _TZ3000_abci1hiu

Smanar, 2023-03-12:
Just create a DDF in the "devices" folder with the PR contain.

> 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

Use thes econd folder.
Call the file what_you_want.json and put inside this contain https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/a8384ae042620c2b4f72f372852687f236e81b7b/devices/tuya/_TZ3000_TS044_4gang_remote.json

Then just restart deconz, it will use this DDF for your device.

caldir65, 2023-03-13:
Thanks, I'll test it and give feedback later

caldir65, 2023-03-21:
Hi, it seems to work. Thanks