dresden-elektronik/deconz-rest-plugin #4900: "No button handler for: TRADFRI remote control"

TobbeJ, 2021-05-20:
## Describe the bug

I got a IKEA Trådfri round remote controll, i have used this for a long time and was working just fine.
but now all of a sudden the big middle power button doesnt react any more to single button presses and whenever i try to use it, i get this in the log:
22:50:02:274 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 59

last number increases at every button press.

the strange part is, when the addon (via home assistant) is restarted, there is a very small time window where it works, but after a few seconds it stops working and errors as described above.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
VM
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
Home Assistent deCONZ Add-on running under hassos as a vm

 - Firmware version: (26xxyy00)
0x26680700
 - deCONZ version: (2.xx.yy)
2.11.5 (via hassos addon version 6.9.0)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
ConBee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
Connected to powered usb hub
 - Is there any other USB or serial devices connected to the host system? If so: Which?
It is a vm so not relevant

## deCONZ Logs

22:49:53:189 verified group capacity: 8 and group count: 2 of LightNode 0x00158d0003247dab
22:49:53:189 0x00158d0003247dab found group 0xFFF0
22:49:53:190 0x00158d0003247dab found group 0x0003
22:49:53:968 verified group capacity: 8 and group count: 2 of LightNode 0x00158d000341e25b
22:49:53:968 0x00158d000341e25b found group 0xFFF0
22:49:53:968 0x00158d000341e25b found group 0x0003
22:49:54:678 verified group capacity: 255 and group count: 2 of LightNode 0x90fd9ffffe655fd5
22:49:54:678 0x90fd9ffffe655fd5 found group 0x0009
22:49:54:678 0x90fd9ffffe655fd5 found group 0xFFF0
22:49:55:433 verified group capacity: 255 and group count: 2 of LightNode 0xd0cf5efffe3081a1
22:49:55:434 0xd0cf5efffe3081a1 found group 0xFFF0
22:49:55:434 0xd0cf5efffe3081a1 found group 0x0009
22:49:56:173 verified group capacity: 255 and group count: 2 of LightNode 0xd0cf5efffe435fef
22:49:56:173 0xd0cf5efffe435fef found group 0xFFF0
22:49:56:173 0xd0cf5efffe435fef found group 0x0009
22:49:56:965 verified group capacity: 255 and group count: 2 of LightNode 0x90fd9ffffe58ef59
22:49:56:965 0x90fd9ffffe58ef59 found group 0xFFF0
22:49:56:965 0x90fd9ffffe58ef59 found group 0x0009
22:50:01:140 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 55
22:50:01:562 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 56
22:50:01:794 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 57
22:50:01:915 GW firmware version: 0x26680700
22:50:01:982 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 58
22:50:02:274 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 59
22:50:02:587 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 60
22:50:02:701 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 61
22:50:02:932 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 62
22:50:04:120 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 63
22:50:11:912 GW firmware version: 0x26680700
22:50:21:912 GW firmware version: 0x26680700
22:50:31:911 GW firmware version: 0x26680700
22:50:41:913 GW firmware version: 0x26680700
22:50:46:911 Current channel 15
22:50:46:917 CTRL got nwk update id 2
22:50:46:924 Device TTL 5284 s flags: 0x7
22:50:48:269 ZCL attribute report 0xD0CF5EFFFE3081A1 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:50:48:810 ZCL attribute report 0x90FD9FFFFE655FD5 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:50:51:912 GW firmware version: 0x26680700
22:50:54:010 ZCL attribute report 0xD0CF5EFFFE3081A1 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:51:01:064 ZCL attribute report 0x90FD9FFFFE655FD5 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:51:01:958 GW firmware version: 0x26680700

## Additional context

TobbeJ, 2021-05-22:
A bit more info that might be relevant.
i got two IKEA Trådfri round remotes, the one i have problem with as described above is the one with probably longest distance to nearest powered zigbee device and i know that the reception is at times not good but good enough for it to work (when it was working)

the other remote (exactly the same model) works just fine with single button presses, it is used a little differently but no problems.
this one got a lot more zigbee devices nearby, some light bulbs (4+2) some other battery powered zigbee devices (on/off button, wall mounted)

adrha, 2021-05-25:
I got exactly the same issue here...

It seems that all buttons are working, except the big power-button in the middle.

I'm running version **2.11.05** with firmware **26350500**

Tin-Bui, 2021-05-25:
I got the same issue last night while setting up both of the new remotes (same model). 
Gave up.
Then I un plugged and moved the deconz server (the pi zero) to its designated location, then the big buttons suddenly work, on both remotes.
Not sure how that works.

adrha, 2021-05-26:
> I got the same issue last night while setting up both of the new remotes (same model).
> Gave up.
> Then I un plugged and moved the deconz server (the pi zero) to its designated location, then the big buttons suddenly work, on both remotes.
> Not sure how that works.

I can confirm - powering down the setup and restarting fixes the issue.

SwoopX, 2021-05-26:
We've checked the code yesterday and there's no apparent reason for that button not to work. The button maps file has a corresponding entry ever since so the most logic root cause is that the file did not get read correctly. That a restart solved the issue supports that and you typically see any loading issues in the deconz log at the very beginning after start.

TobbeJ, 2021-05-27:
Well....
For me i have two identical remotes.
One works fine all the time with single button presses on large power button and one only works for a few seconds after deconz starts then spams above errors on every press.

i can do some more testing and include further logs

s644, 2021-05-28:
Can confirm. Restarting the deconz service worked for me, too.

TobbeJ, 2021-05-30:
Restarting works for a while, then all of a sudden it stops working.
This morning the remote was responding just fine to single press, i was able to turn on and off the lights.
then i used the long press on power button to trigger a different function, also worked then after that it is broken.

relevant part of the log:

> 10:14:00:178 [INFO] - Button 1002 - TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), action: Toggle, payload: None, zclSeq: 36
> 10:14:03:119 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: SCENES (0x0005), command: IKEA_STOP_CT (0x09), payload: 0000, zclSeq: 37
> 10:14:03:658 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: FE0000, zclSeq: 38
> 10:14:04:150 [INFO] - Button 1001 - TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: SCENES (0x0005), action: Setup 10s, payload: 02000000, zclSeq: 39
> 10:14:04:695 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: BASIC (0x0000), command: 0x00, payload: 0100, zclSeq: 40
> 10:14:06:114 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 41
> 10:14:08:845 GW firmware version: 0x26680700
> 10:14:08:852 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 42
> 10:14:09:999 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 43
> 10:14:10:272 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 44
> 10:14:10:451 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0x0008, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 45

SwoopX, 2021-05-30:
Thanks for being sticky on this. I tried yesterday to reproduce that but without any luck. However, there is a code passage that apparently is responsible for that unexpected behavior. Now we need to figure out under which conditions that can occur, your last debug output should definitifely help here.

TobbeJ, 2021-05-30:
Just to be clear, when this happens it only affects one of the two identical trådfri remotes i got.
The one that causes problems got all its buttons handled via home assistant and got a bit more distance to other zigbee devices and the controller.
power button works something like: short press=toggle on/off (this is the one that breaks), long press=reset to defaults (it controls a wled based rgbw led strip, always works even when single press stops)

The remote that tend to not have any problems got a lot more zigbee devices around it and the buttons is partly handled in phoscon.
So single press of the large power button is handled via phoscon to control a set of zigbee lights (on/off) and long pressing the power button gets handled by home assistant to turn everything off (night mode)

TobbeJ, 2021-05-30:
also i forgot to say, this is a recent problem.
i have had this setup for some time and haven't done any hardware changes, only thing that has been done is software updates.
so it was working just fine for a few months at least before it broke.

SwoopX, 2021-05-30:
So, was finally able to reproduce it. From what I can tell now, it happens if the very first command the button gets is a long press on the on/off button. Worth to mention that I was only able to force that when pulling out the battery for a few seconds and put it back in after deconz was restarted.

Can it be that is also just occurs on your end if the very first press is a long press on/off? If so, a short press on one of the arrow buttons first would be the workaround.

TobbeJ, 2021-05-30:
first press could be a long press yes.
i rarely use the arrow buttons on the sides because i got them mapped to functions that i normally don't need (dim up/down and change to different random effect on my wled strip)
i of course don't remove the battery on the remote, but could the bug also be triggered on bad reception? if so, then it is a lot more likely to happen.

today it broke after long press on power button yes, was working before.

lbschenkel, 2021-06-04:
As another data point, I confirm that I have this problem too. Remote that has been working for a long time, upgraded deCONZ, now center button no longer works.

closed 2021-06-13 by manup