dresden-elektronik/deconz-rest-plugin #4902: Downgrading to older version

ipearx, 2021-05-21:
## Describe the question or issue you are having
I'm trying to use my Conbee ii with homebridge and the hue gateway, on a Raspberry Pi with Raspian.

I'm currently using version:
Phoscon-GW: dresden elektronik deCONZ gateway v2.11.5, api v1.16.0

But I need to dowgrade to v2.10.4 after the message:
[Hue] Phoscon-GW: warning: not using recommended deCONZ gateway version 2.10.4

How do I downgrade to an older version?

Listing all versions with madison gives only the one version:
apt-cache madison deconz
deconz | 2.11.05-raspbian-buster-beta | http://phoscon.de/apt/deconz buster-beta/main armhf Packages

Thanks in advance.

## Screenshots
No screenshots needed.

## Environment
Version
2.11.05 / 09/05/2021
Firmware
26660700

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05 / 09/05/2021)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
No logs needed

## Additional context
n/a

ipearx, 2021-05-21:
I found this page, but is it OK to have an older version of deCONZ without downgrading the firmware of the conbee ii?

Smanar, 2021-05-21:
Here you have all version for Raspian http://deconz.dresden-elektronik.de/raspbian/beta/
And it s a good question, I think yes, I think firmware devs have thought at "retro-compatibility"

closed 2021-06-19 by github-actions[bot]