dresden-elektronik/deconz-rest-plugin #4903: KlikaanKlikUit smoke detector not reachable

darkogorgievski, 2021-05-21:
I have same situation as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3970

The REST API returns this:

```
9: {
"config": {
"battery": 100,
"enrolled": 0,
"on": true,
"pending": [],
"reachable": true
},
"ep": 1,
"etag": "6ae2457a69504e434978e204c13db485",
"lastseen": "2021-05-21T07:20Z",
"manufacturername": "Heiman",
"modelid": "SmokeSensor-EM",
"name": "Smoke Sensor Hall Ground Floor",
"state": {
"fire": null,
"lastupdated": "none",
"lowbattery": null,
"tampered": null
},
"swversion": "1.1.1",
"type": "ZHAFire",
"uniqueid": "00:0d:6f:00:14:cd:3a:74-01-0500"
`},
```

but the sensor is not reachable.
See also https://github.com/openhab/openhab-addons/issues/10247

## Describe the bug

## Steps to reproduce the behavior

Happens after restart of the rPi

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (openHabian)
 - Firmware version: (26660700)
 - deCONZ version: (2.10.04 / 14-3-2021)
 - Device: ( ConBee II )
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: yes, Aotec 5 USB Zwave?

## deCONZ Logs

## Additional context
I have headless install so not able to run the app mentioned in the https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3970

Mimiix, 2021-05-21:
Can yuou provide logs?

Also, you can have the gui with a X11 setup. Refer to: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

I keep this on bug report, as @SwoopX mentioned that IAS enrolement shouldve been fixed.  Swoop: can you check if that's indeed the case?

darkogorgievski, 2021-05-21:
Can you let me know where can I find the logs?

I am not sure creation of the VNC display work with openHabian, will try later

Mimiix, 2021-05-21:
As you know how to run headless, this should point in the right direction:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

dbg-error=2 , dbg-aps=2 , dbg-info=2 should be enough.

darkogorgievski, 2021-05-21:
I receive errors when I try to run the command, I read I nead to put -platform minimal in the command but not sure where:
```

Copenhabian@openHABianDevice:~ $ /usr/bin/deCONZ  dbg-error=2 , dbg-aps=2 , dbg-info=2 > debug.txt -platform minimal
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
^[[A^Copenhabian@openHABianDevice:~ $ /usr/bin/deCONZ  dbg-error=2 , dbg-aps=2 , dbg-info=2 -platform minimal
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
^Copenhabian@openHABianDevice:~ $ -platform minimal /usr/bin/deCONZ  dbg-error=2 , dbg-aps=2 , dbg-info=2
-bash: -platform: command not found
openhabian@openHABianDevice:~ $ /usr/bin/deCONZ -platform minimal  dbg-error=2 , dbg-aps=2 , dbg-info=2
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
^Copenhabian@openHABianDevice:~ $ /usr/bin/deCONZ  dbg-error=2 , dbg-aps=2 , dbg-info=2
QCoreApplication::arguments: Please instantiate the QApplication object first
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.
```

darkogorgievski, 2021-06-12:
I read in the OpenHab manuals that it is not possible to run VNC or get some other way to virtually create display so I cant get the logs. Other suggestions?

closed 2021-07-13 by github-actions[bot]