dresden-elektronik/deconz-rest-plugin #4904: Tuya Zigbee Smart Vibration Sensor

Rick-Git, 2021-05-21:
- Product name: Tuya Zigbee Smart Vibration Sensor
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Sensor

Check https://expo.tuya.com/product/753051 and https://www.aliexpress.com/item/1005002294327614.html?spm=a2g0s.9042311.0.0.77d64c4dCSZwIn

## Screenshots
[image]
### Basic
[image]

### Time

[image]

### Power Configuration

[image]

### IAS Zone
[image]
[image]

### OTAU

[image]

Rick-Git, 2021-06-12:
Not yet fixed.

Smanar, 2021-06-12:
Have you a linux machine with full OS to test code modifications ?

Rick-Git, 2021-06-14:
No, I'm sorry. 
I can try, but I have to install and set-up things...

Smanar, 2021-06-14:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_sensor_vibration https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

closed 2021-07-13 by github-actions[bot]

Rick-Git, 2021-08-16:
Hi @Smanar again :-)
So, I tried to add this one to my network via Phoscon app.
It connects, but it's not visible in Phoscon.

Via deCONZ it's visible and connected! 
What do I need to do more? Is there a testmanual, review, something??

Smanar, 2021-08-16:
Hello ^^.
So you have compiled the code, replaced the file (using the command line), re-included the device, and still nothing on phoscon ?

Can be normal, phoscon use black/whitelist, you can use this link http://phoscon.de/pwabeta in help/api information to be sure the device is not in the API.

Rick-Git, 2021-08-16:
`                    {
    "3": {
        "config": {
            "battery": 100,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "d6c7bbe610ea8d1dcb830e210455cd3b",
        "lastseen": "2021-08-16T16:42Z",
        "manufacturername": "_TYZB01_3zv6oleo",
        "modelid": "TS0210",
        "name": "Vibration 3",
        "state": {
            "lastupdated": "2021-08-16T16:42:03.095",
            "lowbattery": false,
            "tampered": false,
            "vibration": true
        },
        "type": "ZHAVibration",
        "uniqueid": "84:fd:27:ff:fe:2c:1c:15-01-0500"
    }
}`

Rick-Git, 2021-08-16:
```
                    {
    "3": {
        "config": {
            "battery": 100,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "d6c7bbe610ea8d1dcb830e210455cd3b",
        "lastseen": "2021-08-16T16:42Z",
        "manufacturername": "_TYZB01_3zv6oleo",
        "modelid": "TS0210",
        "name": "Vibration 3",
        "state": {
            "lastupdated": "2021-08-16T16:42:03.095",
            "lowbattery": false,
            "tampered": false,
            "vibration": true
        },
        "type": "ZHAVibration",
        "uniqueid": "84:fd:27:ff:fe:2c:1c:15-01-0500"
    }
}
```

Smanar, 2021-08-16:
So it works ?
I m seing battery, enrollement, vibration.

Rick-Git, 2021-08-16:
I guess. I'm installing Home Assistant on this Pi to check
But still, it's not visible in Phoscon EXCEPT the API-info.

Smanar, 2021-08-16:
Yep it s possible, phoscon use white/black list, it s just a third app like HA, it choose which one device to display or not.

Rick-Git, 2021-08-16:
It's added nicely to HA, but....

Vibration state is staying the same, all the time.
When moving the device the LED-light blinks 1 time (red), so it's doing something but the API-information/status stays the same.

When hitting the button to check the sensor the API-information is updated (with a green light) but the vibration state is still... true!

Smanar, 2021-08-16:
Have added something to reset vibration state

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

No need to re-include the device, just restart deconz.

Rick-Git, 2021-08-17:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/2609
https://www.zigbee2mqtt.io/devices/TS0210.html

Sorry, tried and no change at all. 
Searched the interwebs for some information, found above. Confirmation that the vibration doesn't reset itself, that's a shame! 
The first link build a 90-sec reset in.

Smanar, 2021-08-17:
Ok so have updated the code again.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I m using the same method than for the Xiaomi vibration sensor (that have the same issue) so no reason for the code don't work.

Rick-Git, 2021-08-17:
The status changes from TRUE to FALSE, that's working
But I don't know if it changes back when vibration is detected.... 
Have to test some more, checking logs also.

Rick-Git, 2021-08-17:
Yes, it works. I was just impatient! 

HURRAY!

Rick-Git, 2021-08-17:
I'll do another good testrun tomorrow
But I tested vibration on/off a couple of times now!

Smanar, 2021-08-17:
If you are fine with the actual code, I can make the PR ?

djhampson, 2022-01-19:
How did you add this thru phoscon? I can see it in deconz but cannot add it tot he Sensors page.

Smanar, 2022-01-19:
The PR is not validated (and I think it will be not)
Better to make a DDF file to add the device

But IDK how will react the "reset time"

saschabrockel, 2023-03-20:
Hey there. What is the state of this issue? I came from https://zigbee.blakadder.com/Tuya_TS0210.html to see that it should work with deCONZ. I have a Conbee 2 USB stick connected to my little server and I'm trying to find it in the Phoscon App as a sensor but it does not find the sensor no matter what I do. So what am I doing wrong if it is supported?

Smanar, 2023-03-20:
Hello can try this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6753
But with your manufacture name.

`"manufacturername": "_TYZB01_3zv6oleo",`

saschabrockel, 2023-03-20:
@Smanar thank you! It was already the same manufacturername. And it worked. I got it in Home Assistant and Phoscon as well now.

saschabrockel, 2023-03-22:
@Smanar did you manage to set the vibration level? I read in the manual that it should be able to be set through the Tuya gateway. Currently, it is a bit bad for me because I cannot set it and every time somebody leaves the door it recognizes vibration...

Smanar, 2023-03-22:
It's possible with the GUI, in cluster 0x0500 (IAS zone), attribute 0x0013 (missing ATM in the GUI) https://github.com/Smanar/deconz-rest-plugin/pull/136

Add to the DDF

```
                {
                    "name": "config/sensitivity",
                    "range": [0, 6],
                    "parse": {
                        "ep": 1, "cl": "0x0500", "at": "0x0013",
                        "eval": "Item.val = Attr.val"
                    },
                    "read": {
                        "ep": 1, "cl": "0x0500", "at": "0x0013"
                    },
                    "write": {
                        "ep": 1, "cl": "0x0500", "at": "0x0013", "dt": "0x20",
                        "eval": "Item.val"
                    },
                    "refresh.interval": 86400
                },
```

Values can go from 0 = hight to 6 = low.

saschabrockel, 2023-03-22:
@Smanar sorry I'm completely new to DDF. I'm wondering where to put the code snippet exactly because it already shows me a linting error in my IDE when I edit the DDF you mentioned earlier.

Smanar, 2023-03-22:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bmfw9ykl",
  "modelid": "TS0210",
  "vendor": "Tuya",
  "product": "Moes vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAPresence",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/sensitivity",
          "range": [
            0,
            6
          ],
          "parse": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013",
            "eval": "Item.val = Attr.val"
          },
          "read": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013"
          },
          "write": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013",
            "dt": "0x20",
            "eval": "Item.val"
          },
          "refresh.interval": 86400
        },
        {
          "name": "config/duration",
          "default": 1
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/presence",
          "awake": true,
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

For device _TZ3000_bmfw9ykl

saschabrockel, 2023-03-22:
Got it thanks @Smanar. But I'm still pretty helpless as I am not able to set the value for the sensitivity neither in deCONZ VNC nor even have the option to set it in Home Assistant. Where do I have to look?

Smanar, 2023-03-23:
You will probably need to remove it from HA for it will be re-added with the new field.
I think a deconz restart is enought to have the new field "config/sensitivity" visible in the API (else can re-include the device, without deleting it)
Can take a look in the  API using phoscon/help/API Information/sensors

After that IDK if HA permet to set this field, if not we will need to use direclty REST request.

saschabrockel, 2023-03-27:
@Smanar thanks I could change it using REST calls. It does unfortunately not appear in HA. Deconz does not show a field for that as well.

Smanar, 2023-03-27:
Have you tried something to update the DDF ? After having updated the DDF
- Just restart deconz (not sure this one will work)
- Re-include the device
- On the DDF editor make a hot reload
- Just re-include the device but without deleting it.

In all situation Phoscon need to be in permit join (add new sensor) for the new field will be added on the device.

Rocket200, 2023-08-20:
Can you make a DDF file for this Device 
„manufacturername":"_TZ3000_fkxmyics"
"modelid":"TS0210„
the DDF from above Show me every second a Motion/Vibration

Smanar, 2023-08-20:
From z2M this device use the same DDF

```
        fingerprint: [{modelID: 'TS0210', manufacturerName: '_TYZB01_3zv6oleo'},
            {modelID: 'TS0210', manufacturerName: '_TYZB01_j9xxahcl'},
            {modelID: 'TS0210', manufacturerName: '_TYZB01_kulduhbj'},
            {modelID: 'TS0210', manufacturerName: '_TYZB01_cc3jzhlj'},
            {modelID: 'TS0210', manufacturerName: '_TZ3000_bmfw9ykl'},
            {modelID: 'TS0210', manufacturerName: '_TYZB01_geigpsy4'},
            {modelID: 'TS0210', manufacturerName: '_TZ3000_fkxmyics'}],
        model: 'TS0210',
```

You have a vibration even without mouvement ?

Rocket200, 2023-08-20:
Looks Like every second a Vibration i not touch the device.

I not Change anything in the DDF
Change what ? Sensivity to 2/3 ?

Mimiix, 2023-08-20:
Looks like this is a complete new device.

Please open a separate GitHub issue with the correct screens.