dresden-elektronik/deconz-rest-plugin #4905: Add firmware 266b0700 to https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

Bascht74, 2021-05-22:
## Describe the question or issue you are having
The information for firmware 266b0700 ist missing at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog
Can you add the relevant information of the last stable release?

## Screenshots
[image]

## Environment
./.

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
./.

## Additional context
Thx

manup, 2021-05-22:
Done, added the missing entry was mixed up with 0x266c0700.

closed 2021-05-22 by manup

Bascht74, 2021-05-22:
@manup Thx!

According to 
[image]

...it is not a Beta-Version, but a stable one !?

Regards, 

Sebastian