dresden-elektronik/deconz-rest-plugin #4922: Strange behaviour Sonoff ZBmini

Try2Fly, 2021-05-24:
## Describe the bug
I have installed several zigbee switches/lights/sensors. I recently replaced 3 Sonoff mini (wifi versions) with 3 Sonoff ZBmini switches, with the idea to enlarge my zigbee mesh network. However the devices have a strange behaviour.

Sonoff mini 1 is closest to the ConbeeII and has no problem.
Sonoff mini 2 is a few meters further, this device acts strange as about half of the time when turning it on wirelessly, it inmediately turns off (through the Phoscon application).
Sonoff mini 3 is acting up even more strange. It is the furthest away. It is at the back of my garden with about 15/20 meters away from sonoff mini 2. This device can only be turned on wirelessly if sonoff mini 2 is already on. If I then turn of mini2, it mini3 stays on, and I can still turn it of wirelessly (within a few seconds). If I wait longer it becomes irresponsive again, and can only be turn on/off if I turn on mini2 first.

Sometimes turning on Mini2 also turns on Mini3 and vice versa. Or turning one on, turns off the other. In the Phoscon application they are not in the same group.

All devices are also each wired to a manual switch. All three of the manual switches work normally.

Can this be caused by Mini3 being to far away from the rest of the network? Mini3 is also often unavailable in Home assistant.

Or are the Sonoff ZBmini's not stable in Deconz yet?

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
 - Host system: NAS
 - Running method: (Marthoc Docker container 
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05 / 9-5-2021)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

LeoeLeoeL, 2021-05-25:
I have 6 ZB mini installed and only one of them suffers of the same issue of your mini 2.
But happens even if I switch it on manually.

Mimiix, 2021-05-25:
Can you provide some logs?

To enable logging, open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

Be sure to capture at least a few minutes worth of logs. When sharing these logs, DO NOT attempt to paste them into a channel. Instead use a pastebin service such as this or this.

When done, don't forget to disable logging. This is especually important when running off an SD card, as it will cause unnessecary writes, and might shorten the life of the media.

Try2Fly, 2021-05-25:
Ok thanks, so it could be that mini2 is faulty? I double checked, but it does not behave this way when switching manually. Also I noticed it also does this sometimes when light is on and I swicht it of, it inmediately turns it on again. So it appears that is is toggeling real fast some times.

I also took a look at the Deconz GUI, and it appears that mini3 is not connected to the network.
But if mini2 is turned on I can toggle mini3 by toggeling the group it is in, in the phoscon application (although the light itself is greyed out in phoscon)

I have another ZGmini laying around. I'll replace mini2 around, and see if the behaviour disappears, and also if a better connection can be set up between mini2 and mini3.

SwoopX, 2021-05-25:
What does on and off mean? Physically disconnected from power?

Try2Fly, 2021-05-25:
I mean the light attached to the ZBmini turns on and off. So the relay is switched. The ZBmini itself is powered whole time.

Try2Fly, 2021-05-27:
I replaced mini2 with a new device, and it appears the strange behaviour of the mini2 is gone. So this was probably a faulty device. It also looks like the mini3 is to far away for a stable connection. In the deconz GUI I can see that there is no connection. It remains strange that it did work while the faulty mini2 was in the on status (although this was also not very stable).

closed 2021-05-27 by Try2Fly