dresden-elektronik/deconz-rest-plugin #4924: Heiman Siren HS2WD won't join

Eisenhauer1987, 2021-05-25:
## Describe the bug
The siren will not join. I press 2 seconds the pairing button on the siren, then the pairing light flashes quick. I start the lightsearch in Phoscon and nothing happens. Also tried touchlink and Deconz for joining. Nothing, no logs will be displayed. Maybe I use the wrong method?

## Steps to reproduce the behavior
Tried several times the same. Search for new lights.

## Expected behavior
A light and Alarm sensor will be detected

## Screenshots

## Environment
 - Host system: Odroid n2
 - Running method: Marthoc Docker container 
 - Firmware version: 26680700
 - deCONZ version:2.12.00 / 9.5.2021
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Yes Enocean

## deCONZ Logs
15:52:15:272 GW firmware version: 0x26680700
15:52:15:821 verified group capacity: 2 and group count: 6 of LightNode 0x00158d0003884d86
15:52:15:822 0x00158d0003884d86 found group 0x0007
15:52:15:822 0x00158d0003884d86 found group 0xFFF0
15:52:15:823 0x00158d0003884d86 found group 0x0005
15:52:15:823 0x00158d0003884d86 found group 0x0013
15:52:15:823 0x00158d0003884d86 found group 0x3F54
15:52:15:823 0x00158d0003884d86 found group 0x000E
15:52:16:083 ZCL attribute report 0x7CB03EAA00AD3BD3 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
15:52:16:104 ZCL attribute report 0x588E81FFFEF37B8E for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:52:16:126 ZCL attribute report 0x588E81FFFEF37B8E for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:52:17:585 ZCL attribute report 0x00158D00031D2D0B for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:52:21:125 verified group capacity: 255 and group count: 5 of LightNode 0x588e81fffee75197
15:52:21:126 0x588e81fffee75197 found group 0xFFF0
15:52:21:126 0x588e81fffee75197 found group 0x000C
15:52:21:126 0x588e81fffee75197 found group 0x0014
15:52:21:126 0x588e81fffee75197 found group 0x0013
15:52:21:126 0x588e81fffee75197 found group 0x000E
15:52:22:336 ZCL attribute report 0x00158D000327B2EA for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:52:26:067 APS-DATA.request id: 233, addrmode: 0x03, addr: 0x60a423fffec8b386, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:52:26:097 APS-DATA.confirm id: 233, status: 0x00 SUCCESS
15:52:26:097 APS-DATA.confirm request id: 233 -> erase from queue
15:52:26:126 aps request id: 233 finished, erase from queue
15:52:26:160 APS-DATA.indication srcAddr: 0xf980, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
15:52:26:944 APS-DATA.indication srcAddr: 0xb78e, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 231, rssi: -65
15:52:27:566 APS-DATA.request id: 243, addrmode: 0x03, addr: 0x588e81fffe9c24e5, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x01 queue: 0 len: 15 tx.options 0x00
15:52:27:595 APS-DATA.confirm id: 243, status: 0x00 SUCCESS
15:52:27:596 APS-DATA.confirm request id: 243 -> erase from queue
15:52:27:646 aps request id: 243 finished, erase from queue
15:52:27:672 APS-DATA.indication srcAddr: 0x061b, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -55
15:52:32:126 Mgmt_Lqi_req zdpSeq: 120 to 0x001788011041DDCD start index 4
15:52:32:127 APS-DATA.request id: 7, addrmode: 0x03, addr: 0x001788011041ddcd, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
15:52:32:153 APS-DATA.confirm id: 7, status: 0x00 SUCCESS
15:52:32:153 APS-DATA.confirm request id: 7 -> confirmed, timeout -1511331435
15:52:32:186 APS-DATA.indication srcAddr: 0x2e8a, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -55
15:52:32:186 APS-DATA.indication request id: 7 -> finished
15:52:32:186 APS-DATA.request id: 7 erase from queue
15:52:32:187 ZDP status = 0x00 -> SUCCESS
15:52:32:187 ZDP Mgmt_Lqi_rsp zdpSeq: 120 from 0x001788011041DDCD total: 20, startIndex: 4, listCount: 2
15:52:32:187     * neighbor: 0x588E81FFFEDE690A (0x4825), LQI: 252, relation: 0x03 rxOnWHenIdle: 1
15:52:32:187     * neighbor: 0x588E81FFFEE0DEE3 (0x5489), LQI: 252, relation: 0x03 rxOnWHenIdle: 1
15:52:32:267 APS-DATA.request id: 9, addrmode: 0x03, addr: 0x00158d00031d0872, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x01 queue: 0 len: 4 tx.options 0x00
15:52:32:294 APS-DATA.confirm id: 9, status: 0x00 SUCCESS
15:52:32:295 APS-DATA.confirm request id: 9 -> erase from queue
15:52:32:336 APS-DATA.indication srcAddr: 0x7d66, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0004, lqi: 231, rssi: -65
15:52:32:337 APS-DATA.request id: 9 erase from queue
15:52:33:267 APS-DATA.indication srcAddr: 0xc196, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 223, rssi: -66
15:52:33:268 APS-DATA.request id: 17, addrmode: 0x02, addr: 0xc196, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:52:33:297 APS-DATA.confirm id: 17, status: 0x00 SUCCESS
[... 408 lines cut ...]
15:53:48:962 APS-DATA.confirm request id: 182 -> confirmed, timeout -1511254635
15:53:48:997 APS-DATA.indication srcAddr: 0x2e8a, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -55
15:53:48:998 APS-DATA.indication request id: 182 -> finished
15:53:48:998 APS-DATA.request id: 182 erase from queue
15:53:48:998 ZDP status = 0x00 -> SUCCESS
15:53:48:998 ZDP Mgmt_Lqi_rsp zdpSeq: 128 from 0x001788011041DDCD total: 20, startIndex: 14, listCount: 2
15:53:48:999     * neighbor: 0x00158D00031D0872 (0x7D66), LQI: 255, relation: 0x03 rxOnWHenIdle: 1
15:53:48:999     * neighbor: 0x588E81FFFEF0B8ED (0x24AD), LQI: 251, relation: 0x03 rxOnWHenIdle: 1
15:53:49:064 APS-DATA.indication srcAddr: 0xd8a9, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 207, rssi: -68
15:53:49:065 ZCL attribute report 0x588E81FFFED623F3 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:53:49:070 sensor 62 (lumi.sensor_motion.aq2): disable presence
15:53:51:095 APS-DATA.indication srcAddr: 0x4825, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 215, rssi: -67
15:53:51:096 ZCL attribute report 0x588E81FFFEDE690A for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:53:52:122 APS-DATA.indication srcAddr: 0x7d66, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 215, rssi: -67
15:53:52:123 ZCL attribute report 0x00158D00031D0872 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:53:52:124 APS-DATA.request id: 200, addrmode: 0x02, addr: 0x7d66, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:53:52:169 APS-DATA.confirm id: 200, status: 0x00 SUCCESS
15:53:52:170 APS-DATA.confirm request id: 200 -> erase from queue
15:53:52:211 aps request id: 200 finished, erase from queue

## Additional context
Big, good running network:
60 lights
10 switches
70 sensors

SwoopX, 2021-05-25:
I also highly doubt pushing whatever button is enough to reset the device. Therefore, I relabel this one as user question.

Smanar, 2021-05-25:
You are sure it s the zigbee model, not the Zwave ?
I m trying to find information on google but It seem there is more than 1 model, on one the Led need to go to green.

Start the light search before pressing the button.

Eisenhauer1987, 2021-05-25:
The Print on the back says zigbee 
[image]

Eisenhauer1987, 2021-05-25:
[image]

Eisenhauer1987, 2021-05-25:
[image]

Smanar, 2021-05-25:
And have found a documentation (chinese translatation) that said like you, 2s on the hole.
And you have a good network so it s not your first device.

I realy don't have idea ... Perhaps the position ? some sonoff device can't be included on other device than the coordinator.

Eisenhauer1987, 2021-05-26:
Yes crazy behavior. 
At the moment I build up a new environment for a friend. So I used this morning a fresh pi4 with docker and a new conbee Stick. But the same problem. For testing the general function of the new environment I pairing a Osram RGBW bulb and this was successful. So the environment is working fine. 

I mangen a second reaction of the device. If I press the button 10 times and the approx 5sec, the red LED ring light up for 2sec and then it does off.

In the paper documentation is written, the siren is ver 1.2. Maybe this is the problem?

Eisenhauer1987, 2021-05-26:
Hi, is there something in the logs, now?

[https://pastebin.com/AvPBDt5E](url)

Smanar, 2021-05-26:
Hu, sorry, I don't see (new) device (re)join on it ....

Eisenhauer1987, 2021-06-02:
No success until now. I order a second one, maybe the device is defect

Eisenhauer1987, 2021-06-03:
Hi second one arrived and working perfect! Slightly different print on the back 

[image]

Smanar, 2021-06-03:
>Slightly different

Hu not realy different for me,

Eisenhauer1987, 2021-06-03:
The First one have no „Ce“ and „home“ icon. Also networking says: zigbee, in the nee one: zigbee 3.0

Eisenhauer1987, 2021-06-17:
additional picture of not working siren:

[image]

closed 2021-07-16 by github-actions[bot]

opechunka, 2021-07-19:
Hi @Eisenhauer1987,

I have very similar issue with my Siren. It looks the same as the 1-st yours with a problem.  I just discovered when completely discharge (or open siren and disconnect/reconnect) the battery, then I was able to add to the network. But at some moment (some days after) it can hang again and stop to communigate with Zigbee gateway. So it's not reliable. May I ask you, where did you buy your 2-nd siren? That one with "CE" and Home icon. I've bought the bad one at Aliexpress Jelee Security Product Store in August 2020.

Eisenhauer1987, 2021-07-21:
Hi, I ordered the working Version from Aliexpress Store name: mFsmart Store. I have three from there and all are working fine.