dresden-elektronik/deconz-rest-plugin #4928: Get information device

RitaSghe, 2021-05-26:
## Describe the question or issue you are having
I wanted to get information and screenshots to open a device request for one Konke button, but I don't understand how to do.
Can you help me?

## Screenshots

## Environment
    Host system: (Raspberry Pi)
    Running method: (Raspbian)
    Firmware version: (26660700)
    deCONZ version: (2.10.04)
    Device: (ConBee II)
    Do you use an USB extension cable: (yes)
    Is there any other USB or serial devices connected to the host system? If so: usb pen

## deCONZ Logs

## Additional context

Mimiix, 2021-05-26:
Did you check this page :)
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

RitaSghe, 2021-05-26:
Yes, I had read ...
but to have this information, I I have to try the pairing?

Mimiix, 2021-05-26:
Yep.

IT needs to be paired. 

I remember that konke devices only work on specific channels. Maybe that;'s why it won't pair.

RitaSghe, 2021-05-26:
Ahhhh ok, I've been trying for the last two hours ....

closed 2021-06-25 by github-actions[bot]