dresden-elektronik/deconz-rest-plugin #4929: Xiaomi Aqara D1 1-gang (neutral wire) - model: QBKG23LM

DBaker85, 2021-05-26:
## Device

- Product name: Xiaomi Aqara D1 1-gang neutral wire (2020 version)
- Manufacturer: Lumi
- Model identifier: lumi.switch.b1nacn02
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch

## Screenshots

SwoopX, 2021-05-26:
Can you please add the screenshot of the node's clusters?

DBaker85, 2021-05-27:
@SwoopX Update the request with the requested image. If there is anything else let me know. I wonder if the non neutral type would work...

SwoopX, 2021-06-20:
@DBaker85 Sorry, this one seemed to have drowned... I prepared something, but I need some updated info to see if that's adequate or not. The screenshot of the analog input (basic) cluster for endpoint 03 is a duplicate. Can you please add a new one while the switch is turned on? I'd expect so have some power values there. Though probably negligible, please also add a screenshot from bionary output (basic) cluster of endpoint 05.

Would you be able to compile and test the code?

DBaker85, 2021-06-22:
@SwoopX 
endpoint 3:
[image]

endpoint 5: 
[image]

I should mention that the switch is not actually connected to a lamp. I am using it as a standalone control with no lights attached.
I have another laying around that I could jerry rig to a light if you need the values. 
I can also give it a shot at compiling and testing the code sure :)

SwoopX, 2021-06-27:
You find the instructions how to compile here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4842#issuecomment-855288853

However, please use the following as the first 2 commands instead

```
git clone --branch=QBKG12LM https://github.com/swoopx/deconz-rest-plugin.git QBKG12LM       # Clone repository
cd ./QBKG12LM
```

DBaker85, 2021-06-28:
Okay i'll give it try later today. Thanks for the build!

DBaker85, 2021-06-29:
@SwoopX I tried to clone this branch, but the branch `QBKG12LM ` does not exist. 
Is this the correct branch name?
if yes, did you push your branch to git?

SwoopX, 2021-06-29:
Oops, typo. Corrected the post.

DBaker85, 2021-06-29:
Ok im running into some issues (I am not very well versed in linux)
So first of all I am running the os version of home assistant on an odroid xu4 (This seems to be based on alpine linux).
I tried installing deconz-dev or even getting a package for qmake to compile the code, but I cannot find them using apk and installs fail.

SwoopX, 2021-07-06:
Ah, that makes thing rather complicated. I got no clue about HA, HASS and to get everything set up for compilation I'm afraid.

I can make the PR without having it tested, but that of course involves some uncertainty...

DBaker85, 2021-07-07:
@SwoopX well if we have no other alternative (Other people that can help) that's fine I guess. It is working as a standard on/off switch right now (which is fine for me right now).
I can test with the released version

SwoopX, 2021-07-29:
Keepalive

closed 2021-10-02 by manup