dresden-elektronik/deconz-rest-plugin #4931: Smart switch in light group

RitaSghe, 2021-05-27:
## Describe the question or issue you are having

In my house I have 6 Vimar smart switch (19592.0).
I couldn't find this switch in list supported devices and so I tried the pairing to send you the information, but the switch has been linked without problem.
The switch works fine, but is in the light group and not in the switch group.
It's a problem?

## Screenshots

[image]

## Environment
Host system: (Raspberry Pi)
Running method: (Raspbian)
Firmware version: (26660700)
deCONZ version: (2.10.04)
Device: (ConBee II)
Do you use an USB extension cable: (yes)
Is there any other USB or serial devices connected to the host system? If so: usb pen

## deCONZ Logs

## Additional context

cllatser, 2021-05-27:
The switches connected to the electric light are in the lights group. It is right. Battery-powered buttons and controls will appear in the switch group.

RitaSghe, 2021-05-28:
So, the switch are light and the button are switch ...
Funny way to subdivide the devices .....

Smanar, 2021-05-28:
Lol, no it s logic it fact, it s based on how device react on zigbee network (router or end device) or their action on group.
Battery switches send command > sensor (or switch, switch is a subtype of sensor)
Wired switches receive command > light.

closed 2021-06-27 by github-actions[bot]