dresden-elektronik/deconz-rest-plugin #4935: Tuya siren alarm pairing ok but not working

llimz, 2021-05-28:
## Describe the question or issue you are having
Hello,

The Tuya siren alarm is correctly paired. I can see 4 entities in HA (alarm, humidity and temperature sensor and 1 smart plug)

## Screenshots
[image]

## Environment
I'm using homeassistant OS with deconz add-on 2.11.05. ConBee II stick with firmware 26660700
Host system: Raspberry Pi

## deCONZ Logs
And this is what I can see with the deconz REST API related to this Tuya device:

curl -X GET -i 'http://192.168.0.29:40850/api/6A0C1670DC/sensors/'

"113": { "config": { "offset": 0, "on": true, "reachable": true }, "ep": 1, "etag": "1311979fdb0d982f4ff8cf6c379554a4", "lastseen": "2021-05-28T05:21Z", "manufacturername": "_TZE200_d0yu2xgi", "modelid": "TS0601", "name": "Temperature 113", "state": { "lastupdated": "none", "temperature": null }, "type": "ZHATemperature", "uniqueid": "84:2e:14:ff:fe:e5:60:61-01-0402" }, "114": { "config": { "offset": 0, "on": true, "reachable": true }, "ep": 1, "etag": "de798011bf6a3a410721ab8308a7d2e7", "lastseen": "2021-05-28T05:20Z", "manufacturername": "_TZE200_d0yu2xgi", "modelid": "TS0601", "name": "Humidity 114", "state": { "humidity": null, "lastupdated": "none" }, "type": "ZHAHumidity", "uniqueid": "84:2e:14:ff:fe:e5:60:61-01-0405" }, "115": { "config": { "enrolled": 0, "humiditythreshold": null, "melody": 6, "on": true, "pending": [], "preset": null, "reachable": true, "temperaturethreshold": null, "volume": 0 }, "ep": 1, "etag": "cc00c66127f638e9e064b2c30129e7e5", "lastseen": "2021-05-28T05:20Z", "manufacturername": "_TZE200_d0yu2xgi", "modelid": "TS0601", "name": "Alarm 115", "state": { "alarm": null, "lastupdated": "none" }, "type": "ZHAAlarm", "uniqueid": "84:2e:14:ff:fe:e5:60:61-01-0500" },

and for the lights:

curl -X GET -i 'http://192.168.0.29:40850/api/6A0C1670DC/lights/'

"26": {
"etag": "7e792124544f4a93db7707ef39be430c",
"hascolor": false,
"lastannounced": "2021-05-28T05:20:57Z",
"lastseen": "2021-05-28T06:05Z",
"manufacturername": "_TZE200_d0yu2xgi",
"modelid": "TS0601",
"name": "Smart plug 26",
"state": {
"alert": "none",
"reachable": true
},
"swversion": null,
"type": "Warning device",
"uniqueid": "84:2e:14:ff:fe:e5:60:61-01"
},

## Additional context
It seems that I can change the melody and volume by sending

`curl -X PUT -i 'http://192.168.0.29:40850/api/6A0C1670DC/sensors/84:2e:14:ff:fe:e5:60:61-01-0500/config' --data '{"volume":1,"melody":3}'=

and if I checked the config afterwards, the changes seems to be reflected.

But when I try to change the state to "lselect" or "select", nothing happens, no sound. I'm using the command

curl -X PUT -i 'http://192.168.0.29:40850/api/6A0C1670DC/lights/84:2e:14:ff:fe:e5:60:61-01/state' --data '{"alert":"lselect"}'

If I change the state to "blink", the light of the device is turned on and turned off after a while (maybe 1 minute), but nothing else. When the light is on if I send {"alert":"none"}, nothing is done. The light is still on.

Any idea? Do you think my device is dead or could it be a newer version which is not handled yet but the plugin?

Thanks for your help !

Mimiix, 2021-05-28:
I'm not sure what the question is?

What happens if you switch the smart plug?

llimz, 2021-05-28:
The main question is how can I do to turn the sound of the siren on?

If I switch on the smart plug, The light of the siren turns on and off after a while (maybe 1 minute). But no sound.

Smanar, 2021-05-28:
You are right, your command is good for me

`curl -H 'Content-Type: application/json' -X PUT -d '{"alert":"lselect"}' http://192.168.0.29:40850/api/6A0C1670DC/lights/26/state`

And your device seem to react to it, have you tried the 'volume' command ?

llimz, 2021-05-29:
Yes, I can change the volume by sending e.g.

`curl -X PUT -i 'http://192.168.0.29:40850/api/6A0C1670DC/sensors/84:2e:14:ff:fe:e5:60:61-01-0500/config' --data '{"volume":"50"}'`

but still no sound if I send {"alert":"lselect"}

Smanar, 2021-05-29:
Hu ?
The melody too, but seriously what can be the utility of a silent melody ...

There is something in the manual that can explain a flashing light moe without sound ?

Can you share logs with "info" when doing the request pls ?

llimz, 2021-05-29:
The manual is really poor. Only describes the way of pairing with the official tuya zigbee bridge and the smartphone app.

I tried again to change the melody by sending

`curl -X PUT -i 'http://192.168.0.29:40850/api/6A0C1670DC/sensors/84:2e:14:ff:fe:e5:60:61-01-0500/config' --data '{"volume":"50","melody":"1"}'``

and then 

`curl -X PUT -i 'http://192.168.0.29:40850/api/6A0C1670DC/lights/84:2e:14:ff:fe:e5:60:61-01/state' --data '{"alert":"lselect"}'``

no sound.

This is the few log lines related to the alert:lselect request. I put in bold what I think it is the command sent.

19:19:05:361 verified group capacity: 255 and group count: 1 of LightNode 0x000b57fffe3245c2
19:19:05:362 0x000b57fffe3245c2 found group 0xFFF0
19:19:07:743 sensor 56 (lumi.sensor_motion.aq2): disable presence
**19:19:08:174 Send Tuya request: Dp_type: 0x01, Dp_identifier 0x68, data: 01**
19:19:09:443 Set sensor check interval to 1000 milliseconds
19:19:10:174 ZCL attribute report 0x00158D0003961757 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000

Smanar, 2021-05-29:
Yeap, it's exactly this one.
The device can work with battery and usb cable ? perhaps there is a mode without the sound ?

Seriously all seem fine for me, perhaps a re-inclusion (without deleting it in deconz) can reset some device settings ...

llimz, 2021-05-29:
I tried with battery without cable, with battery+cable, without battery only cable. Everytime the same wrong behaviour.

I'm now more and more confident that it's my hardware which is the problem. I really think of buying another one. Maybe another brand. Maybe something else than Tuya. After a lot of reading it seems that we have good but also a lot of really bad devices with this brand.

But in the meantime, many thanks for your help ! I close this topic for now. If you have juste in case another idea to test, please let me know. Thanks again

closed 2021-05-29 by llimz