dresden-elektronik/deconz-rest-plugin #4936: 3 in 1 Remote Control - SUNRICHER - SR-ZG2858A

Hector47, 2021-05-28:
## Device

- Product name: SR-ZG2858A  [Product page](https://www.domadoo.fr/fr/peripheriques/5247-sunricher-telecommande-3-en-1-zigbee-30.html)

- Manufacturer: Sunricher
- Model identifier: ZG2858A
- Device type : 
  - Remote

## Screenshots

[image]
[image]

### Basic
[image]
### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics
[image]
### Other clusters that are not mentioned above :
### Power Configuration
[image]

### OTAU
[image]

Hector47, 2021-06-19:
@SwoopX Can I use the same modification as #4995

Hector47, 2021-07-12:
@SwoopX @manup is somethings missing?

Mimiix, 2021-07-20:
@Hector47 Not in my view. However, devs need to implement this and  they rather have their hands on them. So it's a waiting game unfortunately.

Hector47, 2021-08-13:
Hello

Hector47, 2021-09-07:
Hello again

Hector47, 2021-09-25:
Will I be able to add it with json Device Descriptor Files?

Hector47, 2021-11-19:
Please Reopen

Mimiix, 2021-11-19:
@Smanar

Smanar, 2021-11-19:
But new devices are not blocked, waiting for DDF ?

35gh, 2022-02-11:
Hello - would it be possible to re-open this ticket? I'll be able to support with any information if needed. Thanks!

Mimiix, 2022-02-11:
Sure

Hector47, 2022-03-17:
Hello

Smanar, 2022-03-18:
It seem battery switch can works with DDF, but I haven't succed yet.
Can make a try with this one. 

- Just drag and drop a ZHAswitch without config
- press a button, check in logs with "info" if you see an error message
- if not bind the cluster 0x0006, and remake a try (can need some time for the bind will be set, lazy device)

Hector47, 2022-04-13:
Does someone has an exemple of DDF file I can use as a base?

Smanar, 2022-04-14:
Haven't found one, and on my side I never suceed yet for battery switch.

Can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4936#issuecomment-1072592629
And share the DDF for someone can edit it.

And BTW you have an idea for the reason this device use 3 endpoint ?

Hector47, 2022-04-14:
@Smanar i thinks it because you can create up to three light groups with the remote: the 1 2 and 3 button in the last row

Hector47, 2022-05-14:
Hello - would it be possible to re-open this ticket? Thanks!

cc @Mimiix @Smanar

Mimiix, 2022-05-14:
Sure!

Smanar, 2022-05-14:
No one is able to start a DDF ? At least for some tests ?

closed 2022-06-12 by github-actions[bot]

Mimiix, 2022-07-12:
@Hector47 are you able to proceed on what @smanar asked?