dresden-elektronik/deconz-rest-plugin #4938: PARKSIDE Smart Watering Timer PSBZS A1(LIDL)

ojcalvo, 2021-05-28:
## Device

- Product name: PARKSIDE Smart Watering Timer PSBZS A1
- Manufacturer: _TZE200_htnnfasr
- Model identifier: TS0601.
- Device type :  
   - Other: Smart watering timer

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-05-28:
Missing screenshots.

Closing.

closed 2021-05-28 by Mimiix

ojcalvo, 2021-05-28:
[image]
[image]

Mimiix, 2021-05-28:
As the template suggests, please add all screenshots of all clusters.

This one remains closed. Please open another issue with all requested information.