dresden-elektronik/deconz-rest-plugin #4939: Device disconneted every  approx 20min

funbirdcas, 2021-05-28:
## Describe the bug
Nearly every 20min deConz loses connection to device. As a result, my connected logic (Home Assistant) does not always work fine.

I already tested other USB Ports on same PC, issue still there.

## Steps to reproduce the behavior

It is not possible to reproduce the error. It happens sporadically

## Expected behavior

## Screenshots

## Environment

 - Host system: (Mini PC)
 - Running method: (Windows 10 pro 21H1)
 - Firmware version: (266e0700)
 - deCONZ version: (2.12.00)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no)
 - Is there any other USB or serial devices connected to the host system? (no)

## deCONZ Logs

13:46:32:296 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
13:46:35:937 device state timeout ignored in state 3
13:46:36:124 device state timeout ignored in state 3
13:46:39:266 [INFO] - No button map for: lumi.sen_ill.mgl01, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0400, command: 0x0A, payload: 0000211181, zclSeq: 16
13:46:39:269 ZCL attribute report 0x04CF8CDF3C7B5AAE for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:46:41:584 ZCL attribute report 0xCCCC03020103866A for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:46:42:303 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
13:46:47:358 device state timeout ignored in state 3
13:46:52:293 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
13:46:55:021 device state timeout ignored in state 3
13:46:55:208 device state timeout ignored in state 3
13:46:57:939 device state timeout ignored in state 3
13:47:01:801 device state timeout ignored in state 3
13:47:01:988 device state timeout (handled)
13:47:02:019 device disconnected reason: 1, index: 0
13:47:02:081 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
13:47:02:081 device state timeout ignored in state 1
13:47:02:300 wait reconnect 15 seconds
13:47:02:308 start reconnect to network
13:47:03:295 wait reconnect 14 seconds
13:47:04:287 wait reconnect 13 seconds
13:47:05:295 wait reconnect 12 seconds
13:47:06:291 wait reconnect 11 seconds
13:47:07:298 wait reconnect 10 seconds
13:47:07:322 failed to reconnect to network try=1
13:47:08:291 wait reconnect 9 seconds
13:47:09:288 wait reconnect 8 seconds
13:47:10:296 wait reconnect 7 seconds
13:47:11:289 wait reconnect 6 seconds
13:47:12:286 wait reconnect 5 seconds
13:47:12:319 failed to reconnect to network try=2
13:47:13:288 wait reconnect 4 seconds
13:47:14:291 wait reconnect 3 seconds
13:47:15:299 wait reconnect 2 seconds
13:47:16:287 wait reconnect 1 seconds
13:47:17:329 failed to reconnect to network try=3
13:47:18:629 Device firmware version 0x266B0700 ConBee II
13:47:18:692 Device firmware version 0x266B0700 ConBee II
13:47:18:754 unlocked max nodes: 200
13:47:19:290 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
13:47:19:595 device state timeout ignored in state 3
13:47:19:782 device state timeout ignored in state 3
13:47:20:409 Device protocol version: 0x010E
13:47:21:567 CTRL ANT_CTRL 0x03
13:47:22:020 device state timeout ignored in state 3
13:47:22:286 CTRL got stack debug assert code: 0x2019, type: 0x03
13:47:22:344 Device protocol version: 0x010E
13:47:22:344 reconnect network done
13:47:23:032 device state timeout ignored in state 3
13:47:23:706 CTRL ANT_CTRL 0x03
13:47:24:065 CTRL got stack debug assert code: 0x201D, type: 0x03
13:47:24:254 ZCL attribute report 0x7CB03EAA0A04881F for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:47:24:315 ZCL attribute report 0x7CB03EAA0A04881F for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
13:47:27:041 ZCL attribute report 0x000D6FFFFE444206 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

## Additional context

funbirdcas, 2021-05-29:
I tested the following FW Version, same issue

deCONZ_ConBeeII_0x264a0700.bin.GCF 
deCONZ_ConBeeII_0x26660700.bin.GCF   
deCONZ_ConBeeII_0x266b0700.bin.GCF    
deCONZ_ConBeeII_0x266d0700.bin.GCF

Smanar, 2021-05-29:
For me it s hardware. Realy few issue on windows, the only thing to check I m seing are
- The power supply.
- The conbee firmware.

But not the same story if you are using virtual OS (from my memory HA don't work on native  windows ?)
Never make firmware upload with virtual OS.

funbirdcas, 2021-05-29:
@Smanar 

I use 2 systems.

1.
a Beelink Mini PC with Windows 10 and deConz Windows.
No virtualisation!

2.
a Raspi with HA

Smanar, 2021-05-29:
Ok, so for me you can start to contact the support, beelink is a small PC, so can try with a powered USB hub, but there is realy few issue on windows.

You have done the firmware update on the mini Pc with windows ?

funbirdcas, 2021-05-29:
@Smanar 

Unfortunately I don't have a powered USB hub.
But there is a new BIOS Firmware for my beelink und new Device Driver for Windows.
I'm  going to try them.

Yes, I always update my ConBee2 by using GCFFlasher.exe on windows.

funbirdcas, 2021-06-02:
Now some information about my further tests:

USB 3.0 Conbee2 direct connected = still the bug
USB 3.0 + extension cable + Conbee2 = still the bug
USB 3.0 + USB 2.0 powered HUB + Conbee2 = still the bug

no other USB devices were connected at the time of the test.

**Only using an USB 2.0 Port + extension cable + Conbee2 = no more error !**

vsaugey, 2021-06-14:
I have the same problem after one or two days, my device is disconnected.
I have to stop and restart the dock to make things come back to normal.
This issue exist since at least 3 months that I use conbee2.
I running the solution from a Synology host.

Regards,

Smanar, 2021-06-14:
Hello, have you tried same tests ? Avoid USB3.0 and play with USB extension ?

closed 2021-07-13 by github-actions[bot]