dresden-elektronik/deconz-rest-plugin #4942: MoesHouse/Tuya AM43 Blinds Drive not working

junalmeida, 2021-05-29:
## Describe the question or issue you are having

I'm looking for advice on a device that seems to be supported, because it is shown in web ui, although I can't find in docs.
Device is: MoesHouse/Tuya AM43 Blinds Drive

So here is my scenario: I use openhab  3.1.0M4 which is the client of your deCONZ rest API, currently version 2.12-0-beta, Openhab has the following "actions" it sends to the api: `UP`, `DOWN`, `STOP`. When I click `DOWN`, I can see the following in the logs:
```
2021-05-28 18:47:13.752 [TRACE] [internal.handler.DeconzBridgeHandler] - Sending {"on":true,"ontime":0} via http://10.0.0.99:80/api/0D234FD02F/lights/6/state
2021-05-28 18:47:13.757 [WARN ] [ernal.handler.DeconzBaseThingHandler] - Sending command DOWN to channel deconz:windowcovering:000000000000:847127fffe1aa5d901:position failed: 400 - [{"error":{"address":"/lights/6/state","description":"parameter, ontime, not available","type":6}}]
```
And nothing happens. 

If I click `UP`, I can see the following:
```
2021-05-28 19:00:23.708 [TRACE] [internal.handler.DeconzBridgeHandler] - Sending {"on":false} via http://10.0.0.99:80/api/0D234FD02F/lights/6/state
2021-05-28 19:00:23.731 [TRACE] [ernal.handler.DeconzBaseThingHandler] - Result code=200, body=[{"success":{"/lights/6/state/open":true}}]
```
And besides the successful message, and the blinds being closed/down, nothing happens.
Trying to manage the blinds through the web ui also does nothing. 

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.12.00 / 5/9/2021
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? No

Smanar, 2021-05-29:
Hello, have take a fast look on code.
The API core check for valid field in request before all the rest. So if there is a bad field, the command will be ignored.

And I have checked the API https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/ there is no "ontime" field at all.

I think the first command don't work because this field, and the second one because the covering is already in place.

junalmeida, 2021-05-29:
Sorry, on the second command, the blind was in DOWN/CLOSED position, so not in place. Unless somehow the internal state of deCONZ thinks it is in state?

Smanar, 2021-05-29:
Ha yes, and for this openhab is right

on = false is to open the covering. But it can be from the mounting  side.

Can you try the reverse request ?

`PUT, PATCH /api/<apikey>/lights/<id> with as body {"reverse":true}`

junalmeida, 2021-05-29:
This is what I got running PUT/PATCH. No movement on the device.
```
[
  {
    "success": {
      "/lights/6/reverse": true
    }
  }
]
```

Smanar, 2021-05-29:
This command don't make mouvement, but it reverse the side (for some devices)
Try to use the same command than on your first test.

junalmeida, 2021-05-29:
[image]

I just tried manually both `on: true` and `on:false`, it succeeds, but device does not move.

Smanar, 2021-05-29:
Ok so there is an issue, if even with both command and the good param nothing is moving. I need to go for today, will check tommorow.
Just to be sure can you tried too open = true/false and lift = 50

And can you share some lohs with "info" and "info_l2" when making a request with "on":true or false pls ?

junalmeida, 2021-05-30:
Logs when sending `{ on: true }`
```
08:56:25:429 poll node 84:71:27:ff:fe:1a:a5:d9-01
08:56:25:966 poll node 84:71:27:ff:fe:1a:a5:d9-01
08:56:26:048 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x01, data: 00
08:56:26:050 add task 5175 type 41 to 0x847127FFFE1AA5D9 cluster 0xEF00 req.id 71
08:56:26:403 enqueue event config/localtime for /config/
08:56:26:490 poll node 84:71:27:ff:fe:1a:a5:d9-01
08:56:27:018 poll node 84:71:27:ff:fe:1a:a5:d9-01
```

Logs when sending `{ on: false }`
```
08:57:40:403 enqueue event config/localtime for /config/
08:57:40:607 poll node 84:71:27:ff:fe:1a:a5:d9-01
08:57:40:677 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x01, data: 02
08:57:40:679 add task 5475 type 41 to 0x847127FFFE1AA5D9 cluster 0xEF00 req.id 169
08:57:41:180 poll node 84:71:27:ff:fe:1a:a5:d9-01
08:57:41:403 enqueue event config/localtime for /config/
```

Logs when sending `{ lift: 50 }`
``` 
09:00:17:405 poll node 84:71:27:ff:fe:1a:a5:d9-01
09:00:17:406 Idle timer triggered
09:00:17:617 poll node 84:71:27:ff:fe:1a:a5:d9-01
09:00:17:740 Send Tuya request: Dp_type: 0x02, Dp_identifier 0x02, data: 00000032
09:00:17:742 add task 6108 type 41 to 0x847127FFFE1AA5D9 cluster 0xEF00 req.id 179
09:00:17:903 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
09:00:18:190 poll node 84:71:27:ff:fe:1a:a5:d9-01
09:00:18:403 enqueue event config/localtime for /config/
```

Smanar, 2021-05-30:
Have take a look again on logs, I realy don't see something special for this device.
But something strange, when you send the request, the device doesn't answer by a log starting by

> Tuya debug 4 : Address

It s realy like your device was not connected, or in deep sleeping.
No timeout error about for exemple "task 5475" or "task 6108" (those values are from yours previous logs) ?

junalmeida, 2021-05-31:
I can't see any timeout errors.
As for the device being sleeping i don't know how to tell.

Smanar, 2021-05-31:
You haven't manualy command on the device ? To check from return in log from it ?

junalmeida, 2021-05-31:
When I press the buttons on the device it works. Where I can see logs? INFO_L2?

Smanar, 2021-05-31:
yep, "info" and "info_l2".
All tuya request are logged, input and output one, if you relay see nothing, try to re-include the device (without deleting it)

junalmeida, 2021-05-31:
So blinds was in `UP` state, I pressed `DOWN`,  after some secs I pressed `UP` (to make it stop, it only has two real buttons) and then I pressed `UP` again for it to come back to total opening. Here is the log:

```
13:04:44:313 binding for attribute reporting of ep: 0x01 cluster 0x0405 seems to be active
13:04:44:316 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
13:04:44:435 Websocket 10.0.0.99:55414 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2021-05-31T18:04:44.298","temperature":2211},"t":"event","uniqueid":"00:12:4b:00:23:20:bb:5a-01-0402"} (ret = 164)
13:04:44:441 Websocket 10.0.0.89:63952 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2021-05-31T18:04:44.298","temperature":2211},"t":"event","uniqueid":"00:12:4b:00:23:20:bb:5a-01-0402"} (ret = 164)
13:04:44:608 discard sensor state push for 6: state/lastupdated (already pushed)
13:04:44:659 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000102
13:04:44:663 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846594
13:04:44:665 enqueue event state/open for /lights/6
13:04:44:667 enqueue event state/on for /lights/6
13:04:44:810 Websocket 10.0.0.99:55414 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"lift":100,"on":false,"open":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 173)
13:04:44:816 Websocket 10.0.0.89:63952 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"lift":100,"on":false,"open":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 173)
13:04:44:832 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000102
13:04:44:835 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846594
13:04:45:010 discard light state push for 6: state/on (already pushed)
13:04:45:019 enqueue event config/localtime for /config/
13:04:45:035 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00006902000400000028
13:04:45:037 Tuya debug 5 : Status: 0 Transid: 0 Dp: 617 (0x02,0x69) Fn: 0 Data 545460846632
13:04:45:357 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x5E87
13:04:45:499 poll node 84:71:27:ff:fe:1a:a5:d9-01
13:04:45:904 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
13:04:45:907 Skip idle timer callback, too early: elapsed 887 msec
13:04:46:026 poll node 84:2e:14:ff:fe:65:50:5b-01
13:04:46:028 Poll light node Fireplace
13:04:46:074 Poll APS request to 0x842E14FFFE65505B cluster: 0x0006 dropped, values are fresh enough
13:04:46:903 enqueue event config/localtime for /config/
13:04:46:932 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000100
13:04:46:938 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846592
13:04:46:947 enqueue event state/open for /lights/6
13:04:46:950 enqueue event state/on for /lights/6
13:04:46:982 Websocket 10.0.0.99:55414 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 173)
13:04:46:989 Websocket 10.0.0.89:63952 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 173)
13:04:46:996 poll node 00:0d:6f:ff:fe:1f:6e:3d-01
13:04:47:004 Poll light node Ikea Light 2
13:04:47:038 discard light state push for 6: state/on (already pushed)
13:04:47:062 Poll APS request to 0x000D6FFFFE1F6E3D cluster: 0x0006 dropped, values are fresh enough
13:04:47:314 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000704000100
13:04:47:317 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1031 (0x04,0x07) Fn: 0 Data 545460846592
13:04:47:460 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000100
13:04:47:464 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846592
13:04:47:903 enqueue event config/localtime for /config/
13:04:48:325 poll node 00:0d:6f:ff:fe:22:65:c5-01
13:04:48:327 Poll light node Ikea Light 3
13:04:48:373 Poll APS request to 0x000D6FFFFE2265C5 cluster: 0x0006 dropped, values are fresh enough
13:04:48:592 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000101
13:04:48:594 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846593
13:04:48:699 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000302000400000050
13:04:48:703 Tuya debug 5 : Status: 0 Transid: 0 Dp: 515 (0x02,0x03) Fn: 0 Data 545460846672
13:04:48:705 enqueue event state/lift for /lights/6
13:04:48:708 enqueue event state/open for /lights/6
13:04:48:710 enqueue event state/bri for /lights/6
13:04:48:822 Websocket 10.0.0.99:55414 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":203,"lift":80,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 171)
13:04:48:828 Websocket 10.0.0.89:63952 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":203,"lift":80,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:1a:a5:d9-01"} (ret = 171)
13:04:48:870 discard light state push for 6: state/open (already pushed)
13:04:49:041 discard light state push for 6: state/bri (already pushed)
13:04:49:048 enqueue event config/localtime for /config/
[... cut ...]
13:04:53:910 discard light state push for 6: state/bri (already pushed)
13:04:53:917 enqueue event config/localtime for /config/
13:04:53:974 poll node 00:0d:6f:ff:fe:22:65:c5-01
13:04:53:977 Poll light node Ikea Light 3
13:04:54:122 Poll APS request to 0x000D6FFFFE2265C5 cluster: 0x0006 dropped, values are fresh enough
13:04:54:797 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000302000400000064
13:04:54:800 Tuya debug 5 : Status: 0 Transid: 0 Dp: 515 (0x02,0x03) Fn: 0 Data 545460846692
```

Smanar, 2021-05-31:
Down
```
13:04:44:659 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000102
13:04:44:663 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846594
```
??? 0x02 mean value, but don't remember 0x69
```
13:04:45:035 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00006902000400000028
13:04:45:037 Tuya debug 5 : Status: 0 Transid: 0 Dp: 617 (0x02,0x69) Fn: 0 Data 545460846632
```
position reached
```
13:04:48:699 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 0000030200040000050
13:04:48:703 Tuya debug 5 : Status: 0 Transid: 0 Dp: 515 (0x02,0x03) Fn: 0 Data 545460846672
```
Fault
```
13:04:53:135 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000704000100
13:04:53:137 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1031 (0x04,0x07) Fn: 0 Data 545460846592
```

And when you make command
`08:56:26:048 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x01, data: 00`

So same command, but realy not same data ....
Why you have so long data for up and down ? And not the same than usual.

The problem is your device is not a new one, and all previous are working, so need to understand what happen.

Smanar, 2021-05-31:
Down

> 13:04:44:659 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000102
> 13:04:44:663 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846594

UP

> 13:04:46:932 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000100
> 13:04:46:938 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846592

STOP
```
13:04:48:592 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000104000101
13:04:48:594 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1025 (0x04,0x01) Fn: 0 Data 545460846593
```

??? 0x02 mean value, but don't remember 0x69

> 13:04:45:035 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00006902000400000028
> 13:04:45:037 Tuya debug 5 : Status: 0 Transid: 0 Dp: 617 (0x02,0x69) Fn: 0 Data 545460846632

position reached

> 13:04:48:699 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000302 00 040000050
> 13:04:48:703 Tuya debug 5 : Status: 0 Transid: 0 Dp: 515 (0x02,0x03) Fn: 0 Data 545460846672

Fault

> 13:04:53:135 Tuya debug 4 : Address 0x847127FFFE1AA5D9 Payload 00000704000100
> 13:04:53:137 Tuya debug 5 : Status: 0 Transid: 0 Dp: 1031 (0x04,0x07) Fn: 0 Data 545460846592

And when you make command for UP
08:56:26:048 Send Tuya request: Dp_type: 0x04, Dp_identifier 0x01, data: 00

So the command is good. And someone have succed somes days ago with the same device https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4806#issuecomment-848910502

Can not be a "mode" problem ? You are locked in calibration mode, so commands can be disabled.

junalmeida, 2021-05-31:
I don't know what "mode" problem is supposed to be. Should I try pair it again? Should I use pwa/add lights or permit join in gui?

Smanar, 2021-06-01:
You haven't already tried to include it again ? Better to use phoscon/add new light (and no need to delete the older entry, will be easier)

And have checked on other project, there is no "calibration" command for device that use the tuya cluster.

chris199512, 2021-06-01:
test to delete the note in gui. then connect it in phoscon as sensor 2 times. then it works at me.

Smanar, 2021-06-01:
And he have put the procedure for his one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4806#issuecomment-852344148

junalmeida, 2021-06-01:
If by calibration you mean the upper and lower limit I've set it before pairing it. I'm trying to re-pair it as we speak to see if it works.

junalmeida, 2021-06-01:
@Smanar I re-paired it as you suggested, now after I send a command through the API I can see device moving. 
Good that it is working, however I'm wondering, why do we need to pair twice?

Smanar, 2021-06-01:
It s a long story, will be corrected in a new api version, but to make short, deconz is not able to reconize tuya device.
So on the first inclusion it badly reconise the device, but get data, and use them on the second inclusion.

But don't sure you are concerned by that, this issue prevent the device creation and as you can see it in phoscon was another problem

junalmeida, 2021-06-01:
Thanks for the clarification and for your help. Since it is working feel free to close the issue. 
I will follow up with openhab on the extra invalid parameter issue.

closed 2021-06-30 by github-actions[bot]