dresden-elektronik/deconz-rest-plugin #4948: Samotech temperature sensor (SM313)

RitaSghe, 2021-05-30:
## Describe the question or issue you are having
Today I connect one Samotech temperature sensor (SM313).
I have the device in the list of sensor (Multipurpose Sensor) but if I click on the device to change the name nothing's happening, it's is blocked and the temperature is integer and not decimal .....

## Screenshots
[image]

## Environment
Host system: (Raspberry Pi)
Running method: (Raspbian)
Firmware version: (26660700)
deCONZ version: (2.11.05)
Device: (ConBee II)
Do you use an USB extension cable: (yes)
Is there any other USB or serial devices connected to the host system? If so: usb pen

## deCONZ Logs

## Additional context

cooper2k4, 2021-06-04:
I have the same sensor (SAMOTECH SM313 ZigBee-Sensor für Temperatur und Luftfeuchtigkeit from Amazon: https://www.amazon.de/dp/B08GZPJYYR/) and can confirm this.

No floating point output of temperature through Phoscon App but through API. I can see the exact temperature in ioBroker.

My problem ist that the battery status is not read correctly though.  It is on 0% right from the beginning. I could take the following screenshots (couldn't do the cluster info yet).

[image]

[image]

Smanar, 2021-06-04:
Hello, can you show the attribute table of the cluster 0x0001, the table on the bottom (need to press the read button, and making the device awake to fill it)

cooper2k4, 2021-06-27:
Sorry for the delayed answer. I could only now get to the device to press the button. It's just a bit more of information. I hope that helps?

[image]

# Cluster 0000 Basic
[image]

# Cluster 0001 Power Configuration
[image]

# Cluster 0402 Temperature Measurement
[image]

Smanar, 2021-06-27:
IDK, all seem fine, the values was not updated when you have read thoses values ?
From the code, we are using the attribute 0x0021 by defaut.

By curisosity can you check if the reporting is working ?
Cluster 0x0001, attribute 0x0021, double clic on it, and "read config" on the new popup (but you will probably need to ake up the device in same time again)

Defaut values are
```
            rq.minInterval = 300;
            rq.maxInterval = 2700;
            rq.reportableChange8bit = 1;
```

cooper2k4, 2021-06-27:
Attribute 0x0021 seems to be right, but the value only gets read while pressing the button on the sensor. After manually reading the attribute it also gets displayed in Hue Essentials App.

Reading and writing config says `UNSUP_GENERAL_COMMAND`

[image]

Smanar, 2021-06-27:
Bad luck, I m sorry but I don't see how to retreive value if it don't support the command ?
Have tried to check on gooel to find information about "_TZ3000_1x1vtalf" nothing too.

Can you try to make a bind, without the reporting ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916

closed 2021-07-28 by github-actions[bot]

cooper2k4, 2021-07-29:
I wasn't able to do the bind. I'm confident with the current situation now. Humidity and temperature get read automatically and battery status if I press the button. It's just like with a Hue remote.

RitaSghe, 2021-08-31:
Do you have fix the problem of my first post?