dresden-elektronik/deconz-rest-plugin #4949: ConBee II cannot connect to deCONZ after its firmware updated

ankostis, 2021-05-30:
## Describe the bug
(originally submitted as #4894, rejected due to non-standard format)
 
Just purchased a new ConBee II dongle and managed to connect it with latest deCONZ in real linux.
But then tried to upgrade to latest firmware with `GCFFlasher_internal` from my terminal (debian buster), to resolve issues with some devices, but then it started failling to connect to deCONZ.
I tried the steps described below with and without an extension USB cable.

In all cases, whenver `deCONZ` tries  to connect, i get one line for each "connect" push-button in dmesg:
```
[ 3739.493704] cdc_acm 1-1:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
[ 3745.028227] cdc_acm 1-1:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
...
```

Occasionally it manages to connect (out of tens of efforts), while fiddling with inserting the dongle and re-connecting, switching off all wireless interfaces (airplane blutooth & wifi), and seems more probable when downgraded to oldest firmware 0x264a0700 (Apr 2019).

## Steps to reproduce the behavior
I upgraded with the commands & dmesgs logs are interspersed in time, to explain what happened:

### dmesg: plug dongle 
```
[ 2882.716937] usb 1-1: new full-speed USB device number 41 using xhci_hcd
[ 2882.866815] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 2882.866822] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2882.866826] usb 1-1: Product: ConBee II
[ 2882.866829] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 2882.866832] usb 1-1: SerialNumber: DE2436973
[ 2882.870251] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```
### console: issue upgrade command
```
$ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF 

GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
05:15:52:215 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
05:15:52:253 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2436973
05:15:52:253 pin serialno: DE2436973
05:15:52:253 search FTDI device: no.0
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436973-if00 (ConBee II)
05:15:52:259 query bootloader v1 ID after 5 ms
05:15:52:269 RX 60 bytes ASCII 
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
, 08:48:53
 after 15 ms
05:15:52:269 bootloader start after 15 ms
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
05:15:52:270 GCF_ResetDeviceDone
05:15:52:272 bootloader v1 update firmware
flashing 164377 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```
### dmesg: issue upgrade command
```
[ 2893.748615] usb 1-1: USB disconnect, device number 41
[ 2894.136850] usb 1-1: new full-speed USB device number 42 using xhci_hcd
[ 2894.287254] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 2894.287261] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2894.287264] usb 1-1: Product: ConBee II
[ 2894.287267] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 2894.287270] usb 1-1: SerialNumber: DE2436973
[ 2894.289381] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 2897.683480] usb 1-1: USB disconnect, device number 42
[ 2898.068881] usb 1-1: new full-speed USB device number 43 using xhci_hcd
[ 2898.219125] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[... 237 lines cut ...]
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000c
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

## Additional context
Flashing happened in airplane mode, in a desparate effort to stabilize the hw.

Mimiix, 2021-05-30:
Changing label. Looks more to be a user question instead of a bug report.

I suggest flashing again.

ankostis, 2021-05-30:
Thank you for the prompt reply!

Have already flashed it 5-6 times, and just flashed it once more, with latest firmware, again.

I just tried the latest deCONZ from `buster-beta`, nothing.

- What is the `acm_port_activate - usb_submit_urb(ctrl irq) failed` error in dmesg?
- Note that the console logs are very strange, after SUCCESS, it seems to continue trying to flash it, but fails to recognize it:
```
$ sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF 
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
19:59:45:746 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
19:59:45:764 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2436973
19:59:45:764 pin serialno: DE2436973
19:59:45:764 search FTDI device: no.0
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436973-if00 (ConBee II)
19:59:45:865 RX 44 bytes ASCII � after 100 ms
19:59:45:865 RX c00e01000700810069ffc0c00e02000700b20037ffc0c00e03000700830065ffc0c00e04000700800067ffc0 after 100 ms
19:59:45:865 SLIP RX frame length: 7
19:59:45:865 query bootloader v1 ID after 100 ms
19:59:46:365 close serial after 600
19:59:47:368 query bootloader v3 ID after 1603 ms
19:59:47:368 TX c081027dffc0
19:59:47:873 RX 11 bytes ASCII � after 2108 ms
19:59:47:873 RX c00e05000700810065ffc0 after 2108 ms
19:59:47:873 SLIP RX frame length: 7
19:59:47:873 query deCONZ firmware version
19:59:47:874 SLIP RX frame length: 9
deCONZ firmware version 264A0700
19:59:47:874 send watchdog reset 2 seconds
19:59:47:874 TX c00b02000c0005002602000000baffc0
19:59:47:875 set watchdog ttl status: 0x00
19:59:47:876 wait reboot: 2100 ms
19:59:51:977 query bootloader v1 ID after 2000 ms
19:59:51:989 RX 60 bytes ASCII 
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
, 08:48:53
 after 2012 ms
19:59:51:989 bootloader start after 2012 ms
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
19:59:51:990 GCF_ResetDeviceDone
19:59:51:991 bootloader v1 update firmware
flashing 164377 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
20:00:10:779 verify application is running
20:00:10:779 failed to query application
20:00:10:779 retry, elapsed 25 seconds
20:00:10:874 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:10:874 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:10:893 search FTDI device: no.0
20:00:10:893 device /dev/ttyACM0 not found
20:00:10:894 retry, elapsed 25 seconds
20:00:10:989 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:10:989 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:11:008 search FTDI device: no.0
20:00:11:008 device /dev/ttyACM0 not found
20:00:11:008 retry, elapsed 25 seconds
20:00:11:103 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:11:103 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:11:123 search FTDI device: no.0
20:00:11:123 device /dev/ttyACM0 not found
[... 817 lines cut ...]
20:00:45:441 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:45:441 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:45:460 search FTDI device: no.0
20:00:45:460 device /dev/ttyACM0 not found
20:00:45:460 retry, elapsed 59 seconds
20:00:45:565 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:45:565 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:45:585 search FTDI device: no.0
20:00:45:585 device /dev/ttyACM0 not found
20:00:45:585 retry, elapsed 59 seconds
20:00:45:680 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:45:680 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:45:700 search FTDI device: no.0
20:00:45:700 device /dev/ttyACM0 not found
20:00:45:700 retry, elapsed 59 seconds
20:00:45:795 parse commands: /usr/bin/GCFFlasher_internal.bin -t 60 -d /dev/ttyACM0 -x 3 -f deCONZ_ConBeeII_0x266b0700.bin.GCF

20:00:45:795 using firmware file: deCONZ_ConBeeII_0x266b0700.bin.GCF 
20:00:45:814 search FTDI device: no.0
20:00:45:814 device /dev/ttyACM0 not found
retry, failed
```

Smanar, 2021-05-30:
Lol, "mister automation" say to me the original issue was rather https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4945

For me, your procedure are fine, and you are more skilled than me for linux stuff.
Just by curiosity what is your hardware (The PC) ?

ankostis, 2021-05-30:
Lenovo T480 (~2019), and tested with dongle installed in 2 usb-ports, including the one for powering external devices.
Doesn't have USB-2.0 only port :-(
And i dislike Windows enough, to borrow my wife's PC :-)

Smanar, 2021-05-30:
I don't thing it s from the USB 3.0 issue, as you have tried with extension + HUB, and not realy same result.
You have checked power supply too.

I m sorry for you ^^, but even you are skilled enought and have the stuff, make a test on windows can be a good way to checks OS.

Seriously, apart from uploading firmware in random mode and crossing finger, I don't see more to do. If you have the same issue on windows I think you can contact the support.

ankostis, 2021-05-30:
Same issue on Windows :-(

ankostis, 2021-05-30:
The strangest fact is that ZHA detects its with no problem...

@Mimiix , i wouldn't classify this issue as a _User Question_,
this is a typical bug report - falling to connect is buggy behavior

Smanar, 2021-05-31:
>The strangest fact is that ZHA detects its with no problem...

I don't think the deconnexions issues are visible on third app, or if you try to send a request at same moment the gateway disconnect.

Ryssk, 2021-06-14:
Hi, i would like to report that i've got the same issue. I'm using HassOS and using Deconz integration. Updating the firmware through Phoscon isnt possible, it upgrades but then fails to connect and the device is entirely gone and unreachable.

ankostis, 2021-06-14:
There is definetely a bug in Deconz, since ZHA works just fine with upgraded dongles - please remove the _"User Question"_ flag.

Smanar, 2021-06-14:
Oups sorry bad reading.
You men the same device, with the same firmware, works on ZHA but not on deconz ?

ankostis, 2021-06-14:
Exactly!

Ryssk, 2021-06-14:
So upgrading it through the Windows 10 worked. 
This is the guide i used: https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/

Step 1 : Follow the instruction. 
Step 2 : Plug it back into your server
Step 3 : I removed the Conbee stick from my HassOS VM and then rebooted. 
Step 4 : Shut down HassOS & add the conbee device to the VM again. 
Step 5 : Now it shows up under devices, with the exact same ID. 

Even when i plugged it back in, rebooted the VM the device still wouldnt show up, only way it did was if i removed the device entirely from my VM, booted up, shut down and attached the device again. 

It's weird, but it actually worked. Hopefully for someone else it can help!

Smanar, 2021-06-14:
But it s strange it can work using the same firmware and ZHA ...

ankostis, 2021-07-06:
Maybe it's stalled because @Mimiix marked it as "UserRequest" while it clearly is not.

Mimiix, 2021-07-06:
@ankostis It's not like that :) All issues (except backlog/Confirmed bug) get this label. The reason for that is that if an issue is "stale"i get the notification and can set things in motion and act on it. I try to read all issues but sometimes they slip trough or others didn't reply. This way it makes sure to get attendence. 

In this case, I'll pm Manup and ask him to check this issue

manup, 2021-07-06:
From the logs it seems that the firmware is installed but something is messing with USB. After the update the MCU restarts and re-enumerates the USB descriptor but GCFFlasher doesn't seem to be able to reconnect – hence the retries, usually this is caused by other applications accessing the device or in case of Docker that the container fails to see the new enumeration.

Note the USB extension cable is always recommended otherwise chances are way higher to run into troubles.

Just to rule out a few things:
* Is this a native deCONZ installation or running in Docker?
* Are there any other instances of deCONZ running:
  * `$ systemctl status deconz`
  * `$ systemctl status deconz-gui`
  * Any other service like ZHA/z2m running?

voltagex, 2022-04-24:
Linux
```
~/src/gcfflasher$ ./GCFFlasher -f deCONZ_ConBeeII_0x26720700.bin.GCF  -d /dev/ttyACM0
read file success: deCONZ_ConBeeII_0x26720700.bin.GCF (164241 bytes)
flash firmware
command reset timeout
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
```

Windows
```
\GCFFlasher_V3_17> .\GCFFlasher.exe -R10 -t10 -d COM4 -x 3 -f .\deCONZ_ConBeeII_0x26720700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device COM4 (ConBee II)

2411: Error: uart reset failed, check retry
```

I can't flash the firmware on Linux or on Windows

Smanar, 2022-04-24:
From the doc there is a space between -t and 10 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Can you try just
`GCFFlasher -l`

voltagex, 2022-04-24:
Same problem,

 ./GCFFlasher -f deCONZ_ConBeeII_0x26720700.bin.GCF  -d /dev/ttyACM0 -t 10
read file success: deCONZ_ConBeeII_0x26720700.bin.GCF (164241 bytes)
flash firmware
command reset timeout
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
retry connect bootloader /dev/ttyACM0
^C
 ./GCFFlasher -l
1 devices found
DEV [0]: name: ConBee_II (DE2446370),path: /dev/ttyACM0 --> /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2446370-if00
```

Smanar, 2022-04-24:
And you are not using VM/docker ? 
Deconz is stopped before the command ?

voltagex, 2022-04-24:
Ubuntu Server 22.04, not a VM, or Windows 10 21H2, not a VM

Smanar, 2022-04-25:
Nice to know, thx a lot.

manup, 2022-10-07:
Closing this for now. As alternative to GCFFlasher 3.x ther is now also the GCFFlasher 4 in beta.
https://github.com/dresden-elektronik/gcfflasher

GCFFlasher.exe (v4.0.3-beta) can be found in https://github.com/dresden-elektronik/gcfflasher/releases/tag/v4.0.3-beta

closed 2022-10-07 by manup