dresden-elektronik/deconz-rest-plugin #4956: Symfonisk Controller - missing rotation stop buttonevent

TheNON75, 2021-05-31:
## Describe the bug

When rotating slowly, the x001/x003 button events are sent within a short time.
When rotating fast, the x001 is sent at the beginning of the rotation and x003 is sent at the end. 
However, when rotating fast to a direction, then promptly rotating to the other direction the x003 button event that is related to the first rotation is not received, missing.

## Steps to reproduce the behavior

1. Rotate continuously and fast to one direction for few seconds 
2. Without waiting or stopping, promptly start rotating to the other direction
3. The stop event for the first rotation is missing

## Expected behavior

When changing the rotation direction, the stop event shall be generated for the previous rotation

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3B
 - Running method: Marthoc Docker container
 - Firmware version: 26690700
 - deCONZ version: 2.11.05
 - Device: RaspBee II
 - Do you use an USB extension cable: N/A
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

## Additional context

TheNON75, 2021-06-30:
Unstale

Mimiix, 2021-06-30:
Sure.

I have the same controller. I'll check later if i see the same behaviour.

If that's the case it would be intresting to see what sniffer logs tell.

TheNON75, 2021-06-30:
Sounds great, thanks in advance

TheNON75, 2021-07-13:
Sorry for bothering or appearing pushy, did you manage to test the above reported behavior?

Mimiix, 2021-07-13:
I just checked. I'm on version 2.11.0. I am not sure what to look for. Either way, this is what i got:

`Event 30 fired 2:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 3003,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:37:02.743625+00:00",
    "context": {
        "id": "cc5cc2c61f27edd8122c983296d0e0cb",
        "parent_id": null,
        "user_id": null
    }
}
Event 29 fired 2:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 3001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:37:02.323031+00:00",
    "context": {
        "id": "c4411411520b081ce869be3aa2aeff4f",
        "parent_id": null,
        "user_id": null
    }
}
Event 28 fired 2:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 2001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:37:02.233542+00:00",
    "context": {
        "id": "5943021028a9c92928875f7173979db2",
        "parent_id": null,
        "user_id": null
    }
}
Event 27 fired 2:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 3001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:37:01.180646+00:00",
    "context": {
        "id": "67a7dd3b4802d46e62a8bbf28102ba85",
        "parent_id": null,
        "user_id": null
    }
}
[... 434 lines cut ...]
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:36:52.684161+00:00",
    "context": {
        "id": "b2c4382b2f961d343473b9f7260054d9",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 2:36 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 2001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T12:36:52.177536+00:00",
    "context": {
        "id": "eca4b3223b0ad02b8ce43591a8c29fb3",
        "parent_id": null,
        "user_id": null
    }
}`

Mimiix, 2021-07-13:
`Event 29 fired 3:32 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 2003,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T13:32:14.130695+00:00",
    "context": {
        "id": "d4043d3b375260d394823d3dc86daa71",
        "parent_id": null,
        "user_id": null
    }
}
Event 28 fired 3:32 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 2001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T13:32:13.859694+00:00",
    "context": {
        "id": "5919540fc514df4e90d8bea98588d846",
        "parent_id": null,
        "user_id": null
    }
}
Event 27 fired 3:32 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 3001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T13:32:13.740357+00:00",
    "context": {
        "id": "5a65f6d29b77e543744e02f7f1e81fc5",
        "parent_id": null,
        "user_id": null
    }
}
Event 26 fired 3:32 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 2001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T13:32:12.966795+00:00",
    "context": {
        "id": "074ee3d340c4c2b33d2b93753e2e3505",
        "parent_id": null,
        "user_id": null
    }
}
Event 25 fired 3:32 PM:
{
[... 417 lines cut ...]
    "context": {
        "id": "312060e1aaf4a1637b46ae43689e023e",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 3:32 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "sonoscontroller",
        "unique_id": "14:b4:57:ff:fe:65:de:fe",
        "event": 3001,
        "device_id": "02425bcdf3a945ce97f4fb9ddd275c24"
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-13T13:32:06.953695+00:00",
    "context": {
        "id": "ad340e0ea6c15808e0a105ab6557df49",
        "parent_id": null,
        "user_id": null
    }
}`

Log from 2.12.2

TheNON75, 2021-10-12:
The stop event is missing in the latest log you've shared too
[image]

TheNON75, 2022-02-15:
Hi @Mimiix ,

sorry for not coming back earlier, I could not manage so far. Sadly the issue still exists.
[image]

side info: I have experienced with other devices too, which sends start/stop event, that the stop is missing sometimes...

jzee, 2022-03-20:
Just chiming in to confirm the issue - it has behaved this way for me since the very beginning of Symfonisk encoder support. 

@TheNON75 - seeing that you use node-red, I wanted to share my workaround which made the encoder usable for me: 

[image]

Describing the flow above, the solution 
a) feeds the "turn" events (x001) each into a repeat loop - this will keep increasing or decreasing whatever I want to modify until a STOP is received
b) have the stop events (x003) produce STOP messages for the corresponding loop timer
c) (this being the workaround): have a turn event (x001) produce a STOP event for the _opposite_ turning direction