dresden-elektronik/deconz-rest-plugin #4957: Zigbee Tuya Single Phase 65A Din Rail Smart Energy Meter timer Monitor kWh Meter Wattmeter 110V 220V 50/60HzDevice name

MiAutomations, 2021-05-31:
## Device

- Product name: Zigbee Tuya Single Phase 65A Din Rail Smart Energy Meter timer Monitor kWh Meter Wattmeter 110V 220V 50/60Hz - https://s.click.aliexpress.com/e/_AOs9lj
- Manufacturer: _TZE200_byzdayie
- Model identifier:  TS0601
- Device type : 
  - Other:  Smart Energy Meter

## Screenshots
[image]

  In the Cluster Info panel press "read" button to retreive the values. Please note that at least "Manufacturer Name" and "Model Identifier" must be populated with data (therefore, must not be empty), otherwise that information will not be usable. For battery powered devices, after pressing read it is required to wake-up the device by pressing a button or any other means of interaction.
-->

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

I just can control this device as relay and I don't' see any info abou the Energy Meter functions, can you please  help me in oder to solve this ?

Thank you for your support

Smanar, 2021-06-03:
Ha nice device, and have found information about it.
Are you able to compile code to test code modification ? (Need a linux machine)

juvenalandres, 2021-06-07:
I'm interested in this as well.

Smanar, 2021-06-07:
@juvenalandres Have you  a linux machine to test code modifications ?

juvenalandres, 2021-06-07:
@Smanar  I do, but I don't have the device

Smanar, 2021-06-07:
Lol, so hard to find someone with both.

MiAutomations, 2021-07-07:
the device is not yet supported can you please keep this request open? 
Thank you

Mimiix, 2021-07-07:
Sure!

Smanar, 2021-07-07:
BTW @MiAutomations you don't have a linux machine to test code modification ?

MiAutomations, 2021-07-08:
> BTW @MiAutomations you don't have a linux machine to test code modification ?

I will try to get a machine with a linux OS in order to test this and I let you now 

Thank you

diegomontanaduarte, 2021-08-01:
@Smanar I will get this device in the next days and I'm able to test the code modification. Could you please share the steps?

Smanar, 2021-08-02:
Hello.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch energy_meter https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

You can have a different device, so you can too compare the MAC adress or manufacture name.

So (all will not work on first try)
- You will have a "light" entry for the on/off
- You will have 2 sensor a ZhaPower and a ZHAConsumption

leonardpitzu, 2021-08-09:
i have one of these on the way (should arrive in about 4 weeks). once i get it i will be able to test the integration.

diegomontanaduarte, 2021-09-01:
@Smanar I followed the steps but when I open deCONZ in GUI mode using the command line "/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > test_debug" I can just see the Coordinator node. I have already tried to put the energy meter device in pair mode many times but doesn't work.

What I'm doing wrong?

diegomontanaduarte, 2021-09-01:
So I think the "Permit Join" was disabled. I'm progressing... Please let me know what information is needed from my end. 

[image]

diegomontanaduarte, 2021-09-01:
Basic cluster info

[image]

Groups cluster info

[image]

Scenes cluster info

[image]

Tuya specific cluster info

[image]

OTAU cluster info

[image]

Smanar, 2021-09-02:
@diegomontanaduarte all seem like the PR
Can you try the code modification ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4957#issuecomment-891193893

diegomontanaduarte, 2021-09-02:
@Smanar I followed the exact steps, is there any way to check if I'm using the correct "libde_rest_plugin.so" file?

[image]

[image]

Smanar, 2021-09-02:
Ha ok sorry.

So can you check if you have no new sensors device ?

Can use http://phoscon.de/pwabeta in help / API information.
And no light device in phoscon too ?

diegomontanaduarte, 2021-09-02:
No sensor but got a light device

[image]

Smanar, 2021-09-03:
Ok so was my fault, have updated the code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then re-include the device, if it s still not working can you share some logs with "info" and "info_l2" ?

diegomontanaduarte, 2021-09-03:
Now I was able to add as a sensor, please check the Phoscon App API Information

{
    "2": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "257f7f832d151ba502a3bc64dfe7f317",
        "lastannounced": null,
        "lastseen": "2021-09-03T19:39Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Consumption 2",
        "state": {
            "consumption": 14,
            "lastupdated": "2021-09-03T19:37:03.769",
            "power": 0
        },
        "type": "ZHAConsumption",
        "uniqueid": "60:a4:23:ff:fe:10:fa:d7-01-0702"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "b07c84e87fb086232165478eeefcd52f",
        "lastannounced": null,
        "lastseen": "2021-09-03T19:39Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Power 3",
        "state": {
            "current": 1,
            "lastupdated": "2021-09-03T20:49:18.672",
            "power": 215,
            "voltage": 235
        },
        "type": "ZHAPower",
        "uniqueid": "60:a4:23:ff:fe:10:fa:d7-01-0b04"
    }
}

diegomontanaduarte, 2021-09-03:
Light API Information

{
    "1": {
        "etag": "8f59f3aab17e3e3f4a1718515684d6b3",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2021-09-03T20:57Z",
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Configuration tool 1",
        "state": {
            "reachable": true
        },
        "swversion": "0x26720700",
        "type": "Configuration tool",
        "uniqueid": "00:21:2e:ff:ff:06:cf:c5-01"
    },
    "2": {
        "etag": "fabb89be817a2db79f2c98d223c02976",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2021-09-03T20:57Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Smart plug 2",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "60:a4:23:ff:fe:10:fa:d7-01"
    }
}

Smanar, 2021-09-03:
```
"consumption": 14
"current": 1,
"power": 215,
"voltage": 235
"power": 0
```

Need to check if the device support power.

>switch (state), voltage, power, current, energy, linkquality

Arf.

diegomontanaduarte, 2021-09-03:
Logs "info"  and "info_l2"

22:04:17:685 Tuya debug Request : Address 0x60A423FFFE10FAD7, Endpoint 0x01, Command 0x01, Payload 000f13020004000008da
22:04:17:685 Tuya debug 4 : Address 0x60A423FFFE10FAD7 Payload 000f13020004000008da
22:04:17:686 Tuya debug 5 : Status: 0 Transid: 15 Dp: 531 (0x02,0x13) Fn: 0 Data 2266
22:04:18:144 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:18:144 Poll light node Smart plug 2
22:04:18:194 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:18:686 Tuya debug Request : Address 0x60A423FFFE10FAD7, Endpoint 0x01, Command 0x01, Payload 000f1402000400000916
22:04:18:687 Tuya debug 4 : Address 0x60A423FFFE10FAD7 Payload 000f1402000400000916
22:04:18:687 Tuya debug 5 : Status: 0 Transid: 15 Dp: 532 (0x02,0x14) Fn: 0 Data 2326
22:04:19:140 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:19:140 Poll light node Smart plug 2
22:04:19:193 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:19:244 Idle timer triggered
22:04:19:768 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
22:04:19:768 neigbor 0x00158d000678e5d7 is unknown child
22:04:20:099 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:20:099 Poll light node Smart plug 2
22:04:20:147 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:20:233 CTRL skip creating node 0x00158D000678E5D7 while permit join is disabled
22:04:21:069 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:21:070 Poll light node Smart plug 2
22:04:21:122 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:22:069 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:22:070 Poll light node Smart plug 2
22:04:22:122 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:22:175 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x7EBB
22:04:22:244 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
22:04:23:068 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:23:068 Poll light node Smart plug 2
22:04:23:116 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:24:061 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:24:062 Poll light node Smart plug 2
22:04:24:110 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:25:057 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:25:058 Poll light node Smart plug 2
22:04:25:106 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:26:036 Tuya debug Request : Address 0x60A423FFFE10FAD7, Endpoint 0x01, Command 0x01, Payload 000f11020004000005c3
22:04:26:036 Tuya debug 4 : Address 0x60A423FFFE10FAD7 Payload 000f11020004000005c3
22:04:26:036 Tuya debug 5 : Status: 0 Transid: 15 Dp: 529 (0x02,0x11) Fn: 0 Data 1475
22:04:26:053 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:26:054 Poll light node Smart plug 2
22:04:26:102 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:26:967 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
22:04:26:967 neigbor 0x00158d000678e5d7 is unknown child
22:04:27:048 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:27:048 Poll light node Smart plug 2
22:04:27:096 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:27:433 CTRL skip creating node 0x00158D000678E5D7 while permit join is disabled
22:04:28:042 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:28:042 Poll light node Smart plug 2
22:04:28:093 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:29:040 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:29:041 Poll light node Smart plug 2
22:04:29:093 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:29:372 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x7EBB
22:04:29:948 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:29:949 Poll light node Smart plug 2
22:04:29:997 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:30:893 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:30:893 Poll light node Smart plug 2
22:04:30:944 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:31:819 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:31:819 Poll light node Smart plug 2
22:04:31:871 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:04:32:244 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
22:04:32:744 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:04:32:744 Poll light node Smart plug 2
22:04:32:794 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
[... 317 lines cut ...]
22:05:45:683 Tuya debug Request : Address 0x60A423FFFE10FAD7, Endpoint 0x01, Command 0x01, Payload 000f0101000101
22:05:45:683 Tuya debug 4 : Address 0x60A423FFFE10FAD7 Payload 000f0101000101
22:05:45:683 Tuya debug 5 : Status: 0 Transid: 15 Dp: 257 (0x01,0x01) Fn: 0 Data 1
22:05:46:167 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
22:05:46:167 neigbor 0x00158d000678e5d7 is unknown child
22:05:46:196 poll node 60:a4:23:ff:fe:10:fa:d7-01
22:05:46:197 Poll light node Smart plug 2
22:05:46:245 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
[... cut ...]
22:05:48:168 Poll light node Smart plug 2
22:05:48:216 Poll APS request to 0x60A423FFFE10FAD7 cluster: 0x0006 dropped, cluster doesn't exist
22:05:48:575 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x7EBB
22:05:48:688 Tuya debug Request : Address 0x60A423FFFE10FAD7, Endpoint 0x01, Command 0x01, Payload 000f14020004000008f4
22:05:48:688 Tuya debug 4 : Address 0x60A423FFFE10FAD7 Payload 000f14020004000008f4
22:05:48:689 Tuya debug 5 : Status: 0 Transid: 15 Dp: 532 (0x02,0x14) Fn: 0 Data 2292

Smanar, 2021-09-03:
Ha nope it's my fault, you have state/power on both sensor, with one working and not the other

Smanar, 2021-09-03:
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Have updated the code, I think it can work if you just restart deconz, else you probably need to delete and re-incude the device.

The code will remove the "power" on the ZHAConsumption device.

diegomontanaduarte, 2021-09-03:
zigbee2mqtt

[image]

Smanar, 2021-09-03:
Ha current is more presice for them, wait

diegomontanaduarte, 2021-09-03:
I think a precise value for current will be better.

Cannot test on/off right now because if it works I'll throw the entire house down :)

diegomontanaduarte, 2021-09-04:
Using the latest updated code I got the following API Information. Please let me know if any information is needed from my end.

{
    "2": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "dcee4d3a044bb5ee0f3eb210133ef90b",
        "lastannounced": null,
        "lastseen": "2021-09-03T22:15Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Consumption 2",
        "state": {
            "consumption": 15,
            "lastupdated": "2021-09-03T22:16:02.478"
        },
        "type": "ZHAConsumption",
        "uniqueid": "60:a4:23:ff:fe:10:fa:d7-01-0702"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "33654e33c844e4d14f041726835b6ca1",
        "lastannounced": null,
        "lastseen": "2021-09-03T22:15Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Power 3",
        "state": {
            "current": 1,
            "lastupdated": "2021-09-04T15:17:19.762",
            "power": 190,
            "voltage": 233
        },
        "type": "ZHAPower",
        "uniqueid": "60:a4:23:ff:fe:10:fa:d7-01-0b04"
    }
}

Smanar, 2021-09-04:
So all is fine now ?
Except the current value that is rounded ATM, I have asked, I m sorry but on the actual code, we can use only integer, so the result will be less precise.

diegomontanaduarte, 2021-09-05:
@Smanar I'm just missing the Link Quality (signal strength) metric

Smanar, 2021-09-05:
The deconz API don't expose this parameter.

You need to use the application to see it.

diegomontanaduarte, 2021-09-05:
OK, thanks for clarifying! How can I test the device on/off function? Could you please provide the API call? Or is there an easier way?

Mimiix, 2021-09-05:
@diegomontanaduarte that's something to ask on the forums.

diegomontanaduarte, 2021-09-06:
@Mimiix sorry but I'm not a developer and very new to this deCONZ REST-API debugging thing. I'm trying to help adding support to this specific device and change this communication channel to the forums doesn't make any sense to me.

Mimiix, 2021-09-06:
Git is only for debugging indeed :) Asking for general stuff (that applies to deconz intself) is on the forums.

Either way, you can find your answer here: https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Smanar, 2021-09-06:
But this device is WIP, we are not at the good place ?
We have just tested the sensor, now we are testing the on/off.

Try to use the "state" switch you have on HA, or use the device direclty in phoscon (the light one).

If it don't work can you share the log you have when doing the command ? (with "info" and "info_l2" ?

Mimiix, 2021-09-06:
@Smanar it is. But it's not the place to ask how to use the rest api :)

diegomontanaduarte, 2021-09-06:
@Smanar I was able to test the device on/off function using the API and it's working fine!

Smanar, 2021-09-07:
Ok thx So I make the PR.

leonardpitzu, 2021-09-18:
I have received the meter and succesfully installed and set it up. There is an issue though - the sensors (power and energy) are "not reachable" thus they do not appear in Home Assistant. The associated "Light" is reachable though. Anything i need to do/set-up/tweak? Is this a bug?

Smanar, 2021-09-18:
Ha good question, have checked the code again and all seem fine.
Can you share the device JSON ?(can have it here http://phoscon.de/pwabeta on help/api information)

leonardpitzu, 2021-09-18:
the "sensors"
`    "60": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "69499d33a4ccc76dc22787c6a0ce521e",
        "lastannounced": "2021-09-18T09:11:51Z",
        "lastseen": "2021-09-18T19:16Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Mains Energy",
        "state": {
            "consumption": 8,
            "lastupdated": "2021-09-18T18:14:23.961"
        },
        "type": "ZHAConsumption",
        "uniqueid": "b4:e3:f9:ff:fe:c9:fc:7a-01-0702"
    },
    "61": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "7d4b856ff09519d894041191b8aa2350",
        "lastannounced": "2021-09-18T09:11:51Z",
        "lastseen": "2021-09-18T09:11Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Mains Power",
        "state": {
            "current": 2,
            "lastupdated": "2021-09-18T19:30:45.766",
            "power": 538,
            "voltage": 246
        },
        "type": "ZHAPower",
        "uniqueid": "b4:e3:f9:ff:fe:c9:fc:7a-01-0b04"
    }
}
`

the "light"
`    "36": {
        "etag": "d2482bcd92e064907bdd26cf046cc8b9",
        "hascolor": false,
        "lastannounced": "2021-09-18T09:11:51Z",
        "lastseen": "2021-09-18T19:18Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Mains Switch",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "b4:e3:f9:ff:fe:c9:fc:7a-01"
    }
`

Smanar, 2021-09-19:
Yep and "lastseen" is working ...

Can you take a look in log if you have not this message, for the device b4:e3:f9:ff:fe:c9:fc:7a

>[WARNING] - Missing cluster in sensor fingerprint: b4:e3:f9:ff:fe:c9:fc:7a ...............

leonardpitzu, 2021-09-19:
Yes, i can compile and test anything needed.

Smanar, 2021-09-19:
Ok, have updated the code again to force the update.
I think it will work, but not sure the "hack" will be accepted.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Smanar, 2021-09-19:
For information, I have found the problem, but need more time to fix it.
I think I will need to add a new "tuya hack" somewhere ....

leonardpitzu, 2021-09-20:
I tried the "hack" and it basically works. If you manage to implement the "real tuya hack" i'll test that one too.

Smanar, 2021-09-20:
>looks like it's refreshing very very slow

But to go to state "unreachable", the device need to be 24h without update.
And on the code I m updating RX at every report from it.

Can you check on the device Json "lastseen" and "last updated" value ?

leonardpitzu, 2021-09-20:
energy
`"lastseen": "2021-09-20T17:28Z",
"lastupdated": "2021-09-20T17:25:28.317"`

power
`"lastseen": "2021-09-20T17:54Z",
"lastupdated": "2021-09-20T17:55:37.475"`

as you can seethe energy sensor is updated very slow and in HA it looks like this.
[image]

leonardpitzu, 2021-09-22:
i can't change the log level right now as i am not at home. i can check the API though and yes, reachable is false:

`"60": {
        "config": {
            "on": true,
            "reachable": false
        },
        "ep": 1,
        "etag": "8a5496d6aaa38c9a3cf90efd3c8c85bf",
        "lastannounced": "2021-09-18T09:11:51Z",
        "lastseen": "2021-09-22T07:32Z",
        "manufacturername": "_TZE200_byzdayie",
        "modelid": "TS0601",
        "name": "Mains Energy",
        "state": {
            "consumption": 30,
            "lastupdated": "2021-09-22T06:46:52.183"
        },
        "type": "ZHAConsumption",
        "uniqueid": "b4:e3:f9:ff:fe:c9:fc:7a-01-0702"
    }`

but this only happens to one of the sensors, the power sensor works as expected...

leonardpitzu, 2021-09-23:
this is how it looks in HA. I'll come back with some logs later today.
[image]

leonardpitzu, 2021-09-23:
the log: [log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7218328/log.txt)

Smanar, 2021-09-23:
So yes, there is something strange

```
New value and good notification
16:37:36:979 Tuya debug 5 : Status: 3 Transid: 233 Dp: 529 (0x02,0x11) Fn: 0 Data 4400
16:37:36:982 Websocket 192.168.0.10:37056 send message: {"e":"changed","id":"60","r":"sensors","state":{"consumption":44,"lastupdated":"2021-09-23T13:37:36.980"},"t":"event","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"} (ret = 163)
16:37:36:984 Websocket 192.168.0.10:37056 send message: {"attr":{"id":"60","lastannounced":"2021-09-18T09:11:51Z","lastseen":"2021-09-23T13:37Z","manufacturername":"_TZE200_byzdayie","modelid":"TS0601","name":"Mains Energy","productid":"Smart Energy Meter","swversion":null,"type":"ZHAConsumption","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"},"e":"changed","id":"60","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"} (ret = 383)
16:37:36:986 Websocket 192.168.0.10:37056 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"60","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"} (ret = 134)
```

Useless notification
`
16:40:09:932 Websocket 192.168.0.10:37056 send message: {"attr":{"id":"60","lastannounced":"2021-09-18T09:11:51Z","lastseen":"2021-09-23T13:40Z","manufacturername":"_TZE200_byzdayie","modelid":"TS0601","name":"Mains Energy","productid":"Smart Energy Meter","swversion":null,"type":"ZHAConsumption","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"},"e":"changed","id":"60","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"} (ret = 383)`

But it was set to false ...
`16:46:23:014 Websocket 192.168.0.10:37056 send message: {"config":{"on":true,"reachable":false},"e":"changed","id":"60","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:c9:fc:7a-01-0702"} (ret = 135)`

leonardpitzu, 2021-09-23:
`de_web_plugin.cpp: In member function 'void DeRestPluginPrivate::checkSensorNodeReachable(Sensor*, const deCONZ::NodeEvent*)':
de_web_plugin.cpp:3916:25: error: request for member 'fingerPrint' in 'sensor', which is of pointer type 'Sensor*' (maybe you meant to use '->' ?)`

leonardpitzu, 2021-09-24:
I fixed the bug and checked the code. It compiled ok and runs as expected. One thing i noticed is the scale - i think it's wrong as i see about 200W constant usage and just 0.00x kWh after about 2 hours. This is too low, the energy is scaled wrong (or something else is wrong).

Smanar, 2021-09-24:
200W during 1 hour = 200W/H = 0.2 Kw/H

So you need * 10 ?

leonardpitzu, 2021-09-24:
by * 100

Smanar, 2021-09-24:
Last version pushed, just need to replace the file, no need to re-include the device.

[11 more replies]

closed 2023-01-18 by manup