dresden-elektronik/deconz-rest-plugin #4958: LIDL PARKSIDE Smart Watering Timer PSBZS A1

ojcalvo, 2021-05-31:
Hello, this smart watering dont refresh its state and dont work well.
Thanks

## Device

Product name: PARKSIDE Smart Watering Timer PSBZS A1
    Manufacturer: _TZE200_htnnfasr
    Model identifier: TS0601.
    Device type :
        Other: Smart watering timer

## Screenshots
[image]

[image]

[image]

### Other clusters that are not mentioned above

ottelo9, 2021-06-01:
Do you mean this one:
https://www.lidl.de/de/parkside-smarter-bewaesserungscomputer-zigbee-smart-home/p375570

stefangries, 2021-06-01:
I'm having issues with the device, too. It does pair via Phoscon, but switching it manually is neither reported in the UI, nor does controlling the device via the UI work.

ojcalvo, 2021-06-01:
Yes, is exactly like this. In phoscon It's integrated likea light, but dont refresh the state and dont show any sensor of battery level, etc.
Thanks.

stefangries, 2021-06-01:
As some screenshots are hard to read, I added some in higher resolution:

[image]
[image]
[image]

Btw, for me, these buttons are working:
[image]

BUT, clicking the light button in phoscon isn't doing anything:
[image]

When controlling the device via deconz, however, phoscon is able to show the correct state of the device (on/off).

When controlling it via the Home Assistant integration, it's a little bit different. You can switch the device on/off correctly, but the switch in Home Assistant stays at the old state. When switching it via deconz, the correct state is indicated in Home Assistant.

Smanar, 2021-06-01:
Hello, can you show the device JSON pls ?

stefangries, 2021-06-01:
Can you give me a hint, where to find it? I couldn't find the information, where to get the JSON [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support).

flausen, 2021-06-01:
I am not able to find the parkside watercomputer via phoscon. Any hints?

ojcalvo, 2021-06-01:
> I am not able to find the parkside watercomputer via phoscon. Any hints?

It shows like a light in phoscon.

stefangries, 2021-06-01:
> Hello, can you show the device JSON pls ?

Is it this, what you are asking for?

```
{
   "etag":"9e8cc3af3dd044b93eafa4d7c1cafe6f",
   "hascolor":false,
   "lastannounced":null,
   "lastseen":"2021-06-01T17:16Z",
   "manufacturername":"_TZE200_htnnfasr",
   "modelid":"TS0601",
   "name":"On/Off light 14",
   "state":{
      "alert":"none",
      "on":false,
      "reachable":true
   },
   "swversion":null,
   "type":"On/Off light",
   "uniqueid":"84:71:27:ff:fe:d3:4a:81-01"
}
```

osochor, 2021-06-01:
The auto-off timer is the best thing about this device! 😁 unfortunately you can’t change the timer yet. I hope the device support comes very soon. 😀 I have 3 devices in use. And with this price and the auto-off function, there are sure to be even more!

flausen, 2021-06-01:
By now I am able to open the waterflow, but the switch turns out immediately, so that I can’t Stop the waterflow using a device. Only manual at the parkside computer

Smanar, 2021-06-01:
For thoses one that can act on the device using deconz, but not working using phoscon (or API)

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch watering https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Still need to check for timer.

Smanar, 2021-06-01:
Ok, so the timer, nothing about this device on other project, so you need to search ^^. On the description they say the device can work with philips HUE gateway, so I don't think they use the tuya cluster (good new)

With the GUI :
Take a look on cluster 0x0006 (on/off), on attribute 0x4002 "off wait time".
Can try to the command "on with timed off"

stefangries, 2021-06-01:
So this should just turn the device on for 90 seconds, right?
[image]
It does turn on, but goes off after 60 seconds.

Btw, this device is very new, offical release was yesterday. I was not successful in pairing it with the hue bridge.

ottelo9, 2021-06-02:
I ordered, too. I‘m hoping that deconz will include this new device :)

Smanar, 2021-06-02:
Just by curiosity, someone can take a look in logs for line starting by 
>Tuya debug ....................

Smanar, 2021-06-02:
And neither if using the device manualy ?

stefangries, 2021-06-02:
Regardless of whether I switched the device via deconz or manually, I could not find "tuya" in the log. I use the Home Assistant addon, and have set all debug flags to 1.

Smanar, 2021-06-02:
Ok, so we are blocked for the moment.
No tuya request.

Have you tried too with the "off wait time" in the deconz GUI ?

Someone have tried to compile the new code to check if the device is working in the API with it ?

stefangries, 2021-06-02:
> Ok, so we are blocked for the moment.
> No tuya request.

Just to make sure: which debug level should be enabled to see the tuya request?
[image]

> Have you tried too with the "off wait time" in the deconz GUI ?

You mean this?
[image]
Didn't changed anything.

Attributes are read only:
[image]

> Someone have tried to compile the new code to check if the device is working in the API with it ?

No, I'm only running the docker container and don't have a full linux machine, sorry :-/

osochor, 2021-06-02:
The off timer starts from the new one, if On is sent again. My workaround For a watering duration of 5 min (295s), the sequence is repeated 5 times (on, wait 59s)

Smanar, 2021-06-02:
To have tuya log you just need the flag "info", so you was fine

>You mean this?

Yep, but with changing the 0 value (yeah I know, not logic, but tuya is magic ...)

>No, I'm only running the docker container and don't have a full linux machine, sorry :-/

 It s possible too with docker https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-852157541

stefangries, 2021-06-02:
> Yep, but with changing the 0 value (yeah I know, not logic, but tuya is magic ...)

Now I tried this command, which instantly turns on the device. But it turns off automatically after 60s, too.
[image]

> It s possible too with docker [#4841 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-852157541)

Building containers is quite a stressing task for my NUC J4005 :-D
Do I have to use tuya_trv_12 branch as well?
I was able to build it, but currently having trouble to pair the device.

stefangries, 2021-06-02:
Now I was able to pair it. Here are some debug outputs (filtered for "tuya"):
```
21:49:03:602 Tuya debug 1 : Time sync request
21:49:03:605 Send Tuya command 0x24, data: 000060b7fccf60b7fccf
21:49:32:532 Send Tuya request: Dp_type: 0x01, Dp_identifier 0x68, data: 01
21:49:35:178 Send Tuya request: Dp_type: 0x01, Dp_identifier 0x68, data: 00
```

When trying to toggle in phoscon:
[image]

The server replies:
`[{"error":{"address":"/lights/1/state","description":"parameter, transitiontime, not available","type":6}}]`

Smanar, 2021-06-03:
Have just updated the code for the "transitiontime" error.

But for the tuya log it s strange, this command is send when using "alert" in the request ....

ottelo9, 2021-06-03:
To avoid the 60s auto off timer I use two scripts in Home Assistant to switch on (for 1h) and switch off the valve. See the pics:

[image]
[image]

fluppie, 2021-06-04:
The Parkside watering meter is a Tuya device: https://zigbeealliance.org/zigbee_products/smarter-water-computer/

ottelo9, 2021-06-06:
Is there any chance that this device will be implemented in near future? Otherwise I send it back and buy WOOX R7060.

Smanar, 2021-06-06:
Can try to compile the actual code to test https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958#issuecomment-852388777

The only thing that will not work if to set the timer, need to use a workaround for the moment.

ottelo9, 2021-06-06:
Mhh thats bad. The timer is very annoying. I cant compile it because I use the deconz addon in my Home Assistant. So ... I will send it back

Smanar, 2021-06-06:
You can with docker https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-852157541
But more complicated.

stefangries, 2021-06-06:
@Smanar 

I have now tested the current code from your repo and the "watering" branch. Here is what I noticed:

- The device now powers on correctly via phoscon and home assistant. The status also changes back and forth. Thanks for your fix!
- If the device turns off automatically after 60 seconds, the status in phoscon / home assistant does not update and is therefore incorrectly still on "on".
- The 60s timer cannot be changed yet.

Can I somehow help with tests / logs to fix this?

Smanar, 2021-06-06:
Can you check the reporting for cluster 0x0006 attribute 0x0000 ?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2196#issuecomment-695843587

stefangries, 2021-06-06:
First step:
[image]

Second step:
[image]

Smanar, 2021-06-06:
Ok so something have not worked.
Defaut values are 1 / 300

But I don't see why it don't work on code.
Can you first take a look if you see some error about binding or reporting for this device  ? (deconz monitor devices and can try to remake the procedure regulary)

Else can you try to change values yourself (and make bind), but as the device is battery powered, you probably need to awake it in same time.

It s possible the device don't support regular bind/reporting and use the tuya cluster instead.

stefangries, 2021-06-06:
@Smanar Unfortunately, I don't quite understand what I need to do in deconz to provide you with needed information. Can you describe in more detail what to do?

Smanar, 2021-06-06:
Just enable log in deconz / help with "info" and "error" and check for error message about "report" or "bind"

Then try to set yourself the values 1 / 300, some procedure than before but press "write config" instead (you have already done the bind.

You will probably have error message when trying to set them.

stefangries, 2021-06-06:
This time, when pressing "read config" i did receive 1 / 300:

[image]

Writing works, too...
[image]

Sorry for not testing it directly multiple times on the last try....

stefangries, 2021-06-06:
I just paired the device new to get all logs (filtered for 847127FFFED34A81). Maybe there is something, that might help?

```
18:03:17:344     * neighbor: 0x847127FFFED34A81 (0xC577), LQI: 252, relation: 0x01 rxOnWHenIdle: 0
18:03:23:771 void zmController::nodeKeyPressed(deCONZ::zmNode*, int) node 0x847127fffed34a81 deleted
18:03:25:789 APS-DATA.request id: 130, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:03:30:778 APS-DATA.request id: 165, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:03:35:778 APS-DATA.request id: 209, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:03:40:778 APS-DATA.request id: 250, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:03:45:777 APS-DATA.request id: 30, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:03:50:778 APS-DATA.request id: 62, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0034, ep: 0x00 -> 0x00 queue: 1 len: 10 tx.options 0x00
18:04:12:345 unknown node 0x847127FFFED34A81 (0x9827), lqi: 255
18:04:12:345 APS-DATA.indication from unknown node 0x847127FFFED34A81
18:04:12:345 ZDP device announce: 0x847127FFFED34A81, 0x9827, 0x80
18:04:12:346 new node - ext: 0x847127fffed34a81, nwk: 0x9827
18:04:29:088 Node_Descriptor_req zdpSeq: 28 to 0x847127FFFED34A81
18:04:29:476 ZDP Node_Descriptor_rsp 0x847127FFFED34A81 - 0x9827
18:04:29:638 ZDP active ep response for 0x847127fffed34a81
18:04:30:820 Node_Descriptor_req zdpSeq: 29 to 0x847127FFFED34A81
18:04:30:907 ZDP Node_Descriptor_rsp 0x847127FFFED34A81 - 0x9827
18:04:31:169 ZDP active ep response for 0x847127fffed34a81
18:04:32:464 Active_EP_req zdpSeq: 30 to 0x847127FFFED34A81
18:04:32:706 ZDP active ep response for 0x847127fffed34a81
18:04:34:153 Simple_Descr_req zdpSeq: 31 to 0x847127FFFED34A81 endpoint 1
18:04:34:199 ZDP Simple_Descriptor_rsp 0x847127FFFED34A81 - 0x9827
18:04:34:277 APS-DATA.request id: 22, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
18:04:34:778 APS-DATA.request id: 26, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x01 queue: 0 len: 6 tx.options 0x00
18:04:35:998 read binding table for 0x847127fffed34a81 (0x9827) 
18:04:36:949 APS-DATA.request id: 32, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
18:04:36:949 Mgmt_Bind_req id: 32 to 0x847127FFFED34A81 send
18:04:36:990 MgmtBind_rsp id: 32 0x847127fffed34a81 seq: 36, status 0x00 
18:05:01:278 APS-DATA.request id: 152, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:05:06:528 APS-DATA.request id: 197, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x01 queue: 0 len: 4 tx.options 0x00
18:05:07:278 APS-DATA.request id: 201, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x01 queue: 2 len: 4 tx.options 0x00
18:05:13:277 APS-DATA.request id: 148, addrmode: 0x03, addr: 0x847127fffed34a81, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
18:05:13:278 Mgmt_Bind_req id: 148 to 0x847127FFFED34A81 send
18:05:16:004 verified group capacity: 255 and group count: 1 of LightNode 0x847127fffed34a81
18:05:16:004 0x847127fffed34a81 found group 0xFFF0
18:05:18:428 MgmtBind_rsp id: 148 0x847127fffed34a81 seq: 13, status 0x00 
18:05:21:043 verified group capacity: 255 and group count: 1 of LightNode 0x847127fffed34a81
18:05:21:043 0x847127fffed34a81 found group 0xFFF0
18:05:28:529 verified group capacity: 255 and group count: 1 of LightNode 0x847127fffed34a81
18:05:28:529 0x847127fffed34a81 found group 0xFFF0
18:05:31:519 verified group capacity: 255 and group count: 1 of LightNode 0x847127fffed34a81
18:05:31:520 0x847127fffed34a81 found group 0xFFF0
```

osochor, 2021-06-08:
Another workaround who owns a Lidl Gateway. You can change the auto-off time within the Lidl Gateway. After delete the device in the lidl App, the new value will be stored in the device. Now you can use the device with the new Auto-off time within deconz. 😀

ottelo9, 2021-06-09:
@MrSolo570 This will not interrupt the flow because I retrigger the device every 50s (60s = timer off).

osochor, 2021-06-09:
> @MrSolo570 This will not interrupt the flow because I retrigger the device every 50s (60s = timer off).

59s works also. 😁

ottelo9, 2021-06-09:
> > @MrSolo570 This will not interrupt the flow because I retrigger the device every 50s (60s = timer off).
> 
> 59s works also. 😁

No in my case it's not working! Because of a delay of ~5s between send command to device and device reacts (Home Assistant on RPI3B with deconz).

Smanar, 2021-06-09:
>Is there some test I can do here to help this device getting supported?

For the moment, we need someone sniff the zigbee request to found the Tuya request to use.
And for the missing return when using it manualy, it s same, I think the device use a tuya request, as all seem fine in the code.

flausen, 2021-06-09:
Tomorrow, maybe the day after I receive the Lidl gateway. Should I sniff s.th.?

Smanar, 2021-06-09:
If you can yes (you need some stuff too), else there is someone on z2m project that is trying too.

profigriller, 2021-06-09:
I am using deconz 2.12 with Openhab 3.1.0-M3 . In Phoscon device shows as on/off device. I also use Openhab to connect to Deconz using Deconz binding. In Openhab I can set the timer AND the state. I was able to set the time to 2 min. But the device does not report the state change to deconz. Not a really big problem. Nice, cheap and simple. Lets see how long it lasts.

Smanar, 2021-06-09:
How you can set the timer ? Using workaround ?

Can hope for a tuya log when the timer is finished. Nothing in logs at same moment ?

forast3r, 2021-06-09:
> > Is there some test I can do here to help this device getting supported?
> 
> For the moment, we need someone sniff the zigbee request to found the Tuya request to use.
> And for the missing return when using it manualy, it s same, I think the device use a tuya request, as all seem fine in the code.

Sadly I don't have the lidl gateway or the sniffer hardware avaliable (yet) but will keep reading this thread. Thanks for clarifying!

superpedro, 2021-06-10:
Hi,

I´m trying to integrate the lidl watering into phoscon via conbee II but its not found as a light. Pushing the button till the red led of network sign is blinking nothing happened.

Is there a trick to do?

Thanks for help

Smanar, 2021-06-10:
Yep @superpedro > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958#issuecomment-852388777

MrSolo570, 2021-06-11:
> I´m trying to integrate the lidl watering into phoscon via conbee II but its not found as a light. Pushing the button till the red led of network sign is blinking nothing happened.
> 
> Is there a trick to do?

Mine just arrived yesterday, here is what I did:
put in the incuded batteries. startet pairing a light in phoscon (searching time 10 minutes). Long pressed the button on the lidl-device until the connection lamp (the left lamp) started blinking. and a few seconds later it appeared as on/off light at the bottom in phoscon.

I use Home Assistant and have the state problem too. The device is shown as off all the time, even if you turn it on. After 60s it turns off automatically but you can work-around this with an automation (as shown above)

flausen, 2021-06-12:
I received my gateway a couple of minutes ago. How can I help you with some data?

Smanar, 2021-06-12:
We need someone sniff the device, but for the moment nothing usefull > https://github.com/Koenkk/zigbee2mqtt/issues/7695#issuecomment-858399023

Main-pinguin, 2021-06-14:
I have timely limited access to a Lidl / Tuya gateway. On last weekend I had the posibility to play around a bit..
There is an interval timer which is set by default to 1 minute.. This could be extended with hours and minutes... This value is the watering period even if you press the button or switch via gateway... In addition there 2 or 3 "Watering Programms" which allow to watering with  time and intervals based on date and time... If one of this values changed (added) the valve will work "autonomus" ...  Will try to sniff zigbee traffic when I have access to Lidl Gateway next time

Smanar, 2021-06-14:
But It s possible to change the duration using the app ?

flausen, 2021-06-14:
Yes it is. Next to the „Einfach ein/aus“. What you entered there will be set as the default timer.

p-rintz, 2021-06-14:
I'd be willing to buy the device for a dev, if it helps.

(or give you the 25€ so you can buy it yourself)

Smanar, 2021-06-14:
It s more complicated, we need someone with the device, a sniffer and the original gateway. But don't worry, we're not alone on this device ^^, it seems a lot of people are interested.

One day or another someone will succed to sniff the good request and share it to other.

p-rintz, 2021-06-14:
I dont have a problem with buying the gateway as well if you need it. (or donating the money, as said)

Buying the Parkside Device twice and the gateway once is still cheaper than other automatic watering solutions (+theyre zigbee which is nice)

[202 more replies]

closed 2022-11-18 by github-actions[bot]