dresden-elektronik/deconz-rest-plugin #4959: Raspberry POE and POE+ HAT compatibility

TheNON75, 2021-05-31:
## Describe the question or issue you are having

Hello,
Hearing about interference issues with USB3, I am curious about if using the original/official Raspberry POE HAT (or the new, not yet on sale POE+ HAT) is causing any interference or other issues.
I am using a RaspBee II but could be interesting in relation to ConBee II too.

Thank you

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian or Marthoc Docker container
 - Firmware version: 26690700
 - deCONZ version: 2.11.05 / 5/9/2021
 - Device: RaspBee II
 - Do you use an USB extension cable: N/A
 - Is there any other USB or serial devices connected to the host system? Planned: Raspberry Pi PoE+ HAT

## deCONZ Logs

## Additional context

JornDL, 2021-06-12:
Hi

I have been using the Raspberry Pi 4B from the start, with both the POE hat but also with the new POE+ hat and with the ConBee II, without any problems.

TheNON75, 2021-06-17:
Thanks @JornDL 
I hope it is valid to raspbee ii as well. :)
Are you using your stick directly attached or is it sitting at the end of a lengthy cable?

closed 2021-07-16 by github-actions[bot]