dresden-elektronik/deconz-rest-plugin #4963: Pulling out the stick during operation

jabred, 2021-06-01:
## Describe the question or issue you are having

After using deCONZ for three weeks, im try to change the sensibility of a PIR-Sensor in the deCONZ-GUI. After I couldn't find the window for the parameters, I made the mistake and pulled out the stick. After a second i plugged the Stick back in. After that action, i lost all connections to the ZB-Devices. Only the Conbee Stick remains visible. I try a reboot - but that dosen`t help. I try many things - at least the system behaves weird. 

I lost the Connection to the Devices in OpenHAB. But the Devices remains in the Phoscon-App. With no succsess, i try to reestablish the Connection to OpenHAB.  But after a short period of time, i was unable to establish even a connection to the Conbee-Stick. 

At this Point i discovered; when i stop de deconz-daemon and starting only the Gui, i was able to connect to the Stick. It was also possible to rebuild the connection to OpenHAB. But when the daemon was runnig, the connection to the Stick was not possible. 

I purge deCONZ and reinstallt it. But this behaviour remains. After i start the deconz-deamon the connection to the stick is not possible. At this point everyting is fine - but im confused. Why is it possible to using deconz when the daemon is stopped? Is there a second service? I remember - after the fresh installation, it was necessary to start the daemon after a systemreboot.  I want to make sure that everything is stable before i connect other devices. 

## Screenshots

## Environment

 - Host system: Intel NUC
 - Running method: Ubuntu 20.04.02 LTS
 - Firmware version: 26660700
 - deCONZ version: 2.11.05 / 9.5.2021
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Smanar, 2021-06-02:
I m not sure to understand what you have done but deconz can be run on both way.
- With the GUI on desktop environement.
- Headless.

You CAN'T use both mode in same time. Something to test to be sure
```
sudo systemctl status deconz
sudo systemctl status deconz-gui
```

You need to have an error at one of those command, or you can close both

```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
```

And restart the one you need.

If you have only 1 deconz instance, and the conbee still not connected, can you share some logs pls ?

jabred, 2021-06-02:
Thanks for the answer. 

I think i can see my mistake. I was always trying to start de Headless-daemon bevor i start the GUI: 

`sudo service deconz start`
`sudo deCONZ`

But at this time the Headless-daemon already taken the Stick - so it was impossible to claim the Stick over the GUI. Two Services - one Stick - that cant work. 

Am i correct in assuming that if i start deconz via the GUI - it must always be open in order to function? And if this assumption is correct, does the headless mode and the GUI mode use the same configuration? So is it possible to configure deCONZ via the GUI and for the normal daily use switch to the Headless-daemon?

Smanar, 2021-06-03:
>Am i correct in assuming that if i start deconz via the GUI - it must always be open in order to function?

Yes

>And if this assumption is correct, does the headless mode and the GUI mode use the same configuration?

Yes

>So is it possible to configure deCONZ via the GUI and for the normal daily use switch to the Headless-daemon?

Yes
Somes users do that when they use a raspberry 1, to weak for GUI during the day. But on recent hardware, the desktop mode don't stress the hardware, the CPU usage difference is not realy significative.

closed 2021-07-03 by github-actions[bot]