dresden-elektronik/deconz-rest-plugin #4964: Unable to turn KNYCKLAN receiver off

frassefraze, 2021-06-02:
## Describe the bug
In Phoscon (and there by HomeAssistant) I'm unable to turn the "KNYCKLAN receiver" off. If I use the buttons in the deCONZ inteface, under the "On/Off Cluster" it works just fine. 

I did notice that someone reported a similair issue for another device, were the on/off-buttons also work in DeCONZ but not in Phoscon (I can however turn mine on in Phoscon, but not off)
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958

## Steps to reproduce the behavior
Go to Phoscon, turn the device on. Then turn the device off (wich doesnt work).
Go to deCONZ and turn the device off. 

## Expected behavior
Being able to control both the on and off prcedure from Phoscon (and HomeAssistant)

## Screenshots

## Environment
 - deCONZ version: 2.11.05 / 09/05/2021
 - Firmware version: 26390500
 - Device: ConBee I
 - Host system:  PC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26390500
 - deCONZ version: 2.11.05 / 09/05/2021
 - Device: ConBee I 
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? Yes If so: ZW090 Z-Stick Gen5 EU

## deCONZ Logs

## Additional context
Just let me know what additional information is needed.

AutoHass8, 2021-06-05:
Noticed the same, got two of those. 
Created light groups in Phoscon (one for each Knycklan), then turning them on/off with the individual light groups works.

frassefraze, 2021-06-05:
Amazing @AutoHass8, thanks a lot for the workaround, creating a group works for me as well.

ebaauw, 2021-06-05:
Looks like the receiver doesn’t support _Off with Effect_.  What’s the device type?  Can you please list the resource create for the receiver and attach a screenshot of the node with endpoints and cluster (like when requesting support for a new device)?

frassefraze, 2021-06-05:
Pasting screenshots below, just let me know if there is any addional details needed. 

- Product name: TYPE E1842 - KNYCKLAN
- Manufacturer: IKEA of Sweden
- Model identifier: KNYCKLAN receiver
- Device type :  
  - Other: 

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ebaauw, 2021-06-05:
Above commit should send a regular _Off_.  Cannot test it myself, though, since I don't have the device.

Looks like the device is also a water (leak) sensor (the _IAS Zone_ cluster on endpoint 0x02), that should be exposed as a ZHAWater `/sensors` resource.  It seems deceptively straightforward to do so (just whitelist the device as sensor), but ironing out the details requires a close develop/test cycle.  I'm not going to do that remotely.