dresden-elektronik/deconz-rest-plugin #4968: long start time of deconz process and old devices in zll.db

bruesche, 2021-06-02:
## Describe the question or issue you are having
After I restart the deconz process it take a very long time with 100% CPU usage. When DECONZ is started and IOBROKER adapter can connect the IOBROKER adapter shows old devices of DECONZ which are not shown in PHOSCON App. A look in the zll.db shows that the old devices are there (table "Devices"). Info: my zll.db has a volume of over 1GB.

Need help that the deconz process start quicker.
Is the only way to delete the old devices to manipulate the zll.db bis sqlitebrowser?
Is it normal that the DB is over 1GB tall?

Can anyone give me some advice?

Thx for help.

## Screenshots

## Environment

Host system: Raspberry Pi
Running method: Raspbian Buster
Firmware version: 266B0700
deCONZ version: 2.12.00
Device: ConBee II
Do you use an USB extension cable: no
Is there any other USB or serial devices connected to the host system? Yes - USB Stick

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

senilio, 2021-06-08:
If you're running Raspbian from an SD card, I'd suspect it's going bad. We've had a couple of these cases before, and this far i believe SD is the common denominator. 

Your DB size is far from normal. My zll.db is around 250Kb in a medium sized network.

I'm not aware of a fix, but I would probably start with migrating to a new media. If you search the closed issues in ths repo, you will find a few dodgy things to try that "might" fix the sqlite DB.

Rick-Jongbloed, 2021-06-18:
Have a look at my post here I had a similar issue and I fixed the issue by expoeting the sql file, fixing it and inporting it in a new database. I have had no issues since I fixed it. Your issue might be something else, but it's worth a try. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3972

bruesche, 2021-06-29:
Yes. It seems to me that it was the same error. I exported the sql file and imported it in a new DB. Now DB size is 275kb. Everything works well.

closed 2021-06-29 by bruesche