dresden-elektronik/deconz-rest-plugin #4969: Symfonisk Controller - duplicated zcl.cmd (button events)

TheNON75, 2021-06-02:
## Describe the bug

Dear All,

I've installed the latest version yesterday and since then, the IKEA Symfonisk Controller sends every button event twice.

thanks in advance for looking after the issue

regards

## Steps to reproduce the behavior

1. press any button or rotate the knob

## Expected behavior

The button events should not be duplicated

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3B
 - Running method: Raspbian host with Marthoc Docker container
 - Firmware version: 26690700
 - deCONZ version: 2.11.05 / 5/9/2021
 - Device: RaspBee II
 - Do you use an USB extension cable: (yes / no) -- N/A
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

```
13:11:19:714 ZCL attribute report 0x00124B00088B8527 for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000 ,
13:11:32:699 Device TTL 6960 s flags: 0x6,
13:11:21:638 saved node state in 0 ms,
13:11:21:638 sync() in 0 ms,
13:11:21:706 dev /dev/ttyAMA0,
13:11:21:707 GW firmware version: 0x26690700,
13:11:23:292 [INFO] - Button 1002 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Toggle, payload: None, zclSeq: 78,
13:11:23:503 Warning duplicated zcl.cmd: 0x02, cluster: 0x0006 with zcl.seq: 78 for IKEA of Sweden / SYMFONISK Sound Controller,
13:11:23:503 [INFO] - Button 1002 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Toggle, payload: None, zclSeq: 78,
13:11:24:617 Skip idle timer callback, too early: elapsed 932 msec,
13:11:29:239 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0x000F, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0108070000, zclSeq: 8,
13:11:29:756 [INFO] - No button map for: 902010/25, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0702, command: 0x0A, payload: 00042A020000000025E7480F000000, zclSeq: 41,
13:11:29:756 ZCL attribute report 0x00124B00088B8527 for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000 ,
13:11:31:685 dev /dev/ttyAMA0,
13:11:31:685 GW firmware version: 0x26690700,
13:11:32:674 Current channel 25,
13:11:32:687 CTRL got nwk update id 5
```

## Additional context

TheNON75, 2021-06-02:
"press any button" is meant for single, double or triple press button event

mat4444, 2021-06-06:
I second this issue. I have observed it as well. Any action on the button is reported twice within a few milliseconds.

17:30:50:500 [INFO] - Button 1004 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008). action: Step Up, payload: 000100000000, zclSeq: 51
17:03:50:601 Warning duplicated zcl.cmd: 0x02, cluster: 0x008 with zcl.seq: 51 for IKEA of Sweden / SYMFONISK Sound Controller
17:30:50:602 [INFO] - Button 1004 - SYMFONISK Sound Controller, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008). action: Step Up, payload: 000100000000, zclSeq: 51

TheNON75, 2021-06-17:
Thanks @mat4444 for confirming the same, this lets me assume we are not alone with this

mat4444, 2021-06-17:
I'm using it in Home Assistant and many other users discovered this issue

Botr16, 2021-06-27:
Same here, with home assistant

TheNON75, 2021-06-30:
It is not related to homeassistant, I am using it with the official docker image

Smanar, 2021-07-01:
Hello, someone here know a little c++ and have a linux machine to make a test ?
There is a code part removed since some versions, and I have no answer about it.

mat4444, 2021-07-10:
> It is not related to homeassistant, I am using it with the official docker image
Yes, I agree. But many users noticed this issue through HomeAssistant.

TheNON75, 2021-07-10:
I guess this only cause many users are using it with homeassistant, simply unavoidable due to the law of large numbers 😄

Mimiix, 2021-07-13:
@Smanar i have a second Conbee here and a Symfonisk. If i give you a VM on my PC with ubuntu/deconz with SSH access, can you  figure it out?

Smanar, 2021-07-13:
It s just a one line edition.
If you want to compile the code to test

```
sudo apt install deconz-dev
git clone --branch symfonisk_test https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Don't need to re-include the device.

Smanar, 2021-08-01:
No one can make test ?

TheNON75, 2021-08-01:
Sorry, I cannot. Running in docker.
Unless the code works in docker, does it?

Smanar, 2021-08-02:
Yes it s possible to compile on docker, but not realy easy, and less on a production machine https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

manup, 2021-11-30:
Above PR adds duplicate check for all IKEA switches I currently don't have the Symfonisk here for testing, but for all the other Ikea switches it works, hopefully here too :)

closed 2021-12-05 by manup