dresden-elektronik/deconz-rest-plugin #4971: Swann Motion Sensor SWO-MOS1PA

6c4491, 2021-06-03:
## Device

- Product name: Swann One Motion Sensor
- Manufacturer: SwannONe
- Model identifier: SWO-MOS1PA
- Device type: 
  - Sensor

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Device Temperature
[image]

### Power Configuration
[image]

SwoopX, 2021-06-15:
IAS Zone cluster is missing and the attributes for the power configuration cluster have not been read. Can you please update the screenshots accordingly?

6c4491, 2021-06-15:
Updated, hopefully that's right

SwoopX, 2021-06-16:
Yes, it is, thanks but similarly unexpected.

Would you be able to compile a test version of the plugin?

6c4491, 2021-06-16:
I can definitely try work out how to do so

6c4491, 2021-06-19:
I think I have the test version set up correctly, what do I need to do next?

SwoopX, 2021-06-20:
If that's a version containing my code changes, just reset and re-pair the device. Afterwards, the device should work.

closed 2021-07-02 by manup