dresden-elektronik/deconz-rest-plugin #4972: effect: colorloop no longer works in most recent version

D34DC3N73R, 2021-06-04:
## Describe the bug

All hue bulbs no longer work with `effect: colorloop` (in HA scripts, dev tools, & directly in HA UI). This can be solved by reverting deconz to amd64-2.11.01

## Steps to reproduce the behavior

  1. update to marthoc/deconz:latest
  2. try `effect: colorloop` on an RGB hue bulb in HA
  3. the bulb turns on but does not colorloop
  4. revert deconz to marthoc/deconz:amd64-2.11.01
  5. `effect: colorloop` now works in HA

## Expected behavior

`effect: colorloop` works in most recent version of deconz

## Screenshots

## Environment

 - Host system: PC Ubuntu Server 20.04 - Docker version 20.10.7
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.12.00 / 5/9/2021 - does not work (2.11.01 does work)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes - HUSBZB-1 but only used for Zwave

## deCONZ Logs

Please let me know if logs are needed and I'll do my best to get them uploaded.

## Additional context

The bulbs in use are Philips Hue White and Color Ambiance A19 562785

D34DC3N73R, 2021-06-27:
This is still an issue for me. I hope someone else is able to confirm.

DanteFireX, 2021-07-13:
I can confirm the issue. ``effect: colorloop`` does not work in HA with the current stable deconz. Could be related to #4846.

D34DC3N73R, 2021-08-04:
I did some more testing and the latest working version is 2.11.2. every 2.11 and 2.12 release after that fails.

Mimiix, 2021-08-06:
I have 2.12.2 and i have it working on my hue spots and ledstrips.

DanteFireX, 2021-08-06:
Switched to 2.12.2 and it is not working for me unfortunately.

D34DC3N73R, 2021-08-06:
neither 2.12.02 or 2.12.03 (:latest) work for me either. Reverting to 2.11.02 fixes the issue.

Mimiix, 2021-08-06:
Important to mention is that you get an error 951 (busy bridge).

DanteFireX, 2021-08-06:
I dug through the code a bit and found the following:

The busy bridge error is returned in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/3eb73ec61d9a6c7da5660d30a82de1522a03405c/rest_lights.cpp#L1053 if ``addTaskSetColorLoop`` returns ``false`` 20 lines above.

This function was changed in #4846 (which was introduced in 2.11.3) to return false if ``!task.lightNode->supportsColorLoop()``.

My guess is that `supportsColorLoop` somehow does not work for hue lights.

ebaauw, 2021-08-06:
In v2.13.3 setting `"effect": "colorloop"` works fine for my Hue LCT003.

Smanar, 2021-08-06:
>My guess is that supportsColorLoop somehow does not work for hue lights.

Possible, can you show the "config/colorcapabilities" for your device ?

D34DC3N73R, 2021-08-06:
This is on 2.12.03
[image]

and this is on 2.11.02 where colorloop is working. 
[image]

DanteFireX, 2021-08-07:
If the value of ``colorcapabilities`` in your snippet is the one used in ``supportsColorLoop``, then 0 should mean that colorloop is not supported, as the function should then return false here: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/36891a17568b783aedb00e1512a63bed7b7e51ff/light_node.cpp#L245

ebaauw, 2021-08-07:
You could double-check in the GUI, but I’ve never seen a Hue _Extended Color Light_ that doesn’t expose _Color Capabilities_.  It looks like deCONZ hasn’t setup the resource correctly.  `ctmax` and `ctmin` are also at their default values, afaik Hue lights expose these as well.

Best reset and re-pair the light.  Don’t delete the resource, or the light will get a new id after re-pairing.

D34DC3N73R, 2021-08-07:
This isn't just a single light. It's the same for every LCA003 bulb in the house.

ebaauw, 2021-08-07:
Do they all show colorcapabilities 0, ctmax 65279, and ctmin 0?

Can you post a screenshot of the GUI showing the _Color Control_ cluster in _Cluster Info Panel_ after reading the attributes?  Particularly interested in 0x400A, 0x400B, and 0x400C.

Are they all on firmware 1.55.7_r28193?  Is that the latest firmware for the bulb?  What is the _Image Type_ for the bulb firmware?  Check in the _STD OTAU plugin_ panel, select the bulb from the map and press_Query_ to populate the table entry.

D34DC3N73R, 2021-08-08:
I have 4 of these bulbs; they are all on 1.55.7_r28193. They were all showing 0x0000 for color capabilities, but after hitting READ on the attributes panel, the color capability values all updated to 0x001f, and the rest API then correctly shows `"colorcapabilities": 31` for all these bulbs. And I'm also happy to report that colorloop is again working with 2.12.03.

before read
[image]

after read
[image]

I'm not too well versed with the UI, as I just run the docker container. but is this the STD OTAU panel info you're looking for? Nothing changes when I hit "Query".
[image]

Side note, I don't have a hue bridge and didn't think it was possible to update the firmware for hue bulbs without one. Is that not the case?

ebaauw, 2021-08-08:
> after read

So the light exposes the correct attribute values.  Did deCONZ update the resource after reading these?  If so, setting colorloop should work now.  If not, I really recommend to try and reset/re-pair one of the lights.  Maybe you paired them originally with a deCONZ version that failed to setup these attributes.

> Nothing changes when I hit "Query".

The OTAU server is disabled (see the note just above the table).  Unfortunately, you can only enable it through the API, not from the GUI.  The old web app shows the under _Advanced Settings_.  It’s a hidden setting; the API only shows it with the right HTTP header.  If you’ve installed, Homebridge Hue, you could also use `ph -p put /config ‘{"otauactive": true}`.

> Side note, I don't have a hue bridge and didn't think it was possible to update the firmware for hue bulbs without one. Is that not the case?

Not anymore.  See https://github.com/dresden-elektronik/deconz-ota-plugin.  I don’t think the colorloop issue is related to the light firmware, though.

D34DC3N73R, 2021-08-08:
> Did deCONZ update the resource after reading these?

Yes, I ran a rest API get on /lights after reading all attributes and color capabilities updated to 31 for each of them. I tried colorloop in HA and it worked.

I enabled OTAU via the rest API, but I'm still not sure this is showing what you're after
[image]
[image]
[image]

I'll keep an eye on the lights and see if color capabilities resets to 0 or stays at 31. If it resets, I'll try the reset/re-pair.

ebaauw, 2021-08-08:
> I'll keep an eye on the lights and see if color capabilities resets to 0 or stays at 31. If it resets

They shouldn’t; once read the API should persist these in the database.

> I enabled OTAU via the rest API, but I'm still not sure this is showing what you're after

Yes, _Image Type_ 0x0114.  There’s a later version linked from the OTAU plugin README.

D34DC3N73R, 2021-08-08:
> There’s a later version linked from the OTAU plugin README.

I see. Would you recommend I update the bulbs using that firmware and the OTAU Update tool?

ebaauw, 2021-08-08:
I keep my lights and other Zigbee devices up to date with their latest firmwares, and have been using the OTAU plugin for that, since we figured out how to handle the Hue firmware files.

The [release notes](https://www.philips-hue.com/en-us/support/release-notes/lamps) for the firmware mention: “this update provides improvements to stability and security.”  That sounds like a good thing, I suppose.

DanteFireX, 2021-08-08:
Thanks to all that contributed! I can confirm that re-reading the attributes (as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4972#issuecomment-894737954) fixes the problem (also on current stable).

D34DC3N73R, 2021-08-09:
Closing this - The color capabilities attributes seem to last through device power off as well as deconz restarts. Thanks to everyone for the help!

closed 2021-08-09 by D34DC3N73R