dresden-elektronik/deconz-rest-plugin #4974: Iris il02_1

Abandonado, 2021-06-04:
## Device

-Product name: Iris il02_1 keypad
Manufacturer: Imagic by GreatStar
Model identifier: 1112-S
Device type :
Lock
Remote
Other: 
[image]
[image]
[image]

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-06-04:
Hello, have you a linux machine to test code modification ?

Abandonado, 2021-06-04:
First sorry for my incomplete previous posts. I do not have much experience. What I use is Hassio in a nuc. Thanks

Smanar, 2021-06-04:
If you can add the device node picture ? (It's the naode with the cluster list)
After I have enought information to make something, but 
- the first keypad support is still in waiting list.
- If you can't compile the code, there is no way to test it ATM.

You don't have a small raspberry or a good VM ?
It s possible on docker, but I prefer let this method for user with more experience.

Abandonado, 2021-06-04:
Tomorrow I send the capture that you request. I have no access but the nuc, sorry. And little experience since I just started in this world. Thanks for your help beforehand

Abandonado, 2021-06-04:
[image]
Is this?

Smanar, 2021-06-04:
Yep exactly, thx.
Will take a look tomorow if I can add it in blind mode, but without guarranty.

Abandonado, 2021-06-04:
Ok very thanks

Smanar, 2021-06-05:
Bad luck, have started to include it https://github.com/Smanar/deconz-rest-plugin/tree/keypad_1112-S
But this device have some extra features, no in code yet, so I relay need to test it first.

For the moment, I can explain how to compile it, but I realy need someone that can compile code.

There is a procedure for docker, but I can help you for this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-852157541

Abandonado, 2021-06-05:
Okay. Thank you very much for your interest. I'm really very new to this. I'll find someone who has docker to see if they help me with this, since I don't know how it works.

closed 2021-07-05 by github-actions[bot]