dresden-elektronik/deconz-rest-plugin #4977: Xiaomi Motion Sensors disappear after power failure

ipodmusicman, 2021-06-05:
## Describe the bug

I have 6 Xiaomi window / door sensors, 4 Xiaomi motion sensors, 5 Xiaomi temperature/humidity sensors and one Xiaomi vibration sensor connected to my Combee II stick running Deconz.

Every time we have a power failure, 2 of the motion sensors disappear from Deconz.  It just so happens that these two motion sensors were purchased about 2 months ago whereas others were purchased in 2019.

When I log onto Phoscan they are marked as not reachable and when I log onto Deconz where I can see the Zigbee network map, they don't even appear there at all.  It is as if, due to the power failure, they have disappeared.  All other sensors connect and are available for use besides these two.

I then have to remove them from Phoscan and re-pair them in order for them to work - until the next power failure that is which is occurring regularly at present in my country. 

## Steps to reproduce the behavior

1.  Pair motion sensor
2. Phoscan picks it up and I am able to use it in Home Assistant
3. Power failure occurs
4. 2 hours later power is restored and my Raspberry Pi 4 with Home Assistant boots up.
5. These two motion sensors do not activate when motion is detected.
6. Log onto Deconz and they are not even on the map.  Logs to indicate some issues though, so see the image attached with the orange rectangle where I trigger the sensor after power has been restored.

## Expected behavior

When power is restored, these two motion sensors should appear back onto the network.

## Screenshots

None

## Environment
 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.11.05 / 5/9/2021
 - Device: (ConBee II)
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system?  No

## deCONZ Logs

Unable to get logs as I am using the Deconz add-on in Home Assistant, but I did see this in the logs under debug view.  Here is a screen grab of the logs from start up.  Notice the orange rectangle around where I triggered the sensor itself.

[image]

Mimiix, 2021-06-05:
The lack of routers causes these issues. Xiaomi devices tend to drop off this way. Having some routers really increases stability

ipodmusicman, 2021-06-06:
During the test, the motion sensor was close to the Combee II stick whereas other motion sensors, door sensors, etc were at their usual spots far away from the stick and they came online just fine.  It is only these two - and these are the ones I purchased about 2 months ago or so.  I am going on a whim here, but I suspect that they are not 100% compatible with Deconz and I am hoping that the log I supplied above would give a clue.

I really don't think it has anything to do with the lack of routers hence why I logged it as a bug report or not a user question.

I even tested using ZHA in Home Assistant (alternative to using Deconz) and that worked fine.  When I restored power, the same motion sensor that I used to reproduce this issue came online and worked fine.  I am really hoping not to have to be forced to re-pair ALL my sensors and move over to ZHA in order to solve this. 

I want to give the opportunity to have this investigated and hopefully fixed in the next release of Deconz and if there is anything that you need from me in order to troubleshoot this, please let me know.

Smanar, 2021-06-06:
Are you able to check too the node picture in the GUI ?
Sometime the sensor are not correctly included (not finished), so not saved.

It s possible too seing it in log, but I don't remember the line.

Was your 2 last devices ? or you have included more after them ?

ipodmusicman, 2021-06-06:
Hi Smanar, the specific motion sensor is not even in the map after power is restored, but when I re-pair the motion sensor, by firstly removing it from Phoscan and adding it again, it appears on the map.  See the screenshot of the detail when I click on the sensor as well as screenshots from the log when I paired the sensor and triggered its motion.

These were my last 2 devices that I paired when I noticed the issue about a week or so ago, but for the purpose of assessing this, I've only involved one of the sensors.

[image]

[image]

Smanar, 2021-06-06:
Can you show the node picture, with the cluster list (and the node title)

ipodmusicman, 2021-06-06:
Here is the entire node map and node list, but when I click on cluster info, it says that I need to click on a cluster to get more information, but I am not familiar  with clusters and it seems I don't have a cluster to click on.

[image]

Smanar, 2021-06-06:
And if you clic on the circle on the node title, all clusters are present ?

ipodmusicman, 2021-06-06:
@Smanar here you go.  I right-clicked on the sensor and executed the various menu commands which caused the little left circle to blink red and when clicking on the cluster info, I got the list.

[image]

Smanar, 2021-06-06:
Ok, so it s that, your device is not correclty included.
You can see the "ffff" ? I think if you compare with other you don't have same thing, and more cluster.

You can restart an inclusion (without delting the old one, why you delete the old sensor to re-include it ? just re-include it, some data can be re-used), and re-include it using the 2/3s short press tips to prevent the device go to sleep.

Else, set phoscon in permit join, and resend all the "descriptor" command you have on the first capture on your last post, but you need to awake the sensor in same time by pressing shortly the button.

You need to have same result than on other sensors.

BTW have you check battery ?

SwoopX, 2021-06-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

ipodmusicman, 2021-06-06:
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

No ways!!!   Thanks VERY much!  I tested this and rebooted my Pi and voila!  My sensor is still available!  Of course I'll keep an eye on things as we are going through rolling blackouts currently.

Thanks so much @SwoopX  for pointing me in the right direction and thanks to @Smanar for all your assistance.

ipodmusicman, 2021-06-06:
Just as a parting shot, is there anything in Deconz that can be done to prevent this step from having to occur?

closed 2021-06-06 by ipodmusicman

Smanar, 2021-06-06:
Not in deconz, but the short press help.