dresden-elektronik/deconz-rest-plugin #4979: No connection in Docker possible

nskorwi-bewoop, 2021-06-06:
## Describe the question or issue you are having

Good day!
I've been trying to integrate my Conbee 2 stick into my Docker system for days now. But no matter what I do, I just can't do it. In addition, I had the system running for a long time until the SD card broke. (and the Conbee also worked with all devices)
I had to reinstall the system (just like before) but a connection to the Conbee is still impossible. The stick is apparently recognized and is also displayed with lsusb. However, the Phoscon app shows me: Firmware not connected. I already thought my stick is defective, which is why I bought a second one. The result is the same.

But apparently the Docker image does not recognize the device or does not accept the information in the Docker Compose. (last picture)

## Screenshots

[image]
[image]
[image][image]
[image]
[image]

## Environment

 - Host system: (Raspberry Pi4 8gb RAM)
 - Running method: ( Marthoc Docker container)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes ) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?
The system runs on a 1/2 terabyte HDD hard drive. But when I install deCONZ on Raspian normally it works. (So ​​that can't be the problem)
## deCONZ Logs

  If applicable provide some logs. 
  Debug logs can help to investigate certain problems like device pairing and communication issues.
  Logging can be enabled on the command line as described in:
  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-switches
  Secondly, for versions above 2.05.87: In the deCONZ GUI: click Help > debug view. Output of APS,ZDP, ZCL and INFO can be handy. 

## Additional context

To install the Docker, I use a ready-made installation script from: https://ei23.de/diy-smarthome/
But even if I install everything manually, it doesn't work.

I despair and am happy about every answer. With best regards -nskorwi-

senilio, 2021-06-06:
You are running both deCONZ and zigbee2mqtt. Are both of them configured to use the same coordinator? Some logs from from the deCONZ container would be helpful.

My guess is that you either have multiple containers trying to access the device at the same time, or you need to run deCONZ in privileged mode due to the host running a restrictive apparmor docker profile.

nskorwi-bewoop, 2021-06-06:
I have deactivated Zigbee2mqtt and now the Conbee has a connection. Thanks a lot for this. Now, however, the stick looks for my lights for a long time and after searching 4 times it only found 2 of my 11 lamps. Do you have any suspicions?

senilio, 2021-06-07:
There can be lots of reasons for this.

Signal quality / weak mech
Interference
A zigbee device can be successfully paired but still not show in phoscon
Failure to put the zigbee device into pairing mode

It usually works best when you set Phoscon to search for devices for 10 minutes, then don't give up as it might take a few attempts.

Since the original issue seems to be solved, would you mind closing this ticket? 

For support with pairing stubborn zigbee devices, I think your best bet is to ask for help in discord, or search around in the github issues (both open and closed), as there may be vendor specific tricks to try out.

Discord link https://discord.gg/58yJPaWz

Thanks

Mimiix, 2021-06-07:
As it seems fixed, i'll close it.

closed 2021-06-07 by Mimiix