dresden-elektronik/deconz-rest-plugin #4984: Conbee 1 windows 10 drivers

clockworkant, 2021-06-07:
## Describe the question or issue you are having

I need to get access to the Conbee 1 windows 10 driver. Currently the conbee one doesn't have drivers in the device manager.

## Screenshots

[image]

## Environment

Windows 10

 - Host system: PC
 - Running method: Windows
 - Firmware version: na
 - deCONZ version: na
 - Device: ConBee I
 - Do you use an USB extension cable: (yes / no) -- yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Smanar, 2021-06-11:
No one have a link for working drivers ?
I have same issue, I have tried a "random" driver, It have worked, but the device is not usable with deconz even it s reconised.

Smanar, 2021-06-14:
Ok so perhaps some news
Have tried https://ftdichip.com/drivers/ the conbee is reconized but not working
Can try this one https://ftdichip.com/drivers/d2xx-drivers/ (not no tested on my side, prod machine is used)

closed 2021-07-13 by github-actions[bot]