dresden-elektronik/deconz-rest-plugin #4985: Ikea Shortcut Button not correctly recognized

T0mWz, 2021-06-07:
## Describe the bug
Hi all,

When I try to add a Ikea Shortcut Button via the Deconz web environment, the device will not be displayed after pairing.
However, via the  Deconz interface I see the Shortcut button correctly. Also in Home Assistent, the devices will be shown without any options.

## Steps to reproduce the behavior
Just add a Shortcut Button via Deconz Webenvironment

## Expected behavior
I should expect a button in Deconz web environment and some options available within Home Assistent

## Screenshots
[image]
[image]
[image]
[image]

## Environment
 - Host system: PC
 - Running method: Home Assistent deCONZ Add-on and Marthoc Docker container
 - Firmware version: 264A0700
 - deCONZ version: 2.11.05 / 09/05/2021 , also in 2.12.00
 - Device: ConBee II

## deCONZ Logs
```
20:24:58:226 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, broadcast to: 0x0015, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 3
20:24:58:227 enqueue event state/buttonevent for /sensors/47
20:24:58:228 enqueue event state/lastupdated for /sensors/47
20:24:58:228 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
20:24:58:229 Force binding of attribute reporting for sensor TRADFRI SHORTCUT Button
20:24:58:233 0x680AE2FFFE1473A9 [TRADFRI SHORTCUT Button] create binding for client cluster 0x0006 on endpoint 0x01
20:24:58:234 queue binding task for 0x680AE2FFFE1473A9, cluster 0x0006
```

## Additional context

Mimiix, 2021-06-07:
As far as I know it should produce deconz events in home assistant.

Can you check that?

T0mWz, 2021-06-07:
Hi @Mimiix ,

Thanks for your help!

Not sure of I correctly understand what you mean, but within HA I don't see any option/event for this device.

[image]

Thanks for your help! :-)

Mimiix, 2021-06-07:
Hi there! 

In developer settings there's a event listener

There you need to listen to deconz_event

T0mWz, 2021-06-08:
Maybe I'm to stupid for this.. With a normal Tradfri on/off switch, I retrieved an event on `deconz_event` channel.
However with the Shortcut Button I retrieved nothing..  The button is available within Home Assistent without any option, in the Phoscon Webapp it isn't visible at all. I only found the device via de Phoscon GUI app

Is it maybe possible to integrate the button like a motion sensor for example? Then you will get a trigger for a event and maybe also the battery status?

[image]

[image]

Mimiix, 2021-06-08:
I think it isn't paired correctly. it should always send events in there (like the on off button).

On the HA intergration, you need to ask @kane610 :).

T0mWz, 2021-06-08:
> I think it isn't paired correctly. it should always send events in there (like the on off button).
> 
> On the HA intergration, you need to ask @Kane610 :).

I don't know, what I can change in the pairing. 😅 
Have two of them and they are both shows exactly the same behaviour. 

Tips are welcome.

Kane610, 2021-06-08:
> Maybe I'm to stupid for this.. With a normal Tradfri on/off switch, I retrieved an event on `deconz_event` channel.
> However with the Shortcut Button I retrieved nothing.. The button is available within Home Assistent without any option, in the Phoscon Webapp it isn't visible at all. I only found the device via de Phoscon GUI app
> 
> Is it maybe possible to integrate the button like a motion sensor for example? Then you will get a trigger for a event and maybe also the battery status?
> 
> [image]
> 
> [image]
> 
> However in the deconz logs I got some `events` when I push the button, so something is received. :-)
> 
> ```
> 08:56:46:913 enqueue event event/validgroup for /sensors/50
> 08:56:46:914 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, broadcast to: 0x0018, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 28
> 08:56:46:915 enqueue event state/buttonevent for /sensors/50
> 08:56:46:916 enqueue event state/lastupdated for /sensors/50
> 08:56:46:916 0x680AE2FFFE34CF23 (TRADFRI SHORTCUT Button) create binding for attribute reporting of cluster 0x0001 on endpoint 0x01
> 08:56:46:917 queue binding task for 0x680AE2FFFE34CF23, cluster 0x0001
> 08:56:46:918 Force binding of attribute reporting for sensor TRADFRI SHORTCUT Button
> 08:56:46:921 Websocket 172.19.0.3:48226 send message: {"e":"changed","id":"50","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-06-08T06:56:46.916"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:cf:23-01-1000"} (ret = 165)
> 08:56:46:922 Websocket 172.19.0.3:43012 send message: {"e":"changed","id":"50","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-06-08T06:56:46.916"},"t":"event","uniqueid":"68:0a:e2:ff:fe:34:cf:23-01-1000"} (ret = 165)
> 08:56:46:925 discard sensor state push for 50: state/lastupdated (already pushed)
> 08:56:47:249 enqueue event config/localtime for /config/
> ```

It is not supported as a device trigger yet. Create an issue here and share the events of the different button presses, short long double click etc. and label them so I can map them and I will get it in for the HA 2021.7 release

T0mWz, 2021-06-10:
Support added in HA; https://github.com/home-assistant/core/commit/79996682e570342272675c88dffdc0b25ac70242

T0mWz, 2021-06-10:
@Mimiix is there any possibility to make the Ikea Shortcut Button also visible via the Phoscon WebApp after adding. Currently nothing happend. Also a join is not visible there.

SwoopX, 2021-06-11:
Closing this one as it is not deconz related.

@T0mWz Iirc, it is already working in Phoscon beta. Check out the corresponding repo.

closed 2021-06-11 by SwoopX