dresden-elektronik/deconz-rest-plugin #4988: Sunricher SR-ZG2833PAC-FOH detected but no button events

kvanisterdael, 2021-06-09:
## Describe the bug
According to #4703 and #4707 this device is supported in version 2.12 beta.
I can detect the push button coupler without a problem and it gets added as a FOHSwitch:

From the REST api, I can see this:
```
{
    "config": {
        "on": true
    },
    "ep": 242,
    "etag": "8200b9cda239fd72505e38bb2c826040",
    "lastseen": "2021-06-09T08:32Z",
    "manufacturername": "PhilipsFoH",
    "modelid": "FOHSWITCH",
    "name": "Friends of Hue Switch",
    "state": {
        "buttonevent": 7002,
        "lastupdated": "2021-06-09T08:32:52.591"
    },
    "type": "ZGPSwitch",
    "uniqueid": "00:00:00:00:01:71:95:7b-f2"
}
```

## Steps to reproduce the behavior
I added the device by holding the K1 contact closed for 10 seconds (my gateway is on Zigbee channel 11).
After 10 seconds a light flashes and the Phoscon app says it was detected.
According to https://phoscon.de/en/support#pairing-friends-of-hue-switch, I shortly closed contact K1 and K4, but I don't know if this is really necessary.

## Expected behavior
I would expect to see the buttonevent updates in the Websockets, but I don't see anything.
I would expect to see the device in my Deconz GUI, but I can't find it anywhere.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: 2.12.00
 - Device: RaspBee I

## deCONZ Logs
closing the contact doesn't seem to generate any debug log

## Additional context
none

Thanks in advance!

LeoeLeoeL, 2021-06-09:
I have the Vimar FOH.
It's not showed in VNC but works.

kvanisterdael, 2021-06-10:
> 
> 
> I have the Vimar FOH.
> It's not showed in VNC but works.

I also have a similar FOH push button from SUNRICHER and that one works fine.

SwoopX, 2021-06-11:
> I would expect to see the buttonevent updates in the Websockets, but I don't see anything.

The REST API output shows that there are button events.

> I would expect to see the device in my Deconz GUI, but I can't find it anywhere.

Green Power switches should never show up in deconz GUI.

closed 2021-07-10 by github-actions[bot]

eakoning, 2022-03-07:
> Green Power switches should never show up in deconz GUI.
Can you explain why that is @SwoopX ?