dresden-elektronik/deconz-rest-plugin #4990: Tuya Smart Air Box

triumvirrr, 2021-06-09:
## Device

- Product name: Tuya Smart Air Box
- Manufacturer: _TZE200_8ygsuhe1
- Model identifier: TS0601
- Device type : 
  - Sensor (temperature, humidity, co2, voc, formaldehyd)
 
  

## Screenshots
[image]
[image]

### Basic
[image]
[image]

Smanar, 2021-06-09:
Hello, have you a linux machine will full OS to test code modification ?
Can you show at least one tuya return from log ?
On deconz / help / debug view, I just need one the dual line starting by

```
Tuya debug 4 : Address .........
Tuya debug 5 : Status: ........
```

triumvirrr, 2021-06-09:
Hello!

Thank you very much for your fast response!

At the moment, deconz is running on raspberry with a RaspBee1. If I can be of any help testing changes there, I gladly will!

Here are a few recent debug lines:
18:30:52:770 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000012020004000000f9
18:30:52:781 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 249
18:30:52:977 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000012020004000000f9
18:30:52:982 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 249
18:30:53:079 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000012020004000000f9
18:30:53:084 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 249
18:30:53:269 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001302000400000200
18:30:53:274 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 512
18:30:55:497 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001302000400000201
18:30:55:503 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 513
18:30:56:530 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00501602000400000002
18:30:56:548 Tuya debug 5 : Status: 0 Transid: 80 Dp: 534 (0x02,0x16) Fn: 0 Data 2
18:30:56:683 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00501602000400000002
18:30:56:688 Tuya debug 5 : Status: 0 Transid: 80 Dp: 534 (0x02,0x16) Fn: 0 Data 2
18:30:56:873 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00501602000400000002
18:30:56:889 Tuya debug 5 : Status: 0 Transid: 80 Dp: 534 (0x02,0x16) Fn: 0 Data 2
18:30:57:089 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00501502000400000001
18:30:57:094 Tuya debug 5 : Status: 0 Transid: 80 Dp: 533 (0x02,0x15) Fn: 0 Data 1
18:30:57:820 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00500202000400000173
18:30:57:829 Tuya debug 5 : Status: 0 Transid: 80 Dp: 514 (0x02,0x02) Fn: 0 Data 371
18:30:58:028 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000012020004000000fa
18:30:58:033 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 250
18:30:58:997 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001302000400000202
18:30:59:002 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 514
18:30:59:311 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:30:59:323 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:30:59:424 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:30:59:438 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:30:59:537 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:30:59:542 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:30:59:807 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:30:59:821 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:31:00:540 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:31:00:558 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:31:00:768 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
18:31:00:774 Tuya debug 5 : Status: 0 Transid: 82 Dp: 534 (0x02,0x16) Fn: 0 Data 1
18:31:01:170 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521502000400000000
18:31:01:206 Tuya debug 5 : Status: 0 Transid: 82 Dp: 533 (0x02,0x15) Fn: 0 Data 0
18:31:01:379 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00520202000400000167
18:31:01:407 Tuya debug 5 : Status: 0 Transid: 82 Dp: 514 (0x02,0x02) Fn: 0 Data 359

Smanar, 2021-06-09:
And you have a linux machine with full OS.

This device realy look "standard". It will be easy to add but big problem, it realy spam the network ....
And no solution on other zigbee project (or haven't see it), and this device use only the tuya one.

triumvirrr, 2021-06-09:
It seems, that it is already supported in Zigbee2MQTT:
https://www.zigbee2mqtt.io/devices/TS0601_air_quality_sensor.html

Can you suggest a starting point for me to read and learn how to work out the necessary changes for this device to work? I am somehow hoping to just clone the code for another device and copy/paste the new identifiers, could this work?

Smanar, 2021-06-09:
Sure, so you need a linux machine with full OS to compile the code

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_airbox https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You can use log to add features

> 18:31:01:379 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00520202000400000167
> 18:31:01:407 Tuya debug 5 : Status: 0 Transid: 82 Dp: 514 (**0x02,0x02**) Fn: 0 **Data 359**

Values in black are the values used

I have started something https://github.com/Smanar/deconz-rest-plugin/commit/ec4e1cf0a759a0d50d65022d5345459bb89ef057
- the code is untested and not finished, only 3 sensor can work, other are not yet in deconz, but you can add them just with copy paste, but from comment from other project, values are not reliable, so test them before.
- the device can be reconized (have added the mac address in whitelist)
- this code is free to use, if you want to search yourself, you can download it and use it instead of using my branch.

Tell me if you need more help.

triumvirrr, 2021-06-10:
Thanks!
Your code did compile (with a few modifications, see below), but when the .so is replaced, deconz crashes with a segmentation fault. Reverting to the old .so solves this.

My current deCONZ version is the following, do I need another one?
[image]

Additional info:
This is running on a Raspberry 1 at the moment, so I used 'qmake && make' instead of 'qmake && make -j2', but this shouldn't make a difference, should it?

CLI output:
deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Segmentation fault

Modifications to your commit:
de_web_plugin.cpp, 5274:
fpCarbonMonoxideSensor.inClusters.push_back(TUYA_CLUSTER_ID);
fpCarbonMonoxideSensor.inClusters.push_back(IAS_ZONE_CLUSTER_ID);

tuya.cpp, 676:
break;

Smanar, 2021-06-10:
>Within deCONZ, I have now 5 new devices, neither of them having any clusters visible and one device named "CarbonMonoxide" with the same clusters as the "Smart Plug" device which already appeared before.

Hu ?
Thoses devices have different Mac adress and they are end router, are you sure it the same device ?
For the name on deconz, I think it s normal deconz use the one from the last devices created.

>Why do you add 2 centigrades to the temperature when converting the value? Is this just legacy, because it was done so for the siren sensor?

Yep was only for the siren, can compare with your, and remove it if you don't need it (but the siren need it)

>I also think, that this device does not come with an CO alert sensor, but a numerical CO2 sensor.

Right, so you can't use old one, I think you need to create a new one.

Temperature seem working.
Humidity still null ? I m checking (0x02 0x13)

```
18:30:59:002 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 514
18:30:59:311 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00521602000400000001
```

Have updated the code a little, but need to re-include it to remove the ZHACarbonMonoxide and add the ZHAairquality

```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

triumvirrr, 2021-06-10:
About the phantom devices: I am not 100% sure, but I havent seen them before and they (randomly) pop up as I pair the Tuya Smart Air Box. Can it be, that the missing values (I do only see 3 sets of data in the logs, whereas there should be data for temp, hum, co2, voc and formald) are somehow transmitted via different MACs?

About your new commit:
Thanks once again.
Did compile smoothly, only one ")" was missing in line 688 of luya.cpp

I have to admit that coding on this project (more than just bringing your commits to compile and test them) is way beyond my skills, so I will need to hope for your experience and patience to bring this up and running.

Current situation:
Temperature is now fine, Humidity still 0, CO2 sensor is gone via REST API and VOC Airquality is here but shows no value whatsoever.

One possibly quite dumb question:
Which purpose should the "ZHAAlarm" have?

Regarding VOC:
Are you sure, that 0x0215 carries the VOC value at all? The value should be way higher, I am measuring indoor, my other sensors (Develco) read between 10-200 VOC ppb most of the time, in the logs I can only find 0 and 1 as value.

REST:
```json
62: {
"config": {
"offset": 0,
"on": true,
"reachable": true
},
"ep": 1,
"etag": "a8a52d11efbd14efa3149a55dca43351",
"lastseen": "2021-06-10T16:15Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Temperature 62",
"state": {
"lastupdated": "2021-06-10T16:18:10.045",
"temperature": 2550
},
"type": "ZHATemperature",
"uniqueid": "cc:86:ec:ff:fe:62:65:e6-01-0402"
},
63: {
"config": {
"offset": 0,
"on": true,
"reachable": false
},
"ep": 1,
"etag": "8b4c157237ed09aadb15446f4f3c7435",
"lastseen": "2021-06-10T16:15Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Humidity 63",
"state": {
"humidity": null,
"lastupdated": "2021-06-10T16:18:13.897"
},
"type": "ZHAHumidity",
"uniqueid": "cc:86:ec:ff:fe:62:65:e6-01-0405"
},
64: {
"config": {
"enrolled": 0,
"on": true,
"pending": [],
"reachable": false
},
"ep": 1,
"etag": "0294df102f192658d4a677d0edd8e1a2",
"lastseen": "2021-06-10T16:15Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Alarm 64",
"state": {
"alarm": false,
"lastupdated": "none",
"lowbattery": false,
"tampered": false
},
"type": "ZHAAlarm",
"uniqueid": "cc:86:ec:ff:fe:62:65:e6-01-0500"
},
65: {
"config": {
"on": true,
"reachable": true
},
"ep": 1,
"etag": "5e6186c51dd9a23f8f5f00594bd53596",
"lastseen": "2021-06-10T16:15Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "AirQuality 65",
"state": {
"lastupdated": "none"
},
"type": "ZHAAirQuality",
"uniqueid": "cc:86:ec:ff:fe:62:65:e6-01"
}
```
Here are the current logs to compare the values once again, grouped by data id:
```log
18:08:19:604 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000012020004000000ff
18:08:19:620 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 255
18:08:22:041 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001202000400000101
18:08:22:049 Tuya debug 5 : Status: 0 Transid: 0 Dp: 530 (0x02,0x12) Fn: 0 Data 257

18:08:20:145 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001302000400000180
18:08:20:154 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 384
18:08:22:248 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00001302000400000184
18:08:22:257 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 388

18:08:20:233 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00081502000400000001
18:08:20:248 Tuya debug 5 : Status: 0 Transid: 8 Dp: 533 (0x02,0x15) Fn: 0 Data 1

18:08:20:310 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 00080202000400000174
18:08:20:329 Tuya debug 5 : Status: 0 Transid: 8 Dp: 514 (0x02,0x02) Fn: 0 Data 372

18:08:22:335 Tuya debug 4 : Address 0xCC86ECFFFE6265E6 Payload 000a1602000400000002
18:08:22:344 Tuya debug 5 : Status: 0 Transid: 10 Dp: 534 (0x02,0x16) Fn: 0 Data 2
```

Smanar, 2021-06-10:
> About the phantom devices: I am not 100% sure, but I havent seen them before and they (randomly) pop up as I pair the Tuya Smart Air Box. Can it be, that the missing values (I do only see 3 sets of [...]

I don't think, it s one MAC address for one device.
0x001788... are for philips.

>I have to admit that coding on this project (more than just bringing your commits to compile and test them) is way beyond my skills, so I will need to hope for your experience and patience to bring this up and running.

Lol, I think it will be faster if you try yourself, you have the method, the device, and probably more time than me. I m agree I know better than you how work deconz, but lot of work can be done just with copy/paste and mimic code.

>Which purpose should the "ZHAAlarm" have?

None ^^.
My bad, forget to remove 
```
                            fpAlarmSensor.inClusters.push_back(TUYA_CLUSTER_ID);
                            fpAlarmSensor.inClusters.push_back(IAS_ZONE_CLUSTER_ID);
```

Have improved the code a little for air quality, but I realy don't know VOC/CO2 or other, I have tale values from here https://github.com/zigpy/zha-device-handlers/pull/903/files

And I have added some debug line for humidity, just after the lines
>18:08:22:257 Tuya debug 5 : Status: 0 Transid: 0 Dp: 531 (0x02,0x13) Fn: 0 Data 388

You need to have lines starting by "Tuya debug Humidity"

triumvirrr, 2021-06-10:
Got Humidity to work. The value already reads (percentage *10), so multiplying with another 100 got it out of range (0-10000), multiplying by only 10 did the trick.

```
                    case 0x026A: // Siren Humidity
                    {
                        qint16 Hum = static_cast<qint16>(data & 0xFFFF) * 100;
                        ResourceItem *item = sensorNode->item(RStateHumidity);

                        if (item && item->toNumber() != Hum)
                        {
                            item->setValue(Hum);
                            Event e(RSensors, RStateHumidity, sensorNode->id(), item);
                            enqueueEvent(e);
                            update = true;
                        }
                    }
					break;
					
                    case 0x0213: // smart air box humidity
                    {
                        qint16 Hum = static_cast<qint16>(data & 0xFFFF) * 10;
                        ResourceItem *item = sensorNode->item(RStateHumidity);

                        if (item && item->toNumber() != Hum)
                        {
                            item->setValue(Hum);
                            Event e(RSensors, RStateHumidity, sensorNode->id(), item);
                            enqueueEvent(e);
                            update = true;
                        }
                    }
```

```
67: {
"config": {
"offset": 0,
"on": true,
"reachable": false
},
"ep": 1,
"etag": "0185f9b480f5dc755a510aba0c4a711a",
"lastseen": "2021-06-10T17:04Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Humidity 67",
"state": {
"humidity": 4670,
"lastupdated": "2021-06-10T17:04:54.262"
},
"type": "ZHAHumidity",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-0405"
},
```

triumvirrr, 2021-06-10:
One question for a quick and dirty approach to get the values at least readable via REST:
Can I add, say, 4 temperature sensors and 1 humidity sensor to one device and fill said temperature sensors with temperature, co2, voc and HCHO and then just work out the real meanings on the frontend?

triumvirrr, 2021-06-10:
Success, all sensors working :-)

Do you want me to push my changes to your git? If yes, please advise me on how to do this.
```json
78: {
"config": {
"offset": 0,
"on": true,
"reachable": false
},
"ep": 1,
"etag": "d04d771b6e8a82a8bbe495174945ee86",
"lastseen": "2021-06-10T22:31Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Temperature 78",
"state": {
"lastupdated": "2021-06-10T22:32:19.332",
"temperature": 2440
},
"type": "ZHATemperature",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-0402"
},
79: {
"config": {
"offset": 0,
"on": true,
"reachable": false
},
"ep": 1,
"etag": "f2c4e426a18c5a6c02377a039e409287",
"lastseen": "2021-06-10T22:31Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "Humidity 79",
"state": {
"humidity": 5530,
"lastupdated": "2021-06-10T22:32:21.932"
},
"type": "ZHAHumidity",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-0405"
},
80: {
"config": {
"on": true,
"reachable": true
},
"ep": 1,
"etag": "e9519045b8f823b038436ec496fd15f2",
"lastseen": "2021-06-10T22:31Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "AirQuality 80",
"state": {
"airquality": null,
"airqualityppb": 200,
"lastupdated": "2021-06-10T22:32:17.314"
},
"type": "ZHAAirQuality",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-ef00"
},
81: {
"config": {
"on": true,
"reachable": true
},
"ep": 1,
"etag": "17bfd9b43ff091186e1b4da06a31048b",
"lastseen": "2021-06-10T22:31Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "AirQualityCO2 81",
"state": {
"airquality": null,
"airqualityco2ppm": 370,
"lastupdated": "none"
},
"type": "ZHAAirQualityCO2",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-ef00"
},
82: {
"config": {
"on": true,
"reachable": true
},
"ep": 1,
"etag": "b33f508215ab50e5e48a80f73f53f136",
"lastseen": "2021-06-10T22:31Z",
"manufacturername": "_TZE200_8ygsuhe1",
"modelid": "TS0601",
"name": "AirQualityHCHO 82",
"state": {
"airquality": null,
"airqualityhchomgm3": 1,
"lastupdated": "none"
},
"type": "ZHAAirQualityHCHO",
"uniqueid": "cc:86:ec:ff:fe:62:65:d6-01-ef00"
}
```

Smanar, 2021-06-11:
>Can I add, say, 4 temperature sensors and 1 humidity sensor to one device and fill said temperature sensors with temperature, co2, voc and HCHO and then just work out the real meanings on the frontend?

The rules is : 1 sensor by endpoint
Problem, tuya use only 1 endpoint.

So rule 2 : Use existing sensor type.
Problem, you have new sensors.

So rule 3 : Make that like you prefer, devs will choose later ^^, but I think make like you have done is better, a new sensortype. Like this they can be used later by other devices

```
"airquality": null,
"airqualityhchomgm3": 1,
"lastupdated": "none"
```

I thin, you can remove "airquality", and you probably miss something to update the sensor `"lastupdated": "none"`

>Do you want me to push my changes to your git? If yes, please advise me on how to do this.

Download the code, and make your own PR, like this you will be able to correct it yourself it according to other devs reactions, or improve it (the PR validation can take 1 month, enought time to improve it)

triumvirrr, 2021-06-11:
Thank you very much for your help, very much appreciated :-)

I have made my own PR with your changes merged with mine.
I hope no one of the devs gets sick by reviewing my hands-on copy/paste coding ;-)

Here we go:
https://github.com/triumvirrr/deconz-rest-plugin/tree/tuya_smart_airbox

Known problems - I would very much appreciate every hint on how to solve this:
- Device is shown as Smart Plug in Phoscon
- Only one device can be paired, a second device fails pairing
- After restart of deCONZ, the newly inrroduced nodes state/airqualityhchomgm3 and state/airqualityco2ppm disppear. They are only repopulated after deleting and rejoining the device

Smanar, 2021-06-11:
>Device is shown as Smart Plug in Phoscon

Not your fault, and you can do nothing for that, it s the device itself that give this information, and phoscon just display it, but you can prevent the light device by created (as it don't have something usefull)

        if (node->nodeDescriptor().manufacturerCode() == VENDOR_EMBER)
        {
            //Tuya black list
            //_TZE200_aoclfnxz is a thermostat
            //_TZE200_c88teujp same
            if (R_GetProductId(&lightNode).startsWith(QLatin1String("Tuya_THD")))
            {
                hasServerOnOff = false;
            }

Can add your model to prevent the light entry creation

```
            if (R_GetProductId(&lightNode).startsWith(QLatin1String("Tuya_THD")) ||
                R_GetProductId(&lightNode) == QLatin1String("Smart air box"))
```

>Only one device can be paired, a second device fails pairing

Devs are speaking about that ATM, can be a deconz issue.

>After restart of deCONZ, the newly inrroduced nodes state/airqualityhchomgm3 and state/airqualityco2ppm disppear. They are only repopulated after deleting and rejoining the device

It s because you have added your new devices types in de_web_plugin.cpp (used during creation) and not in database.cpp (used on loading)

triumvirrr, 2021-06-11:
Thanks for the hint with database.cpp, now a restart is possible without losing configuration :-)

Have made a new commit, also with some work on the redundant linefeeds.

I am not sure if it is wise to not create the Phoscon light device at all, because without this device it will not be visible in Phoscon at all.
And I like using Phoscon to at least rename and delete devices quite easily...

Is there a way to make a proper sensor device, ideally even with showing the sensor values, within Phoscon?

Smanar, 2021-06-11:
> this device it will not be visible in Phoscon at all

But at least temperature and humidity sensor will be visible, no ?

triumvirrr, 2021-06-12:
No, there is no sensor visible at all in Phoscon.
Phoscon _only_ shows the Smart Switch / Light thing.
The sensors are only available via REST.

BTW:
Either after my update of database.cpp or simply by chance, I was now able to pair a second device with all 10 sensors (2 x 5) visible and functioning via REST.

Smanar, 2021-06-12:
>Phoscon only shows the Smart Switch / Light thing.

Are you sure there is nothing in sensor tab ? From my memory phoscon don't use whitelist for them. And they can be visible on beta version, in help / API information http://phoscon.de/pwabeta

>second device with all 10 sensors

You have 10 sensors entries for only one device ?

triumvirrr, 2021-06-12:
Confirmed, Phoscon does not show this device within sensors, only within lights, also in the Beta version.

@sensors:
No, I have 10 sensor entries, because I now have both of my devices paired. Each device presents 5 sensors.

About the device itself:
I can now confirm the discussions, which can be found elsewhere regarding the quality of the product:
Temperature and Humidity readings differ significantly (2° temp or 10% hum) between two devices sitting next to each other. So not really reliable.

VOC delivers values I cannot really interpret and do not match my Develco VOC sensor at all, even with the assumption, that the value has to be corrected by some calibration. It does show a rise and decrease of the value in a comprehensible way, most of the time, though.

HCHO seems working, shows mostly nothing and goes up if confronted with nail polisher, for example. I have no idea, though, on how to interpret the number in terms of units. I assume that "1" means 0,1 or 0,01 mg/m3.

The CO2 sensor values seem to be "fake". It gives a reading of roughly 360 / 370 through the whole day in a more or less flat line. CO2 only shows a visible change, when you directly blow on the sensor, but it makes no change if you have the window open or if everything in closed and I sleep in the room, the value stays near 360 / 370 as stated above.

triumvirrr, 2021-06-14:
For all of you thinking about trying out this device, here are 24h readings of the sensors.
I have reduced the reporting of the values to once each 5 minutes to improve readability. The sensor sends updates each second, more or less...

Test setup:
2 Tuya test devices directly next to each other and to one Develco device, which I have used for some weeks now and which seems to deliver appropriate values for it's sensors (temp, hum and VOC)

All within a small room, which had the window repeatedly opened at closed until ~16:30, then closed all the time until now and door also closed.

CO2
Unlikely to be "true" values, one would expect this to change at least a bit according to being someone in the room or having the window open/closed.
Tuya 1:
[image]
Tuya 2:
[image]

VOC
The value changed between 1 and 2, so I did multiply it with 100 to get "similar" values compared to my Develco VOC sensor, which seems to give more plausible values (no comparison to a third sensor available, so I can only assume, that Develco's readings are good).
Tuya1:
[image]
Tuya 2:
[image]
Develco:
[image]

HCHO
The value changed between 0 and 1, mostly, climbing to 4 at some points for one sensor. I devided the value by 10, since I assume this to be a plausible value in mg/m3. Probably the right reading would be (value/100) for this unit. No other sensors available for HCHO for comparision.
Tuya 1
[image]
Tuya 2
[image]

Temperature
For both Tuya boxes I substracted 2 degrees permenantly in order to calibrate to a comparable result:
Tuya 1
[image]
Tuya 2
[image]
Develco
[image]

Humidity
I could not find a suitable value to calibrate Humidity, the values are differing per sensor, but seem to be more or less useful, when you do not expect too big acuracy. It seems, that one sensor permanently shows values ~6 percent points below the other one. I assume the values from Develco and Tuya 2 to be more or less correct, since they are matching readings from other sensors as well, which I have tested in the past.
Tuya 1:
[image]
Tuya 2:
[image]
Develco:
[image]

Smanar, 2021-06-14:
>The sensor sends updates each second, more or less...

And no way to reduce it for the moment ....

But I m sorry, but even for temperature it seem this device have a floating of 0.3 degree, not the more reliable ....
Not realy usefull to send a values every seconds if this value is so much variable.

triumvirrr, 2021-06-16:
Totally with you.

The temperature reading would not really work for my current smart home programming, since the threshold for my heating steering is 0.3 degrees above or below target temperature. This works perfectly with my Xiaomi sensors.

I will leave the sensors online and available for further testing, should someone else benefit from the integration of this sensor in deCONZ, but for my personal intended usecase (good representation of the air quality, mostly) they seem to be more or less worthless.

easybeat, 2021-06-17:
@triumvirrr : I've got this device yesterday and was wondering if it was already integrated in deconz. And now a bit frustrated about your results. Specially for CO2 values.

I'm trying to find a CO2 sensor since month which works with zigbee. Did you find one? (which gives accurate measurements)

Thanks for your help.

triumvirrr, 2021-06-18:
Hi @easybeat!

Unfortunately, I also had no luck in finding a decent Zigbee CO2 sensor.

I now went with Netatmo for CO2 (and other readings), Develco (Zigbee, deconz support) for VOC and our Tuya smart air box for HCHO. This is by far not the ideal solution, but OK for my purpose.

Netatmo offers a REST API and is well integrated within my central OpenHab solution. It is a cloud solution, though, which I alway try to avoid, if possible and quite costly.

My personal pros of Netatmo:
- reliable infrared CO2 measurement
- all other readings (which exactly depend on which product you use) seem reliable as well
- if you go for the wheather station (and not the pure indoor station), you can also connect wind and rain sensors, which is quite useful for a lot of automated steerings

But, summed up, I would also be glad for all hints about a functioning CO2 sensor via Zigbee.

triumvirrr, 2021-07-12:
Does anyone of the dev team want to merge my additions for the Tuya smart air box?

It works well for me and although the product does not match all my expectations, I would be really glad to see it supported in an official release.

Probably it will be of help to someone else with the same product, too.

Mimiix, 2021-07-12:
Can you link me the pr?

triumvirrr, 2021-07-12:
Of course, here we go:
https://github.com/triumvirrr/deconz-rest-plugin/tree/tuya_smart_airbox

Mimiix, 2021-07-12:
> Of course, here we go:
> https://github.com/triumvirrr/deconz-rest-plugin/tree/tuya_smart_airbox

That's not a pull request 😅

Mimiix, 2021-07-12:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4997 I think it's this one.

I've forwarded it to @manup

triumvirrr, 2021-07-12:
Sorry for the confusion, still new to this and not familiar with the terms...

triumvirrr, 2021-07-20:
Guys, can I be of any help here for advancing in getting the box officially supported?

Smanar, 2021-07-20:
Don't worry, I have some PR is waiting list since 2 months ^^
It can take time sometime.

jaydisc, 2021-07-28:
FWIW, I am keen for support of this device too.

rowdy238, 2021-08-17:
Any News?
I´ve also the tuya smart air box and would like to know if there is a chance to use it with all sensors?
Actually nothing is working, no sensor data...

Mimiix, 2021-08-17:
Asked @manup  to hceck the PR.

jaydisc, 2021-09-29:
Did this PR ever get merged? Is this device supported now?

Mimiix, 2021-09-29:
ITs still open @jaydisc

manup, 2021-09-29:
Small update: I have the sensor now for testing the PR here, but will be delayed for v2.13.1

Vartkat, 2021-11-01:
Congrat to those testing and devs. 

I bought 4 of these by mistake, discovering on arrival that they need USB powering. 
Now I'm wondering if I keep or return them... As at the begining of this discussion I only get a smart plug in Deconz. 

When can we expect a merge so that I can decide if I keep these ?
 As you say they are broadcasting every minute it seems my USB battery project won't last long, I'll have to find a way to power these...

Kllrv, 2021-12-23:
Looking forward to this device being supported.  Manufacturer _TZE200_yvx5lh6k and more recent Application Version than OP, otherwise appears to be the same device 
[image]

jytou, 2022-02-12:
Hi all,
purchased those before looking here... should have apparently. :( Any plans on getting them supported?
Thanks!

Smanar, 2022-02-13:
No, I think if you want to use it ATM, you need to compile the code.
PR are locked for the moment https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733