dresden-elektronik/deconz-rest-plugin #4992: ERROR LOG SPAM -  void zmController::fetchZdpTick(),6574: assertion '0' failed - 33x per second and other problems

Expello, 2021-06-10:
## My Environment
I use Windows 10 (native) with deCONZ v2.12.0 and Conbee II with FW 266E0700.
And an one meter extension cabel (thick and double shielded), and one USB 3 HDD.

## My issue´s
I user Tradfri LED´s, aqara window door sensors (MCCGQ11LM) and Heiman Smoke Sensors (HS1SA-E).
currently 56 nodes are online.

But the following errors I also have on my RPI4.

The first thing that i notice is that the ERROR L2 log is spammed with the error  "void zmController::fetchZdpTick(),6574: assertion '0' failed" ... und das 33x per second! 

The second problem...the network is never really stable. Sometimes deconz loses the usernames of the nodes, then I have to reassign them via phoscon. New nodes are partially recognized in deconz but only the ID is shown in deconz and therefore not displayed in phoscon, then I have to manually click on "read node description" or "simple desciptor" in deconz so that the name is read. 

Some TRADFRI LED's and remote controls sporadically lose the connection, a reset via F5 unfortunately never works...only power on off helps then.

I also set my wifi channel to 1...the zigbee channel is set to 15. but there was no difference.

I have already tested several firmware versions for the Conbee II stick...starting with 26580700 up to the current beta.
Also deCONZ from version v2.10.01.

All in all not very satisfying, so I can't use this in my house if it runs so unstable.
I hope you have a few more tips for me...I'm really in despair.

Thanks a lot!

## Screenshots

<details>
  <summary>deCONZ</summary>

[image]
</details>

<details>
  <summary>Phoscon</summary>

[image]
</details>

## deCONZ Logs

<details>
  <summary>ERROR LOG -></summary>

  ```
08:19:39:909 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:004 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:004 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:005 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:099 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:100 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:100 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:192 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:192 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:193 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:284 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:284 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:285 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:376 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:376 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:377 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:483 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:483 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:484 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:576 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:576 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:577 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:667 void zmController::fetchZdpTick(),6574: assertion '0' failed
08:19:40:667 void zmController::fetchZdpTick(),6574: assertion '0' failed
[... 2249 lines cut ...]
08:30:12:443 enqueue event config/localtime for /config/
08:30:12:473 Poll APS request to 0xD0CF5EFFFE308547 cluster: 0x0006 dropped, values are fresh enough
08:30:12:534 device state timeout ignored in state 3
08:30:12:641 void zmMaster::handleStateBusy(zmMaster::MasterEvent) unhandled event EVENT_GOT_STATUS
08:30:12:858 void zmMaster::handleStateBusy(zmMaster::MasterEvent) unhandled event EVENT_GOT_STATUS
08:30:12:934 Node 0x14B457FFFE686C13 is known by 16 neighbors, last seen 44312 s
08:30:13:119 device state timeout ignored in state 3
08:30:13:303 device state timeout ignored in state 3
08:30:13:428 void zmMaster::handleStateBusy(zmMaster::MasterEvent) unhandled event EVENT_GOT_STATUS
08:30:13:444 enqueue event config/localtime for /config/
08:30:13:490 Node 0x5C0272FFFECA6212 is known by 20 neighbors, last seen 1800 s
08:30:13:627 poll node 5c:02:72:ff:fe:ca:62:51-01
08:30:13:628 Poll light node DG Buero Wandleuchte 02
08:30:13:643 void zmMaster::handleStateBusy(zmMaster::MasterEvent) unhandled event EVENT_GOT_STATUS
08:30:13:688 Poll APS request to 0x5C0272FFFECA6251 cluster: 0x0006 dropped, values are fresh enough
08:30:13:918 device state timeout ignored in state 3
08:30:14:026 APS-DATA.confirm id: 84, status: 0x00 SUCCESS
08:30:14:027 APS-DATA.confirm request id: 84 -> confirmed, timeout 299434371
08:30:14:057 Node 0xD0CF5EFFFE308547 is known by 10 neighbors, last seen 268052 s
08:30:14:102 device state timeout ignored in state 3
08:30:14:225 void zmMaster::handleStateBusy(zmMaster::MasterEvent) unhandled event EVENT_GOT_STATUS
08:30:14:426 APS-DATA.confirm id: 84, status: 0x00 SUCCESS
08:30:14:427 APS-DATA.confirm id: 84, status: 0x00, match: 0
08:30:14:439 enqueue event config/localtime for /config/
08:30:14:440 void zmMaster::handleSt

Mimiix, 2021-06-10:
**I have no clue what you did to the template, as this isin;'t our template and the bot apparantly likes it. All info is there but please stick to the template next time.** 

There is a firmware bug with ikea lights https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261 

The one in deconz seems to be not paired correctly.  I have asked Manup to check on your issue with the ZDPTICK

Expello, 2021-06-10:
@Mimiix 

Thanks for your fast feedback. :-)

Sorry! with the template I was not so aware...if it is necessary...I write the request again in the template format.

#1261 - The bug seems to be solved with the last fw...or do i misunderstand the last entry of manup?

Mimiix, 2021-06-10:
@Expello No worries :) Keep it as is. 

The firmware of the Conbee and the latest firmware of the lights as well.

Expello, 2021-06-10:
@Mimiix 

ok the firmware of the lights...good idea! :-)

OTAU seems to be disabled...how can i activate that...i cant login to the WEB-APP. delight delight doesnt work.

[image]

[image]

Mimiix, 2021-06-10:
Delight would be the username. The password is the same as Phoscon.

Expello, 2021-06-10:
ok your are fast...yes it works... updates in process! :-)

manup, 2021-06-10:
Sorry the assert shouldn't be there in first place forgot to remove it after debugging :)
This will be fixed in the next version.

closed 2021-07-09 by github-actions[bot]