dresden-elektronik/deconz-rest-plugin #4994: Device name

kurcsics, 2021-06-10:
## Device

- Product name: Tuya 4 buttons scene switch
- Manufacturer: _TZ3000_vp6clf9d
- Model identifier: TS0044
- Device type :   - Switch

## Screenshots
[image]
[image]

### Basic
try with 12.0
Switch not present in switch editor.

Smanar, 2021-06-10:
He is not present in switch editor, but he is not visible in the API ?
Try with looking in http://phoscon.de/pwabeta in help/api Information.

Tuya device can need 2 inclusions without deleting the previous entry.

kurcsics, 2021-06-10:
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-06-10:
So the device is working, "buttonevent" change, the problem is more from phoscon side https://github.com/dresden-elektronik/phoscon-app-beta/issues

closed 2021-07-09 by github-actions[bot]