dresden-elektronik/deconz-rest-plugin #4995: Device name

kurcsics, 2021-06-10:
## Device

- Product name: CCT Zigbee Touch Remote Controller
- Manufacturer: Sunrichter
- Model identifier: ZGRC-TEUR-004
- Device type :   - Switch

## Screenshots
[image]
[image]
[image]

### Basic
try with 12.0
Switch not present in switch editor.

Mimiix, 2021-06-10:
Is it working with deconz rest api tho?

kurcsics, 2021-06-11:
Hi,
Its not working in Api, no buttonevent. 
[image]

Mimiix, 2021-06-11:
Can you check "event"?

kurcsics, 2021-06-11:
I have checked it, No event…

SwoopX, 2021-06-11:
For the button events, you could try adding your model ID to the `iluminizeMap` button map in file `button_maps.json`. The line should then look like this:

`"modelids": ["ZGRC-TEUR-003", "ZGRC-TEUR-004"],`

For the switch editor, you need to move over to phoscon beta repository.

kurcsics, 2021-06-12:
Thanks, its Work in API

closed 2021-07-11 by github-actions[bot]