dresden-elektronik/deconz-rest-plugin #4996: Conbee II stick no longer connects after version update 2.12.0

biiiink, 2021-06-10:
## Describe the bug

I updated the Conbee II stick to 2.12.0 in the Phoscon app.  After reboot, the stick will no longer connect

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Proxmox VM
 - Running method: Marthoc Docker container
 - Firmware version: Unsure
 - deCONZ version: 2.12.0
 - Device: ConBee II
 - Do you use an USB extension cable: yes - 3ft
 - Is there any other USB or serial devices connected to the host system? If so: Aeotec Zwave Z-Stick Gen5+

## deCONZ Logs

[marthoc/deconz] Starting deCONZ...
--
Thu, Jun 10 2021 11:41:33 am | [marthoc/deconz] Current deCONZ version: 2.12.00
Thu, Jun 10 2021 11:41:33 am | [marthoc/deconz] Web UI port: 8092
Thu, Jun 10 2021 11:41:33 am | [marthoc/deconz] Websockets port: 8443
Thu, Jun 10 2021 11:41:33 am | [marthoc/deconz] VNC Disabled
Thu, Jun 10 2021 11:41:33 am | QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Thu, Jun 10 2021 11:41:33 am | libpng warning: iCCP: known incorrect sRGB profile
Thu, Jun 10 2021 11:41:34 am | This plugin does not support propagateSizeHints()
Thu, Jun 10 2021 11:41:34 am | This plugin does not support propagateSizeHints()
Thu, Jun 10 2021 11:41:34 am | 11:41:33:379 HTTP Server listen on address 0.0.0.0, port: 8092, root: /usr/share/deCONZ/webapp/
Thu, Jun 10 2021 11:41:34 am | 11:41:33:516 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
Thu, Jun 10 2021 11:41:34 am | 11:41:33:597 parent process /bin/sh
Thu, Jun 10 2021 11:41:34 am | 11:41:33:597 gw run mode: normal
Thu, Jun 10 2021 11:41:34 am | 11:41:33:597 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB sqlite version 3.27.2
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB PRAGMA page_count: 0
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB PRAGMA page_size: 4096
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB PRAGMA freelist_count: 0
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB file size 0 bytes, free pages 0
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB PRAGMA user_version: 0
Thu, Jun 10 2021 11:41:34 am | 11:41:33:598 DB upgrade to user_version 1
Thu, Jun 10 2021 11:41:34 am | 11:41:34:056 DB write sqlite user_version 1
Thu, Jun 10 2021 11:41:34 am | 11:41:34:063 DB PRAGMA user_version: 1
Thu, Jun 10 2021 11:41:34 am | 11:41:34:063 DB upgrade to user_version 2
Thu, Jun 10 2021 11:41:34 am | 11:41:34:093 DB write sqlite user_version 2
Thu, Jun 10 2021 11:41:34 am | 11:41:34:102 DB PRAGMA user_version: 2
Thu, Jun 10 2021 11:41:34 am | 11:41:34:102 DB upgrade to user_version 6
Thu, Jun 10 2021 11:41:34 am | 11:41:34:154 DB write sqlite user_version 6
Thu, Jun 10 2021 11:41:34 am | 11:41:34:161 DB PRAGMA user_version: 6
Thu, Jun 10 2021 11:41:34 am | 11:41:34:161 DB upgrade to user_version 7
Thu, Jun 10 2021 11:41:34 am | 11:41:34:185 DB write sqlite user_version 7
Thu, Jun 10 2021 11:41:34 am | 11:41:34:196 DB PRAGMA user_version: 7
Thu, Jun 10 2021 11:41:34 am | 11:41:34:196 DB cleanup
Thu, Jun 10 2021 11:41:34 am | 11:41:34:196 DB create temporary views
Thu, Jun 10 2021 11:41:34 am | 11:41:34:197 started websocket server at port 8443
Thu, Jun 10 2021 11:41:34 am | 11:41:34:198 create default username and password
Thu, Jun 10 2021 11:41:34 am | 11:41:34:199 [INFO] - Found file containing button maps. Parsing data...
Thu, Jun 10 2021 11:41:34 am | 11:41:34:203 [INFO] - Button maps loaded.
Thu, Jun 10 2021 11:41:34 am | 11:41:34:203 dlg action: Read binding table
Thu, Jun 10 2021 11:41:34 am | 11:41:34:203 found node plugin: libde_rest_plugin.so - REST API Plugin
Thu, Jun 10 2021 11:41:34 am | 11:41:34:204 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
Thu, Jun 10 2021 11:41:34 am | 11:41:34:217 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
Thu, Jun 10 2021 11:41:37 am | 11:41:37:369 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:37 am | 11:41:37:529 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:37 am | 11:41:37:688 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:37 am | 11:41:37:848 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:009 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:168 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:329 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:489 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:649 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:38 am | 11:41:38:809 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:39 am | 11:41:38:969 device state timeout ignored in state 2
Thu, Jun 10 2021 11:41:39 am | 11:41:39:129 device state timeout (handled)
Thu, Jun 10 2021 11:41:39 am | 11:41:39:131 device disconnected reason: 1, index: 0
Thu, Jun 10 2021 11:41:39 am | 11:41:39:148 wait reconnect 15 seconds
Thu, Jun 10 2021 11:41:39 am | 11:41:39:209 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
Thu, Jun 10 2021 11:41:39 am | 11:41:39:209 device state timeout ignored in state 1
Thu, Jun 10 2021 11:41:39 am | 11:41:39:913 Announced to internet https://phoscon.de/discover
Thu, Jun 10 2021 11:41:40 am | 11:41:40:141 wait reconnect 14 seconds
Thu, Jun 10 2021 11:41:41 am | 11:41:41:141 wait reconnect 13 seconds
Thu, Jun 10 2021 11:41:42 am | 11:41:42:141 wait reconnect 12 seconds
Thu, Jun 10 2021 11:41:43 am | 11:41:43:141 wait reconnect 11 seconds
Thu, Jun 10 2021 11:41:44 am | 11:41:44:141 wait reconnect 10 seconds
Thu, Jun 10 2021 11:41:45 am | 11:41:45:141 wait reconnect 9 seconds
Thu, Jun 10 2021 11:41:46 am | 11:41:46:140 wait reconnect 8 seconds
Thu, Jun 10 2021 11:41:47 am | 11:41:47:141 wait reconnect 7 seconds
Thu, Jun 10 2021 11:41:48 am | 11:41:48:141 wait reconnect 6 seconds
Thu, Jun 10 2021 11:41:49 am | 11:41:49:141 wait reconnect 5 seconds
Thu, Jun 10 2021 11:41:49 am | 11:41:49:164 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
Thu, Jun 10 2021 11:41:50 am | 11:41:50:141 wait reconnect 4 seconds
Thu, Jun 10 2021 11:41:51 am | 11:41:51:140 wait reconnect 3 seconds
Thu, Jun 10 2021 11:41:52 am | 11:41:52:141 wait reconnect 2 seconds
Thu, Jun 10 2021 11:41:53 am | 11:41:53:141 wait reconnect 1 seconds
Thu, Jun 10 2021 11:41:54 am | 11:41:54:141 start reconnect to network

## Additional context

manup, 2021-06-10:
Not sure how the container device path is setup, but you may try the following approach to start the container which makes it resistant to device path changes which can be quite annoying in virtual environments and lead to the above error.

https://github.com/marthoc/docker-deconz/issues/298#issuecomment-732293745

Mimiix, 2021-06-10:
@manup would this be a deconz bug?

manup, 2021-06-10:
No this is more a general problem with USB devices within Docker, not specific to ConBee.
However the described workaround should work quite nicely, at least for the Docker container, can't comment on Proxmox if that adds extra challenges compared to native system.

biiiink, 2021-06-10:
> Not sure how the container device path is setup, but you may try the following approach to start the container which makes it resistant to device path changes which can be quite annoying in virtual environments and lead to the above error.
> 
> [marthoc/docker-deconz#298 (comment)](https://github.com/marthoc/docker-deconz/issues/298#issuecomment-732293745)

I confirmed that the path is  /dev/ttyACM1, which is what I was using 

I also tried using: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2120169-if00

Got the same result.

I also tried using the Home Assistant addon with the same results

biiiink, 2021-06-14:
Any help on this would be greatly appreciated

Smanar, 2021-06-14:
You have just updated deconz to the version 2.12 ? not the gateway firmware ?
Have you tried a rollback ?

biiiink, 2021-06-14:
> You have just updated deconz to the version 2.12 ? not the gateway firmware ?
> Have you tried a rollback ?

Yes.  the gateway firmware was not showing an update was available

Are there instructions somewhere on how to rollback?

biiiink, 2021-06-20:
Anyone have any suggestions??

Smanar, 2021-06-20:
For me the problem is from how the VM detect the USB device.
Check if basic command are working "GCFFlasher_internal -l" "lsusb" and ect ... But I don't use VM on my side.

ottelo9, 2021-06-24:
Hi, I solved the problem with a firmware update via Windows and used this tutorial: https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/

Now the stick is detected in Home Assistan again.

closed 2021-07-24 by github-actions[bot]