dresden-elektronik/deconz-rest-plugin #5002: Rounded Tradfri button

junalmeida, 2021-06-12:
## Describe the bug

Short press on power button used to send 1002 but stopped working out of sudden. All other x002 buttons works.

## Steps to reproduce the behavior

1. Press the button
2. Check logs

## Expected behavior

Should send 1002 event

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: v2.12.0-beta
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

```
08:58:19:831 [INFO] - No button handler for: TRADFRI remote control, broadcast to: 0xAFCE, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 38
08:58:25:789 Skip idle timer callback, too early: elapsed 597 msec 
``` 

For buttons that work I can see the correct button number on INFO log. Long press power button still sends 1001. Only short press does not work anymore.

## Additional context

closed 2021-06-12 by SwoopX

junalmeida, 2021-06-12:
@SwoopX when adding a "duplicate" label would be good to see to which ticket this is a duplicate. Thx

junalmeida, 2021-06-12:
And if possible which PR will fix it.

SwoopX, 2021-06-14:
Duplicate is #4900. It would equally be appreciated to use the search before raising an issue 😉

junalmeida, 2021-06-14:
Sometimes the view of a technical person with enough knowledge of the CODE involved is the best one to identify duplicates, as causes may vary, I'm sure you know that :)