dresden-elektronik/deconz-rest-plugin #5003: Aqara high precision motion sensor (RTCGQ13LM)

vinhthaipham, 2021-06-13:
Hi,
i am running deconz with RPI 3.
i manage to buy one from internet, 
https://www.reddit.com/r/Aqara/comments/kksfnw/quick_test_aqara_high_precision_motion_sensor/
but couldnt add it to deconz as it has not been there.
need some help to guideon how to manual add (source code) then recompile.
Thanks !!!

## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-06-13:
You need to have it added to deconz and have the screenshots to get it supported.

vinhthaipham, 2021-06-13:
thanks! I will do so

vinhthaipham, 2021-06-18:
any help please!

SwoopX, 2021-06-20:
The attributes are all on their defaults. you must read them while the device is awake. You can wake it up by pressing the physical button on it while hitting read within the following 1 - 3 secs.

SwoopX, 2021-07-12:
See #5109

closed 2021-07-12 by SwoopX